johanmeijer/grott #81: Some values wrong - MOD 9000TL3-X

szupi-ipuzs, 2021-12-03:
So, it seems I've got rerouting and proxy mode working, but not all parameters are shown correctly. This is how it looks in HA (via MQTT):

[image]

From the logs I suspect that my model uses another, not yet fully supported layout:

```
Dec 01 09:04:05 mediaserver grott[8973]:          - Grott automatic protocol detection
Dec 01 09:04:05 mediaserver grott[8973]:          - Grott data record length 585
Dec 01 09:04:05 mediaserver grott[8973]:          - layout   :  T060104X
Dec 01 09:04:05 mediaserver grott[8973]:          - no matching record layout found, try generic
Dec 01 09:04:05 mediaserver grott[8973]:          - Record layout used :  T06NNNNX
Dec 01 09:04:05 mediaserver grott[8973]:          - Growatt data decrypted V2
Dec 01 09:04:05 mediaserver grott[8973]:          - Grott Growatt data decrypted
Dec 01 09:04:05 mediaserver grott[8973]:          - Growatt plain data:
Dec 01 09:04:05 mediaserver grott[8973]:                  000c0006024101045847443542475434473400000000000000000000000000000000000000005
Dec 01 09:04:05 mediaserver grott[8973]:                  24a483242464d3039510000000000000000000000000000000000000000150c01090403020000
Dec 01 09:04:05 mediaserver grott[8973]:                  007c0001000016850eb80006000008d40e5400060000089800000000000000000000000000000
Dec 01 09:04:05 mediaserver grott[8973]:                  00000000000000000000000000000000000000000000000000000000000000000000000162f13
Dec 01 09:04:05 mediaserver grott[8973]:                  8908be000a000008be08c00009000007e008de000a000008de0eee0f6b0f4f000000010000008
Dec 01 09:04:05 mediaserver grott[8973]:                  7000463320000000000000030000000000000003c000000000000000000000000000000000000
Dec 01 09:04:05 mediaserver grott[8973]:                  0000000000000000000000000000000000000000000000000000000000000000006c020900ee0
Dec 01 09:04:05 mediaserver grott[8973]:                  0dd000000000bd80bec4e200000000186a0000000000000000000000000000000000000000500
Dec 01 09:04:05 mediaserver grott[8973]:                  1e0000000000000000000000000000000000000000007d00f9000000000000000000000000000
Dec 01 09:04:05 mediaserver grott[8973]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Dec 01 09:04:05 mediaserver grott[8973]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Dec 01 09:04:05 mediaserver grott[8973]:                  000000000000000000000000000000000000000000000000000000000000000000003e849b600
Dec 01 09:04:05 mediaserver grott[8973]:                  0000000015000b001e0015000c00010000000000002672002f001100410000000000000000000
Dec 01 09:04:05 mediaserver grott[8973]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Dec 01 09:04:05 mediaserver grott[8973]:                  00000000000016490000000000000000000000000000000000000000000000000000000000000
Dec 01 09:04:05 mediaserver grott[8973]:                  00000000000cfaf
Dec 01 09:04:05 mediaserver grott[8973]:          - Growatt new layout processing
Dec 01 09:04:05 mediaserver grott[8973]:                  - decrypt       :  True
Dec 01 09:04:05 mediaserver grott[8973]:                  - offset        :  6
Dec 01 09:04:05 mediaserver grott[8973]:                  - record layout :  T06NNNNX
```

What is decoded matches what can be seen in HA.

```
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvstatus             :  1
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvpowerin            :  576.5
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pv1voltage           :  376.8
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pv1current           :  0.6
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pv1watt              :  226.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pv2voltage           :  366.8
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pv2current           :  0.6
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pv2watt              :  220.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvpowerout           :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvfrequentie         :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridvoltage        :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridcurrent        :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridpower          :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridvoltage2       :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridcurrent2       :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridpower2         :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridvoltage3       :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridcurrent3       :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvgridpower3         :  37218394.5
Dec 01 09:04:05 mediaserver grott[8973]:                  -  totworktime          :  91.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvenergytoday        :  14877054.2
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvenergytotal        :  25867451.1
Dec 01 09:04:05 mediaserver grott[8973]:                  -  epvtotal             :  0.1
Dec 01 09:04:05 mediaserver grott[8973]:                  -  epv1today            :  13.5
Dec 01 09:04:05 mediaserver grott[8973]:                  -  epv1total            :  28753.8
Dec 01 09:04:05 mediaserver grott[8973]:                  -  epv2today            :  0.0
Dec 01 09:04:05 mediaserver grott[8973]:                  -  epv2total            :  4.8
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvtemperature        :  52.1
Dec 01 09:04:05 mediaserver grott[8973]:                  -  pvipmtemperature     :  0.0
```

szupi-ipuzs, 2021-12-03:
So I finally read about custom layouts in your wiki. Using your example layout for T060104X and grott seems to be using it. Still no change in the values though.
Also, I've noticed that:
**epv1today** matches what the growatt app shows as "total energy produced"
**epvtotal** matches the "energy produced today" in the app.

Also, I can see that **pvpowerin** seems to hold a correct value for current power (so it should really be mapped to pvpowerout).

I guess I can change in manually in the laout ini file...? Correct?

johanmeijer, 2021-12-04:
Oke that doesn't look good. Let's try to fix it. To be honest I haven't test this record (T060104X) with real 3 phase inverter data I think. 

Yes you can change the T060104X.json yourself of course. But if generic mapping is wrong I want to fix it or Grott in generic (hopefully you want to help with that). 

Powerin and powerout is not totally the same. Powerin is what the solar panels are producing. Power out is what the inverter delivers to the grit (=powerin - engery used/lost by inverter). 

It is a little a pity we can not get the values in a spread sheet from the Growatt site anymore.  It would help if you can give the log output from Grott (and then the original data with the /x in it) and the values in the app. 
I am not sure  what the Growatt app is able to show while I am not allowed to run it on my phone anymore (Company virus scanner stated it is not safe). 

If not all values are shown in the app we have to run a query at the Growatt website. Do want and can you run a simple python script if necessary (I will provide the code)? 

In he mean while I will test with your data to see if there is an easy fix.

szupi-ipuzs, 2021-12-04:
Sure, I will gladly help, please send me the script to run.
I have fiddled with the existing layouts and I got the best results with t060104xsph.json. I modified it a bit and it started to show the values correctly (but I also had to modify the HA yaml to use different mqtt fields). Here they are:
[MOD_9000TL3-X.zip](https://github.com/johanmeijer/grott/files/7653977/MOD_9000TL3-X.zip)

szupi-ipuzs, 2021-12-04:
If this helps I can also provide you with my credentials to growatt.com, but I don't know how to send a private msg via github... How can i contact you privately?

johanmeijer, 2021-12-04:
Thanks. 

I think I found the issue. It is indeed an other layout the the t060104x used now. As you discoverd the xsph looks the most like. 
I created a new layout file t060104xtl3 and put it in /examples/Record Layout/  (branch 2.6). You can copy this one in grott main directory and specify invtype = tl3 in the [Generic] session of .ini. Grott will then select T010604XTL3 record layout. 

In the meanwhile I am going to investigate if I can detect the difference between inverters in the data record. So I can improve the grott automatic detection (and you don't have to specify invtype in the future).

johanmeijer, 2021-12-04:
> Sure, I will gladly help, please send me the script to run. I have fiddled with the existing layouts and I got the best results with t060104xsph.json. I modified it a bit and it started to show the [...]

In the layout file, It is also possible to create 2 field names with the same offset. So if the fieldnames does comply with the HA yaml you can add the fieldnames that are defined in the YAML. Example: 

    "eactoday"          : {"value" :370, "length" : 4, "type" : "num", "divide" : 10}, 
    "pvenergytoday"     : {"value" :370, "length" : 4, "type" : "num", "divide" : 10},

szupi-ipuzs, 2021-12-04:
Great! I will try the new layout tomorrow and let you know.
Thanks!

szupi-ipuzs, 2021-12-05:
The new layout seems to be correct. So I'm closing this issue. Thank you!

closed 2021-12-05 by szupi-ipuzs

logicalshrapnel, 2023-10-07:
I'm on a MOD 15000TL3-X and it seems to work with the same layout.

mikesdnielsen, 2023-12-20:
I'm sorry to post on a closed topic, but I'm facing the same issue. I have the latest version of grott, I think (I checked out the master branch). 

I have a MOD 9000TL3-X as well. 

Values are wrong on some of the data. I've looked the example record t060104xtl3.json, but I'm not sure I understand how to make grott use this layout. 

Johan mentioned something about copying it, but copying the file to my grott directory `/opt/grott/` didn't seem to fix the issue and values are still off. 

[image]

Any help would be highly appreciated - and thanks for making grott, so that we can use our own data 🙂

johanmeijer, 2023-12-24:
I think for the MOD type of inverter you can better use the MIN layout. You can specify this with the option invtype=min (tl3 is also an option but might not be the best one for this inverter). 

If you use a docker image you have to specify the -e ginvtype=min (and use the grott:2.8.2 image).

mikesdnielsen, 2023-12-24:
> I think for the MOD type of inverter you can better use the MIN layout. You can specify this with the option invtype=min (tl3 is also an option but might not be the best one for this inverter).
> 
> If you use a docker image you have to specify the -e ginvtype=min (and use the grott:2.8.2 image).

I tried the `min` format, but nothing seems to be appearing. I'll keep tinkering over Christmas. 

Merry Christmas and thanks for all your work on grott.

johanmeijer, 2023-12-24:
We will wait for the results after Christmas and Merry Christmas for you to!

techmantel, 2024-03-26:
I'm also having a MOD 8000TL3-X and was trying out Grott during the weekend when I faced the same issue.

@mikesdnielsen did you get this working on yours, and if so, how?

techmantel, 2024-03-29:
I tried out `ginvtype=tl3` and that seems to work pretty ok actually. At least the values I'm interested in are working now 👍