johanmeijer/grott #82: Invalid retrieved values for my MOD 5000TL3-X

JasperE84, 2021-12-11:
Today I installed a MOD 5000TL3-X inverter and started up a grott docker container.
However, the returned values seem incorrect, any ideas how I can fix this?

```
 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.16.5', 36666), raddr=('47.91.67.66', 5279)>
 - Growatt original Data:
		 \x00\x9a\x00\x06\x02\x41\x51\x04\x09\x33\x2c\x42\x20\x40\x41\x77\x21\x5f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3d\x32\x29\x45\x36\x76\x45\x5f\x47\x25\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61
		 \x74\x74\x3b\x72\x6e\x77\x61\x75\x67\x4e\x12\x6f\x76\x61\x74\x74\xb7\x77\x08
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x76
		 \x6e\x67\xfc\x4e\x56\x6f\x71\x61\x74\x71\x3b\x7b\x5a\x77\x67\x74\x74\x42\xf4
		 \x66\x40\x61\x72\x74\x47\x77\xe8\x78\x85\x7b\x95\x48\x85\x6f\x77\x61\x7b\x74
		 \x47\x72\x7f\x77\x61\xf2\x8b\x47\x72\x6f\x7f\x61\x74\x74\x4f\x72\x6f\x77\x66
		 \x74\x74\x47\x75\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x6e\x76
		 \x20\x46\x4b\x6e\x5d\x61\x74\x74\x47\x7e\x29\x7b\x21\x3a\x54\x47\x72\x6f\x77
		 \xb9\x8a\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x6a\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x3a\x72\x96\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x70\x44\x3e\xd7\x74\x74\x47\x72\x6f\x62\x61\x78\x74\x4d\x72\x7a\x77\x6d
		 \x74\x7f\x47\x72\x6f\x77\x61\x74\x1a\x69\x72\x07\x77\x78\x74\x3a\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75\x34\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xa6\x2c
 - Grott automatic protocol detection
 - Grott data record length 585
 - layout   :  T065104X
 - no matching record layout found, try generic
 - Record layout used :  T06NNNNX
 - Growatt data decrypted V2
 - Grott Growatt data decrypted
 - Growatt plain data:
		 009a0006024151044e41433541343530533000000000000000000000000000000000000000005
		 24548314231373030440000000000000000000000000000000000000000000000000000020000
		 007c00010000011309600001000000f0056700000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000010f13
		 88092400060000057c093500060000058609370006000005870fe40fe10ff70000000f0000001
		 0000086ff00000008000000080000000700000007000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000000000000000000000000000f025401390
		 12a000000000c460c404e2000000000d8fe000000000000000000000000000000000000000000
		 1e0000000000000000000000000000000000000000007d00f9000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000000000000000000000000000022b49b600
		 0000000015000c000a0015000c000b0000000000006e2e00680019004e0000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000001400000000000000000000000000000000000000000000000000000000000000
		 00000000000d443
 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX

 - Grott data record date/time processing started
 - no or no valid time/date found, grott server time will be used (buffer records not sent!)
 - Grott values retrieved:
		 -  pvserial             :  REH1B17***
		 -  pvstatus             :  1
		 -  pvpowerin            :  27.5
		 -  pv1voltage           :  240.0
		 -  pv1current           :  0.1
		 -  pv1watt              :  24.0
		 -  pv2voltage           :  138.3
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pvpowerout           :  0.0
		 -  pvfrequentie         :  0.0
		 -  pvgridvoltage        :  0.0
		 -  pvgridcurrent        :  0.0
		 -  pvgridpower          :  0.0
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  1776525.6
		 -  totworktime          :  54.6
		 -  pvenergytoday        :  9273750.8
		 -  pvenergytotal        :  26640792.7
		 -  epvtotal             :  1.5
		 -  epv1today            :  1.6
		 -  epv1total            :  3455.9
		 -  epv2today            :  0.8
		 -  epv2total            :  0.8
		 -  pvtemperature        :  59.6
		 -  pvipmtemperature     :  0.0
 - MQTT jsonmsg:
				 {"device": "REH1B17***", "time": "2021-12-11T15:56:30", "buffered": "no",
				 "values": {"pvstatus": 1, "pvpowerin": 275, "pv1voltage": 2400,
				 "pv1current": 1, "pv1watt": 240, "pv2voltage": 1383, "pv2current": 0,
				 "pv2watt": 0, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0,
				 "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2":
				 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0,
				 "pvgridpower3": 17765256, "totworktime": 393216, "pvenergytoday": 92737508,
				 "pvenergytotal": 266407927, "epvtotal": 15, "epv1today": 16, "epv1total":
				 34559, "epv2today": 8, "epv2total": 8, "pvtemperature": 596,
				 "pvipmtemperature": 0}}
 - MQTT message message sent
 - Grott Send data to PVOutput disabled
 - Grott InfluxDB publihing started
 - Timezone local specified default timezone used
 - Grott original time :  2021-12-11T15:56:30 adjusted UTC time for influx :  2021-12-11T14:56:30
 - Grott influxdb jsonmsg:
				 [{'measurement': 'REH1B17***', 'time': '2021-12-11T14:56:30', 'fields':
				 {'pvserial': 'REH1B17***', 'pvstatus': 1, 'pvpowerin': 275, 'pv1voltage':
				 2400, 'pv1current': 1, 'pv1watt': 240, 'pv2voltage': 1383, 'pv2current': 0,
				 'pv2watt': 0, 'pvpowerout': 0, 'pvfrequentie': 0, 'pvgridvoltage': 0,
				 'pvgridcurrent': 0, 'pvgridpower': 0, 'pvgridvoltage2': 0, 'pvgridcurrent2':
				 0, 'pvgridpower2': 0, 'pvgridvoltage3': 0, 'pvgridcurrent3': 0,
				 'pvgridpower3': 17765256, 'totworktime': 393216, 'pvenergytoday': 92737508,
				 'pvenergytotal': 266407927, 'epvtotal': 15, 'epv1today': 16, 'epv1total':
				 34559, 'epv2today': 8, 'epv2total': 8, 'pvtemperature': 596,
				 'pvipmtemperature': 0}}]
 - Grott write to influxdb v2
 - Grott extension processing disabled
```

Stats from growatt UI:
[image]

JasperE84, 2021-12-11:
I managed to fix this with the help from the layout proposed in this topic: https://github.com/johanmeijer/grott/issues/81

closed 2021-12-11 by JasperE84

johanmeijer, 2021-12-11:
Hi @JasperE84 ,

I will create a container that supports the TL3-X as described in issue #81 the comming days. @Tacoos33 (issue79) is also waiting for it.

I really like to have feedback on what is missing or wrong  in the t060104xtl3.json lay-out mapping.

If you create your own container it might be a good idea to use 2.6 (2.6.1f) as basis. It is quite simplex to create a container. It only needs the t060104xtl3.json in the main directory (or a link to it in the docker run command) and -e ginvtype="tl3" at the run command.

JasperE84, 2021-12-11:
I'll be glad to inform you soon. I mapped the JSON in /app using a volume in docker compose, the LAN  gateway should start sending data tomorrow again when the sun comes up (I suppose that's how it works)

JasperE84, 2021-12-12:
Seems to work ok (i stripped the json somewhat for influxdb data reduction):

Growatt App:
[image]

Grott log:
```
         - Grott values retrieved:
                 -  pvserial             :  REH1B1****
                 -  pvstatus             :  1
                 -  pvpowerin            :  35.4
                 -  pv1voltage           :  142.8
                 -  pv1current           :  0.0
                 -  pv1watt              :  0.0
                 -  pv2voltage           :  141.0
                 -  pv2current           :  0.1
                 -  pv2watt              :  14.1
                 -  pv3voltage           :  0.0
                 -  pv3current           :  0.0
                 -  pv3watt              :  0.0
                 -  pvpowerout           :  34.9
                 -  pvfrequentie         :  50.0
                 -  pvgridvoltage        :  234.3
                 -  pvgridcurrent        :  0.6
                 -  pvgridpower          :  140.5
                 -  pvgridvoltage2       :  237.0
                 -  pvgridcurrent2       :  0.6
                 -  pvgridpower2         :  142.2
                 -  pvgridvoltage3       :  236.4
                 -  pvgridcurrent3       :  0.7
                 -  pvgridpower3         :  165.4
                 -  Vac_RS               :  407.8
                 -  Vac_ST               :  407.9
                 -  Vac_TR               :  409.7
                 -  pvenergytoday        :  0.0
                 -  pvenergytotal        :  1.6
                 -  totworktime          :  5.7
                 -  epv1today            :  0.0
                 -  epv1total            :  0.8
                 -  epv2today            :  0.0
                 -  epv2total            :  0.7
                 -  epvtotal             :  1.5
                 -  pvtemperature        :  54.1
                 -  pvipmtemperature     :  25.1
                 -  pvboottemperature    :  24.1
                 -  temp4                :  0.0
                 -  uwBatVolt_DSP        :  0.0
                 -  pbusvolt             :  3149
                 -  nbusvolt             :  3145

```

What I did to get this to work:

Mount json from local path in container using docker-compose:
```
  grott:
    image: ledidobe/grott:latest
    container_name: grott
    restart: always
    networks:
      - hassnet
    ports:
      - "5279:5279"
    volumes:
      - ./data/grott/t065104X.json:/app/t065104X.json

    environment:
      - gblockcmd=False
      - gnomqtt=False
[... 30 lines cut ...]
    "pvgridpower2"      : {"value" :334, "length" : 4, "type" : "num", "divide" : 10},
    "pvgridvoltage3"    : {"value" :342, "length" : 2, "type" : "num", "divide" : 10},
    "pvgridcurrent3"    : {"value" :346, "length" : 2, "type" : "num", "divide" : 10},
    "pvgridpower3"      : {"value" :350, "length" : 4, "type" : "num", "divide" : 10},
    "Vac_RS"            : {"value" :358, "length" : 2, "type" : "num", "divide" : 10},
    "Vac_ST"            : {"value" :362, "length" : 2, "type" : "num", "divide" : 10},
    "Vac_TR"            : {"value" :366, "length" : 2, "type" : "num", "divide" : 10},
    "pvenergytoday"     : {"value" :370, "length" : 4, "type" : "num", "divide" : 10},
    "pvenergytotal"     : {"value" :378, "length" : 4, "type" : "num", "divide" : 10},
    "totworktime"       : {"value" :386, "length" : 4, "type" : "num", "divide" : 7200},
    "epv1today"         : {"value" :394, "length" : 4, "type" : "num", "divide" : 10},
    "epv1total"         : {"value" :402, "length" : 4, "type" : "num", "divide" : 10},
    "epv2today"         : {"value" :410, "length" : 4, "type" : "num", "divide" : 10},
    "epv2total"         : {"value" :418, "length" : 4, "type" : "num", "divide" : 10},
    "epvtotal"          : {"value" :522, "length" : 4, "type" : "num", "divide" : 10},
    "pvtemperature"     : {"value" :530, "length" : 2, "type" : "num", "divide" : 10},
    "pvipmtemperature"  : {"value" :534, "length" : 2, "type" : "num", "divide" : 10},
    "pvboottemperature" : {"value" :538, "length" : 2, "type" : "num", "divide" : 10},
    "temp4"             : {"value" :542, "length" : 2, "type" : "num", "divide" : 10},
    "uwBatVolt_DSP"     : {"value" :546, "length" : 2, "type" : "num", "divide" : 10},
    "pbusvolt"          : {"value" :550, "length" : 2, "type" : "num", "divide" : 1},
    "nbusvolt"          : {"value" :554, "length" : 2, "type" : "num", "divide" : 1}
     } }

```