johanmeijer/grott #84: Shouldn't MQTT messages be retained considering the way ShineLan box sends updates?

JasperE84, 2021-12-12:
I noticed after sundown, there is no traffic from my ShineLan box with packets Grott considers large enough to process. This means, I think, no updates are sent once the solar production reaches 0. The inverter also seems to turn itself in power saving mode after sundown. Hence no frequent updates are sent.

I also noticed the state of most sensors in home assistant switch to 'unknown', resulting in warnings in the hass interface cards.
[image]
[image]

Now MQTT has an option which tells the broker to retain the last known value of a topic. The mosquitto broker can also be configured for retained values to [persist after a broker restart](https://stackoverflow.com/questions/36126725/how-to-retain-messages-in-mqtt-if-the-mosquitto-broker-goes-down).

This retain option in a MQTT topic specifically disabled here: https://github.com/johanmeijer/grott/blob/c6bf04af3c7dd49427938f0a41f2fa821c5cd1da/grottdata.py#L342

Shouldn't the last sent value, for instance PV output wattage = 0 on sundown, be retained until the sun comes up again?

johanmeijer, 2021-12-13:
I am going to look at it. Thanks!

johanmeijer, 2021-12-14:
We can tried to add retain to the MQTT publish from grott. I doesn't hurrt I think. 

I am not sure if this helps with the unknown status of home automation. If I understand it from the documentation the retained MQTT message will be send to the consumer (in this case HA) the moment it subscribes to the topic. I expect HA "listen" to the topic as long it is active (otherwise it might miss messages) and the retain msg only will be read after a restart of HA. 

But again we can try it. I will change and test in it Grott and will publish it after that.

johanmeijer, 2022-02-06:
@JasperE84 in 2.7.1 version (beta) I added a parameter to enable MQTT message retain.  

You can enable this by specifying: 

* . ini [MQTT section] retain = True
* environmental gmqttretain = "True" (docker: -e gmqttretain = "True")

Docker 2.7.1 build is not available yet. I will build it soon (after testing the regular version for a couple of days).

JasperE84, 2022-02-20:
Thanks I'll test that soon!

JasperE84, 2022-03-27:
Testing now :)

johanmeijer, 2022-04-28:
@JasperE84 does it works ?

JasperE84, 2022-04-28:
Hi Johan, I have updated and configured last februari, but havent tested restarting hass or hass and mosquitto toghether, properly yet after sundown. I'll see when I have some time available to do this. Maybe this evening.

The flag itself seems to work though. The retain flag comes trough in mosquitto.

```
$ mosquitto_sub -F ' %r : %q : %m : %l : %t : %j' -q 2 -t '#'

1 : 0 : 0 : 876 : energy/growatt : {"tst":"2022-04-28T12:56:43.964868+0000","topic":"energy/growatt","qos":0,"retain":1,"payloadlen":876,"payload":"{\"device\": \"...\", \"time\": \"2022-04-28T14:55:10\", \"buffered\": \"no\", \"values\": {\"datalogserial\": \"...\", \"pvserial\": \"REH1B1700D\", \"pvstatus\": 1, \"pvpowerin\": 45456, \"pv1voltage\": 2460, \"pv1current\": 96, \"pv1watt\": 23616, \"pv2voltage\": 2299, \"pv2current\": 95, \"pv2watt\": 21840, \"pvpowerout\": 43621, \"pvfrequentie\": 5002, \"pvgridvoltage\": 2413, \"pvgridcurrent\": 60, \"pvgridpower\": 14478, \"pvgridvoltage2\": 2402, \"pvgridcurrent2\": 59, \"pvgridpower2\": 14171, \"pvgridvoltage3\": 2432, \"pvgridcurrent3\": 60, \"pvgridpower3\": 14592, \"Vac_RS\": 4117, \"Vac_ST\": 4189, \"Vac_TR\": 4244, \"pvenergytoday\": 214, \"pvenergytotal\": 14719, \"totworktime\": 9792369, \"epv1today\": 118, \"epv1total\": 8298, \"epv2today\": 100, \"epv2total\": 6527, \"epvtotal\": 14825, \"pvtemperature\": 731, \"pvipmtemperature\": 692, \"pvboottemperature\": 695, \"uwBatVolt_DSP\": 0, \"pbusvolt\": 3273, \"nbusvolt\": 3227}}"}
```

The %r retain flag shows 1 which indicates grott sends it with retain flag enabled.

JasperE84, 2022-04-28:
Yep, works. Thanks

closed 2022-04-28 by JasperE84

gijs007, 2022-08-14:
@johanmeijer Please add these commands to the Wiki documentation :)