johanmeijer/grott #85: Invalid parsing of T065150X resulting in erroneous production spikes

JasperE84, 2021-12-14:
The stats I'm seeing in home assistant had some huge production spikes in it. So I then started to check what caused these.
I'm seeing some messages with huge production numbers in it. The one below is an example which outputs invalid data:
````
         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.16.5', 42900), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x4a\x00\x06\x02\x41\x51\x50\x09\x33\x2c\x42\x20\x40\x41\x77\x21\x5f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x3d\x32\x29\x45\x36\x76\x45\x5f\x47\x25\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x78\x7a\x4c\x71\x4f\x75\x61
                 \x74\x74\x3b\x72\x6e\x77\x61\x62\xab\x4d\xd6\x6f\x7c\x61\x74\x7f\xf3\x7b\x4f
                 \x77\x6a\x74\x74\x4d\x7b\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x61
                 \xe9\x67\xf0\x4e\x62\x6f\x7d\x61\x74\x7d\x57\x7b\x41\x77\x6b\x74\x74\x4e\x5c
                 \x66\x5a\x61\x7f\x74\x47\x78\x78\x78\xaf\x7b\xa6\x48\xa8\x6f\x77\x61\x71\x74
                 \x47\x72\x4e\x77\x63\x49\x67\x47\x72\x6f\x75\x61\x74\x74\x48\x72\x6f\x77\x63
                 \x74\x74\x47\x7c\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x7c\x76
                 \x3a\x46\x32\x6e\x4f\x61\x74\x74\x47\x7e\x56\x7b\x46\x3a\x54\x47\x72\x6f\x77
                 \xb9\x8a\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x6b\x74\x6a\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x3a\x72\x96\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x70\x44\x3e\xd7\x74\x74\x47\x72\x6f\x62\x61\x78\x74\x4a\x72\x7a\x77\x6d
                 \x74\x7a\x47\x72\x6f\x77\x61\x74\x13\xc8\x72\x56\x77\x41\x74\x2d\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x62\xa1\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xfc\x90
         - Grott automatic protocol detection
         - Grott data record length 585
         - layout   :  T065150X
         - no matching record layout found, try generic
         - Record layout used :  T06NNNNX
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 004a0006024151504e41433541343530533000000000000000000000000000000000000000005
                 24548314231373030440000000000000000000000000000000000000000150c0e0b0320020000
                 007c0001000016df0aa4000b00000bb40920000b00000a0900000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000168813
                 840910000a00000910092e000a0000092e092d000b00000a170fce0fd20fda000000050000002
                 100023d13000000020000000f000000020000000e000000000000000000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000000001d024e01400
                 138000000000c390c274e2000000000d8fe000000000000000000000000000000000000000a00
                 1e0000000000000000000000000000000000000000007d00f9000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000000000022b49b600
                 0000000015000c000d0015000c000e000000000000678f0039002000590000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000016d50000000000000000000000000000000000000000000000000000000000000
                 000000000008eff
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06NNNNX

         - Grott data record date/time processing started
         - date-time:  2021-12-14T11:03:32
         - Grott values retrieved:
                 -  pvserial             :  REH1B*****
                 -  pvstatus             :  1
                 -  pvpowerin            :  585.5
                 -  pv1voltage           :  272.4
                 -  pv1current           :  1.1
                 -  pv1watt              :  299.6
                 -  pv2voltage           :  233.6
                 -  pv2current           :  1.1
                 -  pv2watt              :  256.9
                 -  pvpowerout           :  0.0
                 -  pvfrequentie         :  0.0
                 -  pvgridvoltage        :  0.0
                 -  pvgridcurrent        :  0.0
                 -  pvgridpower          :  0.0
                 -  pvgridvoltage2       :  0.0
                 -  pvgridcurrent2       :  0.0
                 -  pvgridpower2         :  0.0
                 -  pvgridvoltage3       :  0.0
                 -  pvgridcurrent3       :  0.0
                 -  pvgridpower3         :  37801664.4
                 -  totworktime          :  100.1
                 -  pvenergytoday        :  16928353.4
                 -  pvenergytotal        :  26542485.8
                 -  epvtotal             :  0.5
                 -  epv1today            :  3.3
                 -  epv1total            :  14670.7
                 -  epv2today            :  0.2
                 -  epv2total            :  1.5
                 -  pvtemperature        :  59.0
                 -  pvipmtemperature     :  0.0
         - MQTT jsonmsg:
                         {"device": "REH1B1*****", "time": "2021-12-14T11:03:32", "buffered": "yes",
                         "values": {"pvstatus": 1, "pvpowerin": 5855, "pv1voltage": 2724,
                         "pv1current": 11, "pv1watt": 2996, "pv2voltage": 2336, "pv2current": 11,
                         "pv2watt": 2569, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0,
                         "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2":
                         0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0,
                         "pvgridpower3": 378016644, "totworktime": 720896, "pvenergytoday":
                         169283534, "pvenergytotal": 265424858, "epvtotal": 5, "epv1today": 33,
                         "epv1total": 146707, "epv2today": 2, "epv2total": 15, "pvtemperature": 590,
                         "pvipmtemperature": 0}}
         - MQTT message message sent
         - Grott Send data to PVOutput disabled
         - Grott Send data to Influx disabled
         - Grott extension processing disabled
````

[image]

johanmeijer, 2021-12-14:
Hi Jasper,

You have TL3-X inverter. The record layout is a little bit different then a normal T065104X. I just updated the 2.6 branche to 2.6.1g (in the comments I  erroneously  mention 2 .6.1h).  It now should be able to proces the tl3-x records as well. 

I this moment I am not able yet to detect the TL3-X automatically (working on that) so you have to specify: 

      invtype=tl3 

in the [Generic] section of the .ini file 

(or env variable : -e ginvtype=tl3 for docker run, docker 2.6.1g build will be available soon)

This is how your data will look with 2.6.1g 

[image]

JasperE84, 2021-12-14:
Cool I'll try that when the new docker is released.

Please note that I already managed to parse the T065**104**X layout fine with the steps I described in #82.

This error in this issue is caused by the T065**150**X layout.
I noticed transmission with this layout after Grott hadn't been relaying data to Growatt for a while due to a misconfigured Influxdb server which caused an exception in Grott.

JasperE84, 2021-12-14:
Great!
Containter updated and ready, will know tomorrow if its working when the data starts flowing after sun up!

JasperE84, 2021-12-15:
Fix confirmed \0/
Parsing the MOD 5000TL3X  data works fine now on the 2.6.1g docker version without custom jsons mounted in the volume

```
         - layout   :  T065150XTL3
         - no matching record layout found, try generic
         - Record layout used :  T06NNNNXTL3
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 00240006024151504e41433541343530533000000000000000000000000000000000000000005
                 24548314231373030440000000000000000000000000000000000000000150c0f091b02020000
                 007c0001000000ca05ac000000000000058200010000008d00000000000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000000000000000c713
                 870910000700000658092900070000066909250007000006660fca0fc20fd7000000000000002
                 b0002c8c400000000000000150000000000000013000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000028020600df0
                 0da000000000c2f0c284e2000000000d8fe000000000000000000000000000000000000000000
                 1e0000000000000000000000000000000000000000007d00f9000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000000000019249b600
                 0000000015000c000e0015000c000f00000000000061a600960019007d0000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000002f80000000000000000000000000000000000000000000000000000000000000
                 00000000000bcd2
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06NNNNXTL3

         - Grott data record date/time processing started
         - date-time:  2021-12-15T09:27:02
         - Grott values retrieved:
                 -  datalogserial        :  NAC5A*****
                 -  pvserial             :  REH1B1700D
                 -  pvstatus             :  1
                 -  pvpowerin            :  20.2
                 -  pv1voltage           :  145.2
                 -  pv1current           :  0.0
                 -  pv1watt              :  0.0
                 -  pv2voltage           :  141.0
                 -  pv2current           :  0.1
                 -  pv2watt              :  14.1
                 -  pvpowerout           :  19.9
                 -  pvfrequentie         :  50.0
                 -  pvgridvoltage        :  232.0
                 -  pvgridcurrent        :  0.7
                 -  pvgridpower          :  162.4
                 -  pvgridvoltage2       :  234.5
                 -  pvgridcurrent2       :  0.7
                 -  pvgridpower2         :  164.1
                 -  pvgridvoltage3       :  234.1
                 -  pvgridcurrent3       :  0.7
                 -  pvgridpower3         :  163.8
                 -  Vac_RS               :  404.2
                 -  Vac_ST               :  403.4
                 -  Vac_TR               :  405.5
                 -  pvenergytoday        :  0.0
                 -  pvenergytotal        :  4.3
                 -  totworktime          :  25.3
                 -  epv1today            :  0.0
                 -  epv1total            :  2.1
                 -  epv2today            :  0.0
                 -  epv2total            :  1.9
                 -  epvtotal             :  4.0
                 -  pvtemperature        :  51.8
                 -  pvipmtemperature     :  22.3
                 -  pvboottemperature    :  21.8
                 -  uwBatVolt_DSP        :  0.0
                 -  pbusvolt             :  3119
                 -  nbusvolt             :  3112
```

closed 2021-12-15 by JasperE84