johanmeijer/grott #86: Grott crashes if configured with influxdb 2.1.1 and infinite retention period

JasperE84, 2021-12-19:
If the bucket retention period on influxdb 2.1.1 is set to infinite, then the retention variable value is set to zero seconds.
This causes a crash on grott startup in grottconf.py. 
I think because the used influxdb client lib doesn't support infinite retention on the latest influxdb version.

I added a stack trace log to grottconf.py at the moment the exception "Grott error: can not contact InfluxDB" was printed. 
Here is the result.

I solved this problem for now by setting a 365d retention period on the influxdb bucket. Then everything works just fine.

```
Grott Influxdb initialisation error
Traceback (most recent call last):
  File "/app/grottconf.py", line 167, in __init__
    buckets = self.ifbucket_api.find_bucket_by_name(self.ifbucket)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/client/bucket_api.py", line 83, in find_bucket_by_name
    buckets = self._buckets_service.get_buckets(name=bucket_name)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/service/buckets_service.py", line 501, in get_buckets
    (data) = self.get_buckets_with_http_info(**kwargs)  # noqa: E501
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/service/buckets_service.py", line 601, in get_buckets_with_http_info
    urlopen_kw=urlopen_kw)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 345, in call_api
    _preload_content, _request_timeout, urlopen_kw)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 182, in __call_api
    return_data = self.deserialize(response_data, response_type)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 254, in deserialize
    return self.__deserialize(data, response_type)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 293, in __deserialize
    return self.__deserialize_model(data, klass)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 642, in __deserialize_model
    kwargs[attr] = self.__deserialize(value, attr_type)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 271, in __deserialize
    for sub_data in data]
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 271, in <listcomp>
    for sub_data in data]
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 293, in __deserialize
    return self.__deserialize_model(data, klass)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 642, in __deserialize_model
    kwargs[attr] = self.__deserialize(value, attr_type)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 271, in __deserialize
    for sub_data in data]
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 271, in <listcomp>
    for sub_data in data]
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 293, in __deserialize
    return self.__deserialize_model(data, klass)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/api_client.py", line 644, in __deserialize_model
    instance = klass(**kwargs)
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/domain/bucket_retention_rules.py", line 51, in __init__
    self.every_seconds = every_seconds
  File "/usr/local/lib/python3.7/site-packages/influxdb_client/domain/bucket_retention_rules.py", line 96, in every_seconds
    raise ValueError("Invalid value for `every_seconds`, must be a value greater than or equal to `1`")  # noqa: E501
ValueError: Invalid value for `every_seconds`, must be a value greater than or equal to `1`
```

johanmeijer, 2021-12-20:
Ok that is strange I use InfluxDB 2.0.3 (docker container) and my database has retention period forever defined (that is what I want for now). I do not have any problems. 

I installed influx 2.1.1 and also get  the every_seconds error. 

So it might has to do something with InfluxDB releases and python libs? 

I have to dive deeper into this.

johanmeijer, 2021-12-25:
@JasperE84 

It seems to be an error in the influxdb_client python library. 

After a pip uninstall and pip install of this library  (upgrade it to 1.24.0) it works fine!

JasperE84, 2021-12-25:
That's good to hear, nice work :) I guess the update will make its way to a future docker rls.

johanmeijer, 2021-12-25:
@JasperE84

Oh yes almost forgotten I have to update the docker container :)

I will look at it.

johanmeijer, 2021-12-25:
@JasperE84 

The docker ledidobe/2.6.1g is updated. Please let me know if it is ok now!

JasperE84, 2021-12-26:
Confirmed!

closed 2021-12-26 by JasperE84

mvank, 2021-12-26:
had same issue error message in docker log:
```ValueError: Invalid value for `every_seconds`, must be a value greater than or equal to `1` ```
with influxdb version: Influx CLI 2.2.1 (git: 31ac783) build_date: 2021-11-09T21:24:22Z

Now running ledidobe/2.6.1g and grott with influxdb v2.2.1 is working!

johanmeijer, 2021-12-26:
Nice!

HarmvZ, 2022-01-13:
I just ran into this same issue when using both the `latest` tags for grott and influxdb images. Pulling 2.6.1g fixed the issue nicely! Perhaps it is time to promote 2.6.1g on docker to the `latest` tag?

johanmeijer, 2022-01-13:
Yes, I will do that soon!

johanmeijer, 2022-01-15:
@HarmvZ versio 2.6.1g is promoted to latest!

(at least for the x86 build!, RPI will follow later)