johanmeijer/grott #87: Format of Energy values

themen71, 2021-12-19:
Hello, I have been using grott since today and got it to work quite easily with my MIN3000 TL-XE and ShineLink-X. There is however one thing that seems a bit strange: the Energy values are read as 0.1/0.2 Wh etc. and then exported to PVOutput as 100/200 Wh etc, which is a bit course, any idea how I can get this a bit more accurate?

Thanks! 
[image]

JasperE84, 2021-12-19:
Could try editing 1f to 2f here. I'm curious if that will bring more detail as the stock app also seems to go by multiples of 100Wh, haven't this tried myself yet.

https://github.com/johanmeijer/grott/blob/c6bf04af3c7dd49427938f0a41f2fa821c5cd1da/grottdata.py#L308

themen71, 2021-12-20:
Nope unfortunately that does not help, I see extra digits in the data but they are all zeros:

[image]

johanmeijer, 2021-12-20:
The growatt inverter sends energy total / today data in 0.1Kwh. That means if the delivered energy is 0.3 Kwh it will send 3. Grott formats this (divide it by 10)  to 0.3 kwh . This is  only to display it in the log (can be customized with personal layouts if you want). 

PVOutput wants the Today energy data in watts hour so Grott multiplies the original data with 100 (in this 3 will be 300) before it is sent to PVOutput. 

Actual delivered power (pvpowerin / pvpowerout) is sent in 0.1 watt. In your data for power out 1954. Grott format this for the log to 195.4  (divide it by 10). 

Pvoutput wants actual power in watts.  So the original data is also divided by 10 before it is sent. PVoutput ignores the decimal values. So 195.4 is sent but PVOutput uses 195.  

Grott always sends the original values to MQTT and Influxdb. 
 
I am reconsidering the data sent to PVoutput while at this moment because of the granularity of the growatt data it shows "strange" average values (it can be explained but ok). If we sent less information (not the Energytoday values) then PVoutput will calculate this and shows much more attractive data (see issue #52).

themen71, 2021-12-21:
@johanmeijer, thanks clear to me that the Energy values are not being sent with more accuracy.

closed 2021-12-21 by themen71