johanmeijer/grott #88: Growatt MOD 7000TL3-X inverter with ShineRFStick-X and a 3-phase Growatt smart meter

ironinhead, 2021-12-28:
Hello,

I have a Growatt MOD 7000TL3-X inverter with ShineRFStick-X and a 3-phase Growatt smart meter (actually Eastron SDM630-MODBUS V3). The meter is connected to the Growatt Raillog.
ShineRFStick-X and Growatt Raillog send data to Growatt ShineLanBox which is connected to the local network and the Internet.
Everything connected as in the attached photo.

[image]

At the moment it is dark and the inverter is turned off, but the meter is sending Raillog data to ShineLanBox.

I installed grott v2.6.1g as a proxy, set invtype = tl3 in grott.ini and I see a problem in the logs:

> 
         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 5279), raddr=('192.168.1.117', 1029)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 22547), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 22547), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x01\x00\x06\x01\x69\x50\x1b\x09\x33\x2c\x42\x20\x46\x4c\x77\x35\x2a\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6e\x34\x74\x78\x68\x54\x71\x73\x76\x5e\x4d\x4d\x69\x4a\x5d\x5b\x52\x4c\x5a
                 \x72\x4a\x43\x45\x56\x40\x5a\x73\x5e\x56\x4f\x4f\x40\x58\x75\x41\x59\x59\x54
                 \x58\x59\x74\x45\x41\x4f\x4d\x59\x43\x70\x5c\x5b\x5b\x4c\x41\x46\x69\x4b\x43
                 \x45\x56\x4c\x5a\x7f\x5e\x5e\x45\x53\x5a\x45\x6b\x40\x5c\x42\x4f\x42\x58\x77
                 \x5c\x56\x5b\x51\x5a\x43\x6b\x42\x41\x4e\x4d\x46\x47\x73\x5c\x5e\x5b\x53\x47
                 \x41\x69\x44\x43\x45\x52\x41\x5a\x75\x5e\x5e\x59\x50\x58\x44\x69\x47\x43\x47
                 \x4f\x4d\x58\x72\x4b\x5e\x59\x53\x58\x42\x76\x4a\x41\x40\x4d\x59\x45\x70\x46
                 \x41\x44\x4d\x44\x5a\x7e\x5e\x5b\x4e\x4f\x4c\x58\x7f\x44\x41\x45\x4d\x45\x47
                 \x69\x44\x43\x45\x4f\x46\x58\x74\x44\x41\x44\x4d\x45\x44\x70\x5c\x5f\x5b\x55
                 \x46\x5a\x77\x5e\x5d\x42\x4f\x42\x58\x74\x40\x41\x46\x4d\x45\x45\x69\x45\x43
                 \x46\x53\x5a\x4c\x6b\x43\x5c\x59\x58\x58\x45\x74\x4a\x41\x43\x50\x58\x40\x77
                 \x44\x41\x4e\x4d\x40\x44\x70\x5c\x57\x5b\x55\x44\x42\x69\x44\x43\x47\x4f\x44
                 \x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59
                 \x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42
                 \x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58
                 \x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51
                 \x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41
                 \x47\x4d\x44\x5a\x77\x5e\x4d\xf7
         - Grott automatic protocol detection
         - Grott data record length 369
         - layout   :  T06501bTL3
         - no matching record layout found, try generic
         - Record layout used :  T06501bTL3
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 000100060169501b4e41433541323830474500000000000000000000000000000000000000000
                 143150c1c13031c013f39392e38322c33382e35382c3237342e342c39382e342c3233362e352c
                 2d33372e382c2d37372e342c2d35322e392c3237382e382c3132322e312c3233352e362c302e3
                 92c302e372c302e392c3233342e312c3233352e362c3233352e322c312e312c302e352c302e39
                 2c3539312e322c3631382e372c2d3137342e332c302e392c34392e382c38362e322c31332e362
                 c322e322c33362e332c3130372e302c34322e302c32352e362c33322e312c31312e372c31322e
                 382c31332e392c3133382e34312c3430362e392c3430372e382c3430362e362c302e302c302e3
                 02c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c30
                 2e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302
                 c302e302c302e302c302e302c302e302c302e302c2280
         - Grott data ack record or data record not defined no processing done

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 5279), raddr=('192.168.1.117', 1029)>
         - Data less then minimum record length, data not processed

        - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 22547), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

johanmeijer, 2021-12-28:
The T06501b is a new type record for me. Do you see other records as well (like T065004)? 

Maybe this a smart meter record record? Let's try to find out!

I will test if it matches any other record layout.

ironinhead, 2021-12-28:
Now the inverter is asleep and I don't see other types of records. I will check and write back tomorrow during the day.

ironinhead, 2021-12-29:
Below are the logs during the day when the inverter is running (and the meter). From what I can read:
T065203XTL3 - no processing
T065204XTL3 -> used T06NNNNXTL3 -> OK (data from inverter)
T065229TL3 - no processing
T06501bTL3 - no processing
T06501bXTL3 - no processing

> 
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 50468), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x7d\x00\x06\x02\x41\x52\x03\x09\x33\x2c\x42\x20\x46\x4c\x77\x35\x2a\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3d\x30\x29\x46\x36\x01\x46\x5f\x42\x35\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61
		 \x74\x74\x3b\x72\x6e\x77\xb8\x8b\x8b\x47\x16\x6f\x77\x46\x64\x74\x46\x63\x1f
		 \x71\x21\x30\x38\x76\x5c\x5f\x77\x3b\x36\x35\x06\x72\x63\x77\x67\x74\x74\x41
		 \x32\x6f\x69\x61\x48\x74\x1d\x72\x35\x77\x61\x26\x33\x0f\x40\x2d\x31\x55\x44
		 \x41\x13\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
		 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x64\xeb\x76\x77\x40\x97
		 \x6f\x7b\x61\x69\x74\x4e\x72\x76\x77\x7e\x74\x77\x4a\x48\x7e\x92\x73\xfa\x60
		 \x17\x7f\x55\x66\x84\x65\x8c\x52\x6a\x62\x4d\x70\x91\x65\xbf\x67\x77\x7a\x5b
		 \x65\x91\x55\xfc\x7b\x27\x61\x32\x74\x75\x72\x29\x77\x53\x74\x46\x47\x40\x6f
		 \x45\x61\x46\x74\x01\x72\x5d\x77\x53\x74\x46\x56\x6c\x6f\x77\x25\x38\x35\x06
		 \x42\x5d\x42\x51\x44\x45\x76\x40\x6e\x46\x61\x74\x74\x47\x61\xf3\x77\x53\x64
		 \xba\x56\x6c\x61\x26\x6f\x75\x74\x53\x72\x6a\x67\x36\x7b\xe4\x60\x76\x48\x73
		 \x47\xfc\x53\x07\x55\x2f\x50\x76\x74\x6a\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
		 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x57\x4f\x72\x6f\x78\x60\x74\x32
		 \x47\x72\x6f\x77\x61\x74\x74\x3a\x72\x96\x27\x37\x54\x54\x67\x45\x5f\x47\x51
		 \x54\x54\x67\x52\x4f\x57\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x7c\x03\x72\xe8\x76\x1f\x72\x6f\x70\xa9\x66\xd8\x56
		 \x6c\x7e\x69\x61\x10\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x40
		 \x7c\xeb\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x25\x26\x3c\x46\x05\x34\x5b\x47\x54\x20\x2c\x1f
		 \x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x74
		 \x74\x47\x72\x6f\x77\x61\x7e\x74\x4d\x72\x65\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x9d\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xc5\x93
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T065203XTL3
	 - no matching record layout found, try generic
	 - Record layout used :  T065203XTL3
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 007d0006024152034e41433541323830474500000000000000000000000000000000000000005
		 24748324246343035540000000000000000000000000000000000000000000000000000020000
		 007c000100d9ffff006400002710000111700640444c312e30005a424141000c0006000006400
		 01e003c005a005a00005247483242463430355400000000000100640000000020202050562049
		 6e76657274657220200000138a020307e5000c001d00090019001f00030d3a11e5128e14500d3
		 a11e511f815180d3a11e511f815180d3a11e5128e145000460032004600320032003200320032
		 0046003200320032111e0000444c41413032353030313132013100000000139c003210ce111e0
		 e510e010014000510570f90270427042688274027402717001e000001e400ff4e2000ff4e2000
		 ff4e2000ff4e20230800000f010046000000000000007d00f9505620202037303030202020202
		 020000000000000000000000000000000000000001374139c0258000007c812ac111e111e0064
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000000000000000032139c0000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0524748324246343035545858585858585858585858585858585858585858000000000000000a
		 000a000a00000000000000000000000000000000000000ef00000000000000000000000000000
		 00000000000b7fc
	 - Grott data ack record or data record not defined no processing done

	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 5279), raddr=('192.168.1.117', 1029)>
[... 343 lines cut ...]
		 \x44\x71\x5c\x58\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59
		 \x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42
		 \x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58
		 \x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51
		 \x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41
		 \x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x11\xfd
	 - Grott automatic protocol detection
	 - Grott data record length 377
	 - layout   :  T06501bXTL3
	 - no matching record layout found, try generic
	 - Record layout used :  T06501bXTL3
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
[... cut ...]
		 2e342c302e362c313637392e362c313732382e332c2d3431322e342c302e392c34392e392c393
		 32e362c31332e362c322e322c33392e312c3131342e392c34352e302c32372e362c33342e352c
		 31322e352c31332e382c31352e302c3134382e36352c3430352e322c3430352e362c3430362e3
		 72c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c30
		 2e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302
		 c302e302c302e302c302e302c302e302c302e302c302e302c302e302c669c
	 - Grott data ack record or data record not defined no processing done

johanmeijer, 2021-12-29:
Ok the inverter data looks fine to me. T065204 is data record. The T065203 is an announce record and not important for ligging I think. 

The T065229 I have not seen before. This might be a smart monitor record. It seems to contain an other device id starting with AZE1.... (do you recognize this?).  

The t06501b are also new for me. Until now I thought the second number (so 50, 51, 52) indicates which device defined at the ShineLink. I am not sure with this one. The record only contains device information of the Shinelink (NAC5A....) and some kind of readable (historical) logging data. 

For the T06501b: 

B107.16,41.39,370.4,60.0,67.6,-113.5,-137.7,-150.3,388.4,150.2,161.0,0.9,0.3,0.3,233.4,234.5,234.6,1.6,0.6,0.7,481.9,655.8,-410.0,0.7,49.9,93.5,13.6,2.2,39.1,114.8,44.9,27.6,34.5,12.5,13.8,15.0,148.56,405.2,406.2,405.3,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0

And the T06501bx: 
G107.21,41.43,367.2,1248.1,45.0,-115.4,-133.4,-149.2,385.1,1253.1,162.4,0.9,0.9,0.2,234.6,233.4,235.2,1.6,5.4,0.6,1679.6,1728.3,-412.4,0.9,49.9,93.6,13.6,2.2,39.1,114.9,45.0,27.6,34.5,12.5,13.8,15.0,148.65,405.2,405.6,406.7,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0

I have to look more at this records if I can detect more information in the header. Normally there is device information and some kind of data/time record. I do recognize this directly

ironinhead, 2021-12-29:
AZE1... is a USB device named ShineRFStick-X plugged into the inverter (RGH2...). It sends data via radio to ShineLanBox (NAC5...)

There are two source of data transfer:
1) Inverter (RGH2 ...) ---(rs485)--- ShineRFStick-X (AZE1...) ---(rf)---> ShineLanBox (NAC5...) ---(lan)--- grott server
2) SDM630-MODBUS V3 ---(rs485)--- Growatt Raillog ---(rf)---> ShineLanBox (NAC5...) ---(lan)--- grott server

(I can see in the admin console of the ShineLanBox that meter has Device Type = SPCT)

[image]

[image]

johanmeijer, 2021-12-29:
Ah OK. The t05229 record seems also to be some kind of announcement record (according some people at the internet). 
So for now I think we can ignore it (but keep it in mind).

You do not see any t06xx20 records? That is what we have found with other sdm-630 users.

ironinhead, 2021-12-29:
There are no entries similar to t06xx20 in the logs. I suppose the difference could be: meter version (V3 vs V2) and
connection via raillog vs eg Modbus to ethernet.

Could I somehow help with data analysis?

johanmeijer, 2021-12-29:
Yes! lets first find the record where the information should be in. 

Are the any other records? Or only the 03, 04, 29, 1b?  

Do you see smart meter information at the Growatt site?

ironinhead, 2021-12-29:
There are no other record types. Most of these are: T065204XTL3 and T06501bTL3

Now, when it is dark and the data is transmitted only by the meter, in the logs I see entries only for T06501bTL3 - I think it is
just the meter.

The meter data affects the appearance of the Growatt website. I can see import and export to the network. Additionally, transmitting RF devices are listed. 
Do you see any entries for the device starting with LAE ...? This is the meter.

[image]

[image]

ironinhead, 2021-12-31:
I am attaching the screen from the ShinePhone application - data from the meter. I hope it will help.

[image]

johanmeijer, 2021-12-31:
Yes this helps a lot! 

If you have an t06501b record and this screen from the same date/time we can try to match it. 

The zero will be a challenge but we probably will find the most important values:)

ironinhead, 2021-12-31:
I'll have to do it again. Need to set the grott to verbose mode and I'll try to capture the same snapshot from the app. I will let you know :)

johanmeijer, 2021-12-31:
It would be nice if you can get 2 sets of record/screen. One at daylight  when the solar panels are generating energy and one in the evening.

ironinhead, 2022-01-03:
Daylight record/screen

> 
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 47449), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x01\x6d\x50\x1b\x09\x33\x2c\x42\x20\x46\x4c\x77\x35\x2a\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6e\x34\x77\x75\x77\x4c\x5e\x73\x76\x22\x45\x43\x77\x5c\x56\x43\x4d\x43\x46
		 \x69\x43\x5e\x5b\x50\x4d\x45\x69\x4b\x43\x44\x55\x5a\x42\x6b\x44\x5e\x42\x4f
		 \x41\x58\x6a\x4a\x57\x59\x54\x58\x59\x76\x41\x5f\x59\x51\x58\x59\x74\x43\x5d
		 \x59\x53\x58\x46\x76\x42\x41\x42\x4d\x45\x47\x71\x5c\x5a\x5b\x57\x4d\x41\x69
		 \x40\x43\x47\x4f\x4d\x58\x77\x5c\x5d\x5b\x51\x5a\x4c\x6b\x40\x5c\x42\x4f\x4d
		 \x58\x75\x41\x5a\x59\x57\x58\x46\x74\x47\x41\x43\x4d\x44\x5a\x7e\x5e\x5f\x59
		 \x55\x58\x46\x69\x4b\x43\x4f\x50\x4d\x5a\x73\x5e\x56\x40\x59\x5a\x40\x6b\x5f
		 \x5a\x42\x53\x5a\x45\x6b\x42\x41\x4f\x4d\x40\x4d\x69\x4b\x43\x46\x54\x41\x5a
		 \x77\x5e\x5e\x42\x4f\x4c\x58\x73\x5c\x5d\x5b\x57\x43\x5a\x7f\x5e\x5e\x4f\x54
		 \x5a\x41\x6b\x44\x56\x59\x58\x58\x40\x73\x5c\x56\x5b\x54\x42\x5a\x77\x5e\x5d
		 \x45\x4f\x40\x58\x75\x41\x41\x44\x4d\x46\x42\x69\x41\x43\x45\x55\x47\x5a\x77
		 \x44\x43\x43\x51\x4c\x5a\x7f\x5e\x5b\x47\x59\x5a\x4d\x6b\x46\x5f\x4f\x4f\x41
		 \x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59
		 \x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42
		 \x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58
		 \x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51
		 \x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41
		 \x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x55\x5c
	 - Grott automatic protocol detection
	 - Grott data record length 373
	 - layout   :  T06501bTL3
	 - no matching record layout found, try generic
	 - Record layout used :  T06501bTL3
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006016d501b4e41433541323830474500000000000000000000000000000000000000000
		 1431601030b2c1c01433137302e39342c37322e31312c3139312e392c33342e362c3631352e35
		 2c2d38382e352c2d3133302e302c2d3331322e322c3231302e352c3133362e352c3639352e322
		 c302e392c302e322c302e382c3233352e392c3233352e362c3233352e342c302e392c302e342c
		 322e392c3831392e342c3937382e342c2d3535322e312c302e382c34392e392c3135352e302c3
		 1352e382c342e322c36372e382c3138352e352c36392e392c34342e392c35362e302c32322e34
		 2c32332e332c32362e332c3234332e30362c3430382e382c3430382e392c3430382e352c302e3
		 02c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c30
		 2e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302
		 c302e302c302e302c302e302c302e302c302e302c302e302c211b
	 - Grott data ack record or data record not defined no processing done

[image]

ironinhead, 2022-01-03:
Evening record/screen

Export data from growatt server: [Electric-meter data evening.xls](https://github.com/johanmeijer/grott/files/7802885/Electric-meter.data.evening.xls)

Grott logs:

> 
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 27273), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x01\x6d\x50\x1b\x09\x33\x2c\x42\x20\x46\x4c\x77\x35\x2a\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6e\x34\x77\x75\x77\x56\x73\x5d\x76\x22\x45\x43\x71\x5c\x58\x44\x4d\x43\x40
		 \x69\x46\x5d\x5b\x56\x4c\x41\x69\x40\x43\x46\x50\x45\x5a\x77\x5e\x5e\x4f\x50
		 \x5a\x41\x6b\x5f\x5d\x47\x53\x5a\x40\x6b\x5f\x59\x4f\x4f\x4d\x58\x6a\x45\x57
		 \x59\x58\x58\x4c\x77\x4a\x41\x4f\x4d\x45\x46\x7e\x5c\x5a\x5b\x50\x4d\x45\x69
		 \x41\x43\x47\x4f\x4d\x58\x77\x5c\x57\x5b\x51\x5a\x4d\x6b\x40\x5c\x44\x4f\x4c
		 \x58\x75\x41\x58\x59\x57\x58\x46\x74\x45\x41\x46\x4d\x46\x5a\x75\x5e\x5f\x59
		 \x54\x58\x44\x69\x4a\x43\x42\x59\x47\x5a\x77\x5e\x59\x44\x57\x5a\x45\x6b\x5f
		 \x5d\x43\x57\x5a\x47\x6b\x42\x41\x4e\x4d\x40\x4d\x69\x4b\x43\x46\x57\x44\x5a
		 \x7f\x5e\x5e\x42\x4f\x4c\x58\x73\x5c\x5d\x5b\x56\x44\x5a\x76\x5e\x5e\x4e\x50
		 \x5a\x43\x6b\x45\x5e\x59\x59\x58\x40\x72\x5c\x59\x5b\x54\x4d\x5a\x75\x5e\x5d
		 \x44\x4f\x44\x58\x75\x46\x41\x47\x4d\x46\x43\x69\x41\x43\x45\x54\x45\x5a\x76
		 \x44\x43\x43\x51\x4c\x5a\x73\x5e\x5b\x46\x50\x5a\x47\x6b\x46\x5f\x4f\x4f\x44
		 \x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59
		 \x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42
		 \x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58
		 \x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51
		 \x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41
		 \x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x1a\x0d
	 - Grott automatic protocol detection
	 - Grott data record length 373
	 - layout   :  T06501bTL3
	 - no matching record layout found, try generic
	 - Record layout used :  T06501bTL3
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006016d501b4e41433541323830474500000000000000000000000000000000000000000
		 14316010311013201433137362e37332c37342e34322c3738352e322c3131312e302c3138312e
		 352c2d3230322e342c2d36382e392c2d37382e392c3830382e382c3132392e352c3139312e332
		 c302e392c302e382c302e392c3233332e382c3233372e362c3233372e312c322e322c302e352c
		 302e382c3538332e302c3633362e312c2d3234362e332c302e392c34392e392c3136302e382c3
		 1352e382c342e322c37302e312c3139312e372c37312e382c34352e362c35392e322c32332e30
		 2c32342e302c32372e332c3235312e31362c3430382e342c3431312e332c3430382e302c302e3
		 02c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c30
		 2e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302
		 c302e302c302e302c302e302c302e302c302e302c302e302c6e4a
	 - Grott data ack record or data record not defined no processing done

ShinePhone app:
[image]

johanmeijer, 2022-01-03:
@ironinhead thanks. This works! I have been able to detect almost all values in the "1b" record. I can not find the positiveActiveTodayEnergy (maybe it is a calculated value) and of course I am not able to map the zero values.

Now if have to change Grott to process the record. The 1b record is totally different formatted than the other records and I have to redesign Grott a little bit. I need some time for that. I will publish it in Version 2.7.x. (probably this or next week)..

ironinhead, 2022-01-03:
@johanmeijer, Great :) Many thanks for your help and time.

Referring to positiveActiveTodayEnergy, maybe it is visible under a different name, such as "meterPositive active today energy" or "PosiActive today power"?

In the list of indicators, I can see that it is rather provided by meter:

[image]

johanmeijer, 2022-01-04:
I do not look at names. The records I see only contains the values. What I did is map the values I see in the record to the values in your spreadsheet/shinephone app. 

I manage to map all values except the positiveActiveTodayEnergy from the shinephone. This value is not in the "1b" record  (I am pretty sure about that )and also not in the spreadsheet (by the way:  how id you manage to retrieve this information, that is a very useful report). I think the spreadsheet contains the same values I see in the 1b record. 

I also do not see today values in the base description of the data record of the SDM630. I do not think the SDM630 provides this information. I even think  the SDM630 does not now the date/time/day. 
Fro more information see: http://support.innon.com/PowerMeters/SDM630-MOD-MID/Manual/SDM630-Modbus_Protocol.pdf. In this document the data record is described that is passed to the growatt server (the input registers) if the SDM630 is directly coupled to the inverter without the Raillog (a T06xx20 record). The raillog formats the data to some kind of readable (comma seperated) fields.   

Where do you have the Energy Trend explanation from? 

The only think I can of is that positiveActiveTodayEnergy is a calculated value. For now I think it is the delta between active_energy in time. So probably from start in the morning and the time of the record is created. If I look at the difference in your report between those values it is: 176,73 - 170.94 = 5.79.  The difference between both positiveActiveTodayEnergy is: 11.9 - 6.1 = 5.8. It might be coincidence and I need more records to really validate this assumption. 

At this moment Grott can not calculate this type of values while it does not have a "memory" (it is stateless). Grott does not know the first value of the active_energy at start of the day. You have to calculated this outside Grott I am afraid. At least you have al values to do it ;).

ironinhead, 2022-01-06:
@johanmeijer 
From what I know, the spreadsheet is available only on the so-called "Share Plant" view under the History Data button. On a normal account, Growatt recently deleted access to historical data. Link to "Share Plant" view can be generated in "Energy-> Plant Management-> Operation Tools->Share Plant". This view does not require logging in. If you provided an email, I can send you such a link to the view of my installation. Github has options for sending private messages?

The Energy Trend explanation is from Growatt www service. There is an "Energy Trend" section and next to the section name there is a tooltip that shows this information.

I will try to provide some more records for analysis.

My plan is to send the data to the Home Assistant and I assume that the appropriate calculation tools are already there.

By the way: do you know how to reset the Energy data in the SDM630 meter? In the document "SDM630 Modbus Protocol" in the chapter "MODBUS Protocol Holding Register Parameters" I can see "Reset Logged Data. Write code to reset data group. Code 1 for Energy.". Do you know how to go about it?

johanmeijer, 2022-01-09:
@ironinhead I just published the 2.7 (beta) branche with support for SDM630 / Raillog. Hopefully you can test it. 

I do not have a SDM630 but I think it (registers) can be reset on the device it self (according the user manual). Maybe it also possible via the device menu (all devices and then settings for the SDM630)?

ironinhead, 2022-01-09:
You're great man, thank you :) Of course I'll check and let you know.

I logged two more records today. It was sunny, electricity was being exported to the grid and a non-zero value appeared in the "Reverse active power (W)" column. I suspect that you did not include this field in the script because this is the first time in my screenshots as there is a non-zero value.
Log in hours:
2022-01-09 13:45:14
2022-01-09 21:31:05

[Electric-meter data - 2022-01-09.xls](https://github.com/johanmeijer/grott/files/7835934/Electric-meter.data.-.2022-01-09.xls)

2022-01-09 13:45:14:
> 
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 21911), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x01\x7c\x50\x1b\x09\x33\x2c\x42\x20\x46\x4c\x77\x35\x2a\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6e\x34\x77\x75\x7d\x4a\x5f\x67\x76\x33\x46\x42\x7f\x5c\x59\x47\x4d\x45\x44
		 \x7e\x5c\x58\x41\x4d\x59\x43\x76\x47\x41\x4e\x4d\x59\x43\x7e\x40\x41\x4e\x4d
		 \x59\x42\x71\x45\x41\x40\x4d\x59\x45\x75\x46\x41\x46\x4d\x59\x45\x76\x47\x41
		 \x43\x4d\x59\x45\x76\x44\x41\x4e\x4d\x47\x46\x72\x5c\x5c\x5b\x59\x44\x47\x69
		 \x4a\x43\x41\x56\x43\x5a\x73\x5e\x42\x47\x4f\x43\x58\x6a\x42\x41\x4e\x4d\x59
		 \x44\x69\x4b\x43\x45\x52\x47\x5a\x7f\x5e\x5d\x44\x54\x5a\x41\x6b\x40\x5c\x43
		 \x4f\x40\x58\x75\x5c\x5d\x5b\x52\x5a\x40\x6b\x40\x41\x4f\x4d\x59\x45\x71\x4a
		 \x5a\x59\x57\x58\x46\x75\x43\x5f\x59\x54\x58\x59\x74\x44\x56\x59\x54\x58\x59
		 \x77\x5c\x56\x5b\x55\x4d\x5a\x7e\x5e\x5d\x43\x50\x5a\x4d\x6b\x40\x59\x59\x57
		 \x58\x42\x69\x4b\x43\x46\x51\x46\x5a\x7f\x5e\x5d\x4e\x51\x5a\x45\x6b\x43\x5f
		 \x4f\x4f\x42\x58\x70\x41\x41\x47\x4d\x4c\x42\x69\x4b\x43\x44\x55\x5a\x44\x6b
		 \x41\x5b\x59\x59\x58\x40\x77\x5c\x57\x5b\x52\x43\x4c\x69\x41\x58\x5b\x55\x44
		 \x42\x69\x41\x43\x43\x51\x43\x5a\x77\x5e\x5b\x47\x54\x5a\x42\x6b\x42\x41\x47
		 \x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c
		 \x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44
		 \x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d
		 \x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f
		 \x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69
		 \x42\x43\x47\x4f\x44\x58\x53\x25
	 - Grott automatic protocol detection
	 - Grott data record length 388
	 - layout   :  T06501bXTL3
	 - no matching record layout found, try generic
	 - Record layout used :  T06501bXTL3
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006017c501b4e41433541323830474500000000000000000000000000000000000000000
		 1431601090d2d0801523236382e36302c3130392e37362c2d3731352e392c2d3739322e392c2d
		 3636372e372c2d3132342e312c2d3131352e342c2d3131362e392c3332352e332c3830332e382
		 c3637372e342c2d302e372c2d302e392c2d302e392c3233332e382c3233352e352c3233342e34
		 2c322e322c332e342c322e382c2d313638352e362c323231302e352c2d3336392e352c2d302e3
		 92c34392e392c3234312e392c32362e362c362e392c3130322e382c3239302e312c3130382e36
		 2c37332e302c38362e392c33342e302c33342e382c34302e382c3337382e33372c3430362e332
		 c3430372e302c3430352e362c302e302c302e302c302e302c302e302c302e302c302e302c302e
		 302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c3
		 02e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e30
		 2c1457
	 - Grott data ack record or data record not defined no processing done

2022-01-09 21:31:05

> 
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 65509), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x01\x6d\x50\x1b\x09\x33\x2c\x42\x20\x46\x4c\x77\x35\x2a\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6e\x34\x77\x75\x7d\x52\x6d\x6b\x76\x22\x46\x43\x73\x5c\x5d\x4f\x4d\x45\x45
		 \x76\x5c\x59\x44\x4d\x40\x44\x72\x5c\x5e\x5b\x54\x4c\x5a\x73\x5e\x5d\x44\x57
		 \x5a\x46\x6b\x5f\x5a\x46\x4f\x40\x58\x6a\x47\x5f\x59\x57\x58\x59\x73\x44\x41
		 \x45\x4d\x40\x44\x7f\x5c\x5c\x5b\x56\x43\x5a\x75\x5e\x5d\x45\x56\x5a\x44\x6b
		 \x42\x41\x4e\x4d\x44\x5a\x70\x5e\x5f\x59\x58\x58\x46\x74\x44\x41\x43\x4d\x46
		 \x47\x70\x5c\x56\x5b\x53\x47\x43\x69\x41\x43\x46\x4f\x43\x58\x77\x5c\x5c\x5b
[... 6 lines cut ...]
		 \x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59
		 \x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42
		 \x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58
		 \x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51
		 \x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41
		 \x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x27\x84
	 - Grott automatic protocol detection
	 - Grott data record length 373
	 - layout   :  T06501bTL3
	 - no matching record layout found, try generic
	 - Record layout used :  T06501bTL3
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006016d501b4e41433541323830474500000000000000000000000000000000000000000
		 143160109151f0401433237342e32382c3131312e36332c3430352e312c35382e342c3233362e
		 322c2d35312e342c2d35302e362c2d34362e322c3430382e332c37372e322c3232372e302c302
[... cut ...]
		 392c3637332e302c3638392e382c2d3134352e382c302e392c34392e392c3234362e392c32372
		 e332c362e392c3130342e362c3239362e312c3131312e312c37342e352c38382e362c33342e36
		 2c33352e342c34312e352c3338352e39312c3431302e382c3431312e362c3431302e332c302e3
		 02c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c30
		 2e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302
		 c302e302c302e302c302e302c302e302c302e302c302e302c53c3
	 - Grott data ack record or data record not defined no processing done

ironinhead, 2022-01-09:
@johanmeijer, It works. Really great job :)

I compared the records with the growatt server dump. The only thing missing: Reverse active power (W). 
(also Positive today power (kWh) and ReverActive today power (kWh) but it's already explained that it's not in the data)

Also, a few minor typos:
"reactie_energy" -> should be "reactive_energy"
"apperent *" -> should be "apparent *"

[Electric-meter data - 2022-01-09_2200.xls](https://github.com/johanmeijer/grott/files/7836042/Electric-meter.data.-.2022-01-09_2200.xls)

> 
         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 52892), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x01\x00\x06\x01\x73\x50\x1b\x09\x33\x2c\x42\x20\x46\x4c\x77\x35\x2a\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6e\x34\x77\x75\x7d\x51\x63\x7c\x76\x28\x46\x43\x73\x5c\x57\x47\x4d\x45\x45
                 \x76\x5c\x58\x42\x4d\x47\x44\x75\x5c\x5a\x5b\x50\x46\x44\x75\x5c\x57\x5b\x50
                 \x4d\x43\x69\x47\x43\x5a\x57\x46\x5a\x75\x5e\x42\x44\x54\x5a\x40\x6b\x5f\x5b
                 \x4f\x4f\x4c\x58\x74\x42\x57\x59\x57\x58\x45\x75\x42\x5b\x59\x52\x58\x46\x77
                 \x42\x41\x46\x4d\x44\x5a\x7e\x5e\x5f\x59\x58\x58\x44\x69\x4b\x43\x45\x52\x41
                 \x5a\x74\x5e\x5d\x44\x55\x5a\x4c\x6b\x40\x5c\x42\x4f\x4d\x58\x76\x5c\x5c\x5b
                 \x54\x5a\x45\x6b\x42\x41\x4f\x4d\x45\x43\x76\x41\x41\x46\x4d\x45\x43\x76\x43
                 \x41\x41\x4d\x59\x45\x73\x45\x41\x46\x4d\x44\x5a\x7e\x5e\x5b\x4e\x4f\x4d\x58
                 \x75\x46\x58\x59\x55\x58\x46\x70\x5c\x5c\x5b\x57\x5a\x4d\x6b\x43\x5f\x43\x4f
                 \x43\x58\x75\x4b\x59\x59\x57\x58\x45\x76\x43\x41\x44\x4d\x43\x40\x69\x44\x43
                 \x4f\x59\x5a\x43\x6b\x41\x5b\x59\x56\x58\x47\x72\x5c\x5b\x5b\x55\x45\x5a\x72
                 \x5e\x5c\x4f\x57\x5a\x41\x71\x5e\x5b\x47\x57\x5a\x4d\x6b\x46\x5f\x40\x4f\x43
                 \x58\x73\x42\x58\x59\x58\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b
                 \x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42
                 \x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a
                 \x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51
                 \x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43
                 \x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x5b\xf8
         - Grott automatic protocol detection
         - Grott data record length 379
         - layout   :  T06501b
         - Record layout used :  T06501b
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 000100060173501b4e41433541323830474500000000000000000000000000000000000000000
                 14316010916111301493237342e38302c3131312e37352c3330322e352c313230322e382c3139
                 372e352c2d36322e322c2d33352e342c2d34382e382c3330382e362c313230342e332c3230302
                 e312c302e392c302e392c302e392c3233352e332c3233342e382c3233352e392c312e332c352e
                 312c302e382c313731332e312c313731312e362c2d3134372e312c302e392c34392e392c32343
                 72e342c32372e332c362e392c3130342e372c3239362e362c3131312e332c37342e362c38382e
                 372c33342e372c33352e342c34312e352c3338362e35362c3430362e392c3430372e372c34303
                 72e392c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e30
                 2c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c302
                 e302c302e302c302e302c302e302c302e302c302e302c302e302c302e302c2f8c
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06501b

         - Grott server date/time used
         - Grott values retrieved:
                 -  datalogserial        :  NAC5******
                 -  active_energy        :  274.80
                 -  reactie_energy       :  111.75
                 -  activePowerL1        :  302.5
                 -  activePowerL2        :  1202.8
                 -  activePowerL3        :  197.5
                 -  reactivePowerL1      :  -62.2
                 -  reactivePowerL2      :  -35.4
                 -  reactivePowerL3      :  -48.8
                 -  apperentPowerL1      :  308.6
                 -  apperentPowerL2      :  1204.3
[... 162 lines cut ...]
                         "values": {"datalogserial": "NAC5******", "active_energy": "274.86",
                         "reactie_energy": "111.76", "activePowerL1": "283.3", "activePowerL2":
                         "60.5", "activePowerL3": "110.9", "reactivePowerL1": "-48.0",
                         "reactivePowerL2": "-51.2", "reactivePowerL3": "-77.4", "apperentPowerL1":
                         "287.4", "apperentPowerL2": "79.5", "apperentPowerL3": "145.2",
                         "powerFactorL1": "0.9", "powerFactorL2": "0.7", "powerFactorL3": "0.8",
                         "voltageL1": "235.2", "voltageL2": "236.1", "voltageL3": "235.9",
                         "currentL1": "1.2", "currentL2": "0.3", "currentL3": "0.5", "active_power":
[... cut ...]
                         "l3Voltage1": "408.1", "device": "SDM630"}}
         - MQTT message message sent
         - Grott Send data to PVOutput disabled
         - Grott Send data to Influx disabled
         - Grott extension processing disabled

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.3', 5279), raddr=('192.168.1.117', 1029)>
         - Data less then minimum record length, data not processed

johanmeijer, 2022-01-10:
@ironinhead  Thanks for the feedback. I fixed the typo's (you have to download grottconf.py again). 

It appears that the Reverse active power (W) is passed in the same field as the active power. If this value is negative it is Reverse active power. If it is positive it is active power. 

Do you think it is necessary to add this field? I am not sure if this will help you with Home Assistent (is it possible to define two fields voor Power?). 

I you think we need it I have to do some rethinking. At this moment Grott does not do any processing based on field names while it is very easy to change these names in an own defined layout file. See also: https://github.com/johanmeijer/grott/wiki/Grott-advanced-(customize-behaviour) This documentation is not yet update for the SDM630 processing but you get an idea how it works. 

But we will find some kind of solution if needed! 

If you specify includeall = True in the [Generic] section of the grott.ini. You will see all fields/values that are in the record. The fields that I was not able to identify yet (because they are 0) has been named posxx (where xx is the position in the record). 
So if I missed something we can always detect which field we have to use/rename. 
Be aware: Using the includeall parameter might also affect the processing of the inverter record (give more information).

johanmeijer, 2022-01-15:
@ironinhead 

I changed the processing a bit and now I am able to make a distinction between positive  and negative power. 

I added fields: 
power (contains  power value both for negative and positive)
reverse_active_power  (contains  power value if negative)

I changed field: 
active_power  (contains  power value if positive)

Hopefully this will work for you.

ironinhead, 2022-01-16:
@johanmeijer 
Thank you, now it will definitely be more in line with what I can see in the Growatt report.
I am at the stage of creating sensors in Home Assistant. The two most important sensors for energy calculations are posiActivePower and reverActivePower. HA handles automatically the calculation of the daily import and export based on these fields.

I had a problem with the device and time fields (last update date) so that for the meter and the inverter they are in separate fields and do not overwrite each other. After many unsuccessful attempts with "if and else" instructions in HA, I changed the mqtt publication topic for the meter to "energy / meter" in grottdata.py (inverter to "energy / growatt").

The change is not very elegant and quick, but sufficient for my needs :)

`if conf.nomqtt != True:

            try:

                mqtttopic = conf.mqtttopic

                if (jsonobj['device'] == "SDM630") :

                    mqtttopic = "energy/meter"

                publish.single(mqtttopic, payload=jsonmsg, qos=0, retain=False, hostname=conf.mqttip,port=conf.mqttport, client_id=conf.inverterid, keepalive=60, auth=conf.pubauth)`

So far it looks like this for me (working screen - sorry not everything is in english):

[image]

johanmeijer, 2022-01-16:
Looking good. 

I will look for a nice way to define a separate MQTT  topic for the energy meter. I can understand that should be different.   
I thought it was enough the be able to specify the device naam in the json message but if that is not the case it should be resolved differently.

johanmeijer, 2022-01-18:
I made a small change. It is now possible to specify separate topic for the smart meter. To enable this you have to specify a new parameter in the [MQTT] section of the .ini: mtopic = True (or in docker if 2.7.0 image is created -e gmqqtmtopic = "True" ). 

If the mtopic = True is specified by default the mqtttopic will be: energy/meter. 

It is possible to change it in the .ini by mtopicname = topicname (or -e gmqttmtopicname = topicname)

johanmeijer, 2022-01-20:
Sorry I comment out the mqtt publish for testing.I will repair this asap.

johanmeijer, 2022-01-21:
Fixed

ironinhead, 2022-01-21:
@johanmeijer, sorry for not resposing. A few private matters popped up. I'll try to check out tonight.

ironinhead, 2022-01-21:
Reading the data for the meter fully works. Tomorrow during the day I will check the regression for the inverter :)

Thanks again

ironinhead, 2022-01-22:
@johanmeijer, Little question: what is the **totworktime** parameter for?

johanmeijer, 2022-01-22:
Totworktime is de total time the inverter has been active. Internally growatt (and in the mqtt message) is based on 0.5 sec. For the grott display I format it in hours (divide by 7200). 

uwBatVolt_DSP I really do not know what that is in de documentation BatVolt(DSP) in 0.1 V .  I think that is the voltage from the battery if connected (before or after it is processed by the inverter DSP). I do not have a battery  I really haven't seen values for that one. 

pbusvolt and nbusvolt are inside inverter  bus (DC) voltages (0.1v) . I think we really have to dig into the inverter schematics to know what that is. 

I know this doesn't really help with explaining, but I have to deepdive in the inverter electronics to know more about that. Maybe there are others who already know more about that.

johanmeijer, 2022-04-28:
@ironinhead Can we close this issue?

ironinhead, 2022-04-29:
@johanmeijer, yes, thank a lot again

closed 2022-04-29 by ironinhead