johanmeijer/grott #89: Growatt server is down

lspaula, 2021-12-30:
The Growatt server has been down for about 2h right now and no data is saved locally when under this condition. I am getting the following messages from grott status:

```
Dec 30 23:17:26 openhabian grott[5636]: [Errno 111] Connection refused
Dec 30 23:17:26 openhabian grott[5636]:          - Can't establish connection with remote server.
Dec 30 23:17:26 openhabian grott[5636]:          - Closing connection with client side ('192.168.0.147', 39589)
```
Is this a normal behaviour? Is there a way of accessing the data locally?

johanmeijer, 2021-12-30:
I also noticed that the growatt server is down. The website does not work either.

If there is no communication with the Growatt server the inverter is not able to send data and Grott can not process it.

However most of the inverters store/buffer the data locally and send it later if the connection is established. Grott will then also process this data (if the date/time is included and that that does not always seems to be the case with some inverters ).

lspaula, 2021-12-30:
Wouldn´t that be a way of, instead of intercepting data that is sent, creating a local server to feed from and ack on messages sent by the shineWiFi devices, keeping all data traffic local?

johanmeijer, 2021-12-30:
It is on the todo list. But I have to reverse engineer the protocol (it is not only ack the messages). I have not completed that totally. 

To be honest maintaining Grott (and the docker containers), interpreter the data of all types of (new) inverters, data loggers and other devices like smart meters, is rather time consuming. I have not found time for creating the server part.

Maybe at one day there will be Grott V3 with this functionality. 

By the way this is the first time in the 3 years I have a growatt inverter I noticed that the Growatt server was down for a longer time.

lspaula, 2021-12-31:
Johan,

I perfectly understand. I know how complicated is to spare some time to a project like this. I might say that I am a great fan of your work and have been using Grott for the last 6 months. I really love to see the data on Grafana and its interaction with my home automation on Openhab. Thank you very much for taking your time on this development.

Growatt services are back online, by the way. 

Happy new year from south Brazil!

Regards,

Luciano

johanmeijer, 2021-12-31:
Thanks and happy New Year from the Netherlands!

kooibosmania, 2022-01-02:
sorry tp for breaking in but i have a question that doesnt need a issue ticket.
great work on this, i got it working. however i got spf 5000ES off grid and i get some values but not the battery side of it. i put inv=spf in grott ini. are u still working on this ?
i understand this is lots of work and apreciate it
regards and happy new year

johanmeijer, 2022-01-02:
This is the information Grott retrieves at this moment for the SPF types. 

[image]

I think there might be more info in the data record for the BMS system but to be honest that is not very well documented (especially the measuring units) in the documents I have (they refer to an other document I do not have). I can try to make the data visibel, maybe we can recognize the information.

kooibosmania, 2022-01-02:
i got it after fiddling around with the ini file somehow the change i made did not stick. then with mqtt explorer i got the same list u have so that is right. after i modified the sensors yaml to accommodate the new naming scheme i got all sensors i need. great work on this again thank u.
[image]

kooibosmania, 2022-01-02:
for those with similar spf batterie driven inverter here is the slightly modified yaml 

[sensors_growatt_nl.zip](https://github.com/johanmeijer/grott/files/7799661/sensors_growatt_nl.zip)

johanmeijer, 2022-01-02:
I am glad it works. 

Thank you for the yaml. Is it ok when I publish in the examples directory

kooibosmania, 2022-01-03:
yes ofcourse orriginaly yours

johanmeijer, 2022-02-16:
@lspaula I have created a first itteration of grottserver that can act as a local replacement for the growatt server.

See discussions #98 for more details. 

I am curious to hear your experiences with it.

lspaula, 2022-02-16:
This is such great news. Thank you very much for this huge effort. I'll check it as soon as I can.

closed 2023-08-13 by johanmeijer