johanmeijer/grott #90: Growatt MOD 6000TL3-X incorrect totals

zwartevogel, 2022-01-11:
I have setup grott and the ShineWifi-X USB device on my MOD 6000TL3-X inverter. Everything is working and data is coming into MQTT and HomeAssistant. Most values are correct (actuals, voltage, ...) but the 'pvenergytoday' and 'pvenergytotal' are filled with values which are not correct.
My MQTT log:
energy/growatt {"device": "xxxxxx", "time": "2022-01-10T15:09:14", "buffered": "no", "values": {"pvstatus": 1, "pvpowerin": 26399, "pv1voltage": 2697, "pv1current": 51, "pv1watt": 13754, "pv2voltage": 2749, "pv2current": 46, "pv2watt": 12645, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 1625756552, "totworktime": 2293760, "pvenergytoday": 530321366, "pvenergytotal": 267259793, "epvtotal": 141, "epv1today": 606, "epv1total": 668346, "epv2today": 76, "epv2total": 310, "pvtemperature": 675, "pvipmtemperature": 467}}

The actual values for that moment are:
- Today: 14 kwh
- total: 60 kwh

It looks like the 'epvtotal' should be in the 'pvenergytoday' and 'epv1today' should be linked to 'pvenergytotal'. 
Or am I doing something wrong?

johanmeijer, 2022-01-11:
@zwartevogel have you specified invtype=tl3 in grott.ini [Generic] section ?

zwartevogel, 2022-01-11:
I added this to my ini file and started docker with "-v /home/pi/growatt/growat.ini:/app/grott.ini". 
The result:
energy/growatt {"device": "xxxxxxxx", "time": "2022-01-11T11:22:04", "buffered": "no", "values": {"pvstatus": 1, "pvpowerin": 7752, "pv1voltage": 2751, "pv1current": 14, "pv1watt": 3851, "pv2voltage": 2787, "pv2current": 14, "pv2watt": 3901, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 490804105, "totworktime": 786432, "pvenergytoday": 179900351, "pvenergytotal": 265424755, "epvtotal": 16, "epv1today": 634, "epv1total": 704283, "epv2today": 8, "epv2total": 325, "pvtemperature": 575, "pvipmtemperature": 318}}

johanmeijer, 2022-01-11:
Sorry I did not realise you uses docker

Which docker image do you use? You need the use the beta / 2.6.1g image

You can specify allmost all parameters at the docker run command as environmental putting a good infront (for this ginvtype="tl3").

Debug info can be seen with  the docker logs <container name> command.

zwartevogel, 2022-01-11:
Ah, that was the problem. I ran latest instead of beta. Thank you!
For future searchers this command solved my problem 👍 :
-  docker run -d -p 5279:5279 --restart on-failure -e ginvtype="tl3" -v /home/pi/growatt/growat.ini:/app/grott.ini ledidobe/grottrpi:beta

closed 2022-01-11 by zwartevogel

johanmeijer, 2022-01-11:
I am glad it is solved. I will try to document it better which docker container to use. 

I also will promote the Beta version to latest soon.