johanmeijer/grott #99: MOD5000TL3-X epvtotal counter data with decreased value results in hass energy dashboard spike

JasperE84, 2022-02-20:
Recently I've seen this happen two or three times on my config:

```
jasper@ubuntu01:~/hass$ egrep -i '(vtotal).+:  [2][1][5].[6-7]' ~/grott.log
                 -  epvtotal             :  215.6
                 -  epvtotal             :  215.6
                 -  epvtotal             :  215.6
                 -  epvtotal             :  215.6
                 -  epvtotal             :  215.6
                 -  epvtotal             :  215.6
                 -  epvtotal             :  215.6
                 -  epvtotal             :  215.7 <---- .1 up
                 -  epvtotal             :  215.6 <---- .1 down, causes reset in HASS and a 215.6 kWh production in energy dashboard
                 -  epvtotal             :  215.6
                 -  epvtotal             :  215.6
```

This results in an home assistant spike in energy dashboard:
[image]

Any ideas how to prevent this? 

This is my home assistant server config:

```
  customize_glob:
    sensor.growatt_*_energ*:
      last_reset: '1970-01-01T00:00:00+00:00'
      device_class: energy
      state_class: total_increasing

```

JasperE84, 2022-02-20:
By the way I reset those spikes by using this SQL bit, not properly tested yet will let you guys know if this breaks my database :)

The SQL code resets the sum value to the measured states which should effectively eliminate the resets.
```
MariaDB [ha_db]> select * from statistics_meta;
+----+-------------------------------------------+----------+---------------------+----------+---------+------+
| id | statistic_id                              | source   | unit_of_measurement | has_mean | has_sum | name |
+----+-------------------------------------------+----------+---------------------+----------+---------+------+
| 20 | sensor.growatt_opgewekte_energie_totaal   | recorder | kWh                 |        1 |       0 | NULL |

update statistics_short_term set `sum` = `state` where metadata_id = 20;
update statistics set `sum` = `state` where metadata_id = 20;
```

gabrielpc1190, 2022-02-20:
I have seen those spikes too, and it totally messed up my statistics. Didn't know it was caused by a reverse on the counter.
How to execute that SQL query? What it does? I'm not SQL trained but I can try to apply this

JasperE84, 2022-02-20:
First off, I believe that my SQL queries will only work if you have the sensor config in ha configuration.yaml set to 'total_increasing'. If set to 'measurement' other columns (mean/min/max i believe) will also be set and the required SQL queries will probably be different. 

In my home assistant I'm using a MySQL backend. Yours is likely to use the default sqlite backend if you haven't configured MySQL/MariaDB in configuration.yaml. 

Whichever database backend, the SQL statements will look the same. Just the SQL client will be different. I used the mysql command line interface client. 

Mind that it's rather easy to mess the database up. Make sure to backup your data.

For the update query to work your data must look like this:

```
MariaDB [ha_db]> select  * from statistics_short_term where metadata_id = 20 limit 10;
+--------+----------------------------+----------------------------+------+------+------+------------+-------+-------+-------------+
| id     | created                    | start                      | mean | min  | max  | last_reset | state | sum   | metadata_id |
+--------+----------------------------+----------------------------+------+------+------+------------+-------+-------+-------------+
| 631366 | 2022-02-10 03:20:10.260390 | 2022-02-10 03:15:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631385 | 2022-02-10 03:25:10.281667 | 2022-02-10 03:20:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631404 | 2022-02-10 03:30:10.258879 | 2022-02-10 03:25:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631423 | 2022-02-10 03:35:10.281366 | 2022-02-10 03:30:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631442 | 2022-02-10 03:40:10.294831 | 2022-02-10 03:35:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631461 | 2022-02-10 03:45:10.373224 | 2022-02-10 03:40:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631480 | 2022-02-10 03:50:10.269229 | 2022-02-10 03:45:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631499 | 2022-02-10 03:55:10.280430 | 2022-02-10 03:50:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631518 | 2022-02-10 04:00:10.725075 | 2022-02-10 03:55:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
| 631537 | 2022-02-10 04:05:10.284439 | 2022-02-10 04:00:00.000000 | NULL | NULL | NULL | NULL       | 148.4 | 148.4 |          20 |
+--------+----------------------------+----------------------------+------+------+------+------------+-------+-------+-------------+
10 rows in set (0.001 sec)

MariaDB [ha_db]> select  * from statistics where metadata_id = 20 limit 10;
+-------+----------------------------+-------------+----------------------------+------+------+------+------------+-------+------+
| id    | created                    | metadata_id | start                      | mean | min  | max  | last_reset | state | sum  |
+-------+----------------------------+-------------+----------------------------+------+------+------+------------+-------+------+
| 49984 | 2021-12-12 08:00:10.332973 |          20 | 2021-12-12 07:00:00.000000 | NULL | NULL | NULL | NULL       |   1.6 |  1.6 |
| 50003 | 2021-12-12 09:00:10.201900 |          20 | 2021-12-12 08:00:00.000000 | NULL | NULL | NULL | NULL       |   1.6 |  1.6 |
| 50022 | 2021-12-12 10:00:10.193075 |          20 | 2021-12-12 09:00:00.000000 | NULL | NULL | NULL | NULL       |   1.7 |  1.7 |
| 50042 | 2021-12-12 11:00:10.509859 |          20 | 2021-12-12 10:00:00.000000 | NULL | NULL | NULL | NULL       |   1.8 |  1.8 |
| 50062 | 2021-12-12 12:00:10.269044 |          20 | 2021-12-12 11:00:00.000000 | NULL | NULL | NULL | NULL       |   1.9 |  1.9 |
| 50082 | 2021-12-12 13:00:10.219029 |          20 | 2021-12-12 12:00:00.000000 | NULL | NULL | NULL | NULL       |   2.1 |  2.1 |
| 50102 | 2021-12-12 14:00:10.164972 |          20 | 2021-12-12 13:00:00.000000 | NULL | NULL | NULL | NULL       |   2.1 |  2.1 |
| 50122 | 2021-12-12 15:00:10.271658 |          20 | 2021-12-12 14:00:00.000000 | NULL | NULL | NULL | NULL       |   2.2 |  2.2 |
| 50142 | 2021-12-12 16:00:10.210584 |          20 | 2021-12-12 15:00:00.000000 | NULL | NULL | NULL | NULL       |   2.2 |  2.2 |
| 50162 | 2021-12-12 17:00:10.188669 |          20 | 2021-12-12 16:00:00.000000 | NULL | NULL | NULL | NULL       |   2.2 |  2.2 |
+-------+----------------------------+-------------+----------------------------+------+------+------+------------+-------+------+
```

What I saw is that the  `sum` column was increased with the full `state` value after each reverse. That's how I deducted that a lower value of a measurement triggers a counter reset in home assistant. (I've seen this mechanism before in graphing applications for 32bit byte counters on network interfaces)

johanmeijer, 2022-02-21:
I Can not help you with home assistent issue I have that still not running (to much busy with new Grott functionality). 

It is strange that an er is an up and down in the epvtotal. Can you see if the records are in sync (by date/time). Are there any buffered (older) records sent in between?

JasperE84, 2022-02-22:
Hi Johan,

Here's the relevant log:
The 215.7 (high) reading has a 'buffered: no' flag.
The 215.6 (low) reading before and after that have a 'buffered: yes' flag

What does buffered mean here, buffered on the ShineLink firmware because growatt server couldn't be reached maybe?

```
         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.16.4', 33734), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\xce\x00\x06\x02\x41\x51\x50\x09\x33\x2c\x42\x20\x40\x41\x77\x21\x5f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x3d\x32\x29\x45\x36\x76\x45\x5f\x47\x25\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x77\x76\x60\x4d\x45\x63\x75\x61
                 \x74\x74\x3b\x72\x6e\x77\x61\x60\x5b\x4d\x55\x6f\x7d\x61\x74\x7e\x60\x7a\x97
                 \x77\x6b\x74\x74\x4f\x8a\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x64
                 \x83\x67\xfe\x4e\x52\x6f\x7e\x61\x74\x7c\x71\x7b\x59\x77\x69\x74\x74\x40\x2c
                 \x66\x4f\x61\x7d\x74\x47\x7a\x23\x78\x81\x7b\x94\x48\x85\x6f\x77\x61\x70\x74
                 \x47\x7a\xc1\x77\x5d\xb4\x3c\x47\x72\x6f\x75\x61\x74\x71\x91\x72\x6f\x77\x60
                 \x74\x74\x45\xe4\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x7f\x0d\x76
                 \x2e\x46\x35\x6e\x37\x61\x74\x74\x47\x7e\x28\x7b\x20\x3a\x54\x47\x72\x6f\x77
                 \xb9\x8a\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x68\x74\x6a\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x3a\x72\x96\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x70\x44\x3e\xd7\x74\x74\x47\x72\x6f\x61\x61\x76\x74\x54\x72\x79\x77\x63
                 \x74\x60\x47\x72\x6f\x77\x61\x74\x6e\x41\x72\x57\x77\x61\x74\x4c\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x67\x9b\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x14\x28
         - Grott automatic protocol detection
         - Grott data record length 585
         - layout   :  T065150XTL3
         - no matching record layout found, try generic
         - Record layout used :  T06NNNNXTL3
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 00ce0006024151504e41433541343530533000000000000000000000000000000000000000005
                 245483142313730304400000000000000000000000000000000000000001602140a370c020000
                 007c00010000142f0a27000a00000a2708f8000a000008f800000000000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000000000000013e213
                 8a0920000900000836093600080000075e093800090000084c0fe00fe00ff700000004000008a
                 e003cc04800000002000005d60000000100000296000000000000000000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000000086c025a01470
                 140000000000c470c414e2000000000d8fe000000000000000000000000000000000000000900
                 1e0000000000000000000000000000000000000000007d00f9000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000000000022b49b600
                 0000000016000200130016000200140000000000001a060038000000380000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000013ef0000000000000000000000000000000000000000000000000000000000000
                 000000000006647
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
[... 345 lines cut ...]
                 -  {'d': '20220220', 't': '10:55', 'v1': 400, 'v2': 509.0, 'v6': 233.6}
         - Grott PVOutput response:
[... cut ...]
                         'pvenergytoday': 4, 'pvenergytotal': 2222, 'totworktime': 3981384,
                         'epv1today': 2, 'epv1total': 1494, 'epv2today': 1, 'epv2total': 662,
                         'epvtotal': 2156, 'pvtemperature': 602, 'pvipmtemperature': 327,
                         'pvboottemperature': 320, 'uwBatVolt_DSP': 0, 'pbusvolt': 3143, 'nbusvolt':
                         3137}}]
         - Grott write to influxdb v2
         - Grott extension processing disabled

```
And then follows another messages with buffered: yes and value 215.6

johanmeijer, 2022-02-22:
Ah that explains a lot. The inverter (datalogger) buffered the record if it does not have a connection with the growatt server.

It will sent them later as soon the connection is established (is then a "50" in stead of "04" data record). Grott flagged this records as buffered. Important with processing of these records is taking care of the timestamp in the mqtt record and not use the time of send. It took me some time to get this right with the influxdb interface. I can understand it will be difficult for HA processing as well.

You can disable the processing of buffered records bij specifying the sendbuf = False in the .ini (gsendbuf = "False"for docker). 
You might miss some historical records though.

cspiby, 2022-02-24:
I worked around this by using generated energy total rather than the daily.

JasperE84, 2022-03-27:
> You can disable the processing of buffered records bij specifying the sendbuf = False in the .ini (gsendbuf = "False"for docker). You might miss some historical records though.
Thanks Johan! I changed the conf to do that now.

closed 2022-03-27 by JasperE84

JasperE84, 2022-04-10:
Additional note, a new home assistant update provides a developer tool to adjust long time stats. Previously this was only possible by direct SQL queries:

`https://www.home-assistant.io/blog/2022/04/06/release-20224/#adjusting-long-term-statistics`