johanmeijer/grott #101: Grott not receiving data anymore

dannybeeckman, 2022-02-23:
Hi fellow solar wizards.

Since a few weeks I can't receive any data anymore from my Growatt monitor stick.
My setup is using a local DNS sending the monitor stick a spoof IP 192.168.1.114 for server.growatt.com.
But since the issue the stick keeps on requesting for the A DNS record for server.growatt.com every second.
I already tried to change the IP to a public IP but that didnt fix the issue.
It seems like a firmware update might have added some kind of check to ensure it is communicating with Growatt.

johanmeijer, 2022-02-23:
Grott needs the data being transmitted to server.growatt.com.

Normally you route the data from the growatt inverter (wifi stick) to the system grott is running on and grott send the data after processing to server.growatt.com (this in proxy mode).

A reset of the wifi stick will probably cause a fallback in destination to server.growatt.com(if you changed it).

It is normal that it search for server.growatt.com. in your cause your dns should change that to the grott ip (and port). While in grott for the growatt server the ip address is specified your dns will not see it and does nothing. 

I am not sure why the stick request  is know not being rerouted to grott. Doe you see any message in the grott log?

dannybeeckman, 2022-02-25:
The grott proxy still has access to the growatt cloud as only the shinelan wifi stick now thinks it's on 192.168.... because of the dns spoofing. 
There is nothing configured on the wifi stick itself.

> It is normal that it search for server.growatt.com. in your cause your dns should change that to the grott ip (and port). While in grott for the growatt server the ip address is specified your dns will not see it and does nothing.
This is how it's configured. But the data doesn't even arrive anymore on the grott proxy.

There is no log on the grott proxy since there is no data arriving, only the startup log.

So for some reason it doesnt accept the grott anymore as a proxy. Everytime I provide the local Ip 192.18.1.X for grott instead of 47.91.67.66 it keeps on polling for the IP in the DNS. Once I leave it to 47.91.67.66 it is OK.

There seems to be a feature added with a firmware update to verify some thing but I'm not sure what.
I'll try to sniff the stick whenever I have some spare time.

johanmeijer, 2022-03-21:
@dannybeeckman do you still have this error?

dannybeeckman, 2022-03-28:
As far as I know this error still exists. But since configuring the growatt container ip/port in the stick manually "fixed" the issue and resolving the base issue would be on growatt side, I propose to close the issue.

closed 2022-03-28 by dannybeeckman

johanmeijer, 2022-03-28:
Ok. thank you for the information.