johanmeijer/grott #103: Allow non IP (hostname) as value for environment variable gifip

bkleef, 2022-03-02:
When I set the environment variable `gifip` to `influxdb`  in version 2.6.1g it shows the error:

> Grott InfluxDB server IP address env invalid

In my Docker Compose the hostname `influxdb` [is available](https://docs.docker.com/compose/networking/) as this is another service in the same file:

```
version: "3.4"
services:
  grott:
    container_name: grott
...
  influxdb:
    container_name: influxdb
...
```

Please allow hostnames in addition to IPs. InfluxDB client supports them as well [like `localhost`](https://influxdb-client.readthedocs.io/en/latest/api.html#influxdbclient).

johanmeijer, 2022-03-02:
Thanks for the feedback. I check on valid ip address indeed.

Although it might introduce some unexpectec configuration errors but I do understand this is an improvement. I will implement it in one of the next releases.

mvysny, 2023-08-03:
Sorry for a ping kind of message, but any progress on this please? This basically blocks me from running grott+influxdb2 in docker-compose.

mvysny, 2023-08-03:
Workaround is to patch `grottconf.py` and comment out line 500:
```
#ipaddress.ip_address(os.getenv('gifip'))
```
Also note that the hostname must contain at least one dot character, otherwise Python influxdb2 client will fail with a non-sensical error message `No host specified.`.

Afterwards, you can include the patched `grottconf.py` file in your `docker-compose.yaml` file by mounting it as a volume.

I'm attaching the `docker-compose.yaml` file which works for me:
```yaml
version: '2'

# Grott:
#   Homepage: https://github.com/johanmeijer/grott
#   Docker : https://hub.docker.com/r/ledidobe/grott
#   Influx support: https://github.com/johanmeijer/grott/wiki/InfluxDB-Support
#   Grott Configuration: https://github.com/johanmeijer/grott/wiki/Grott-Configuration

# InfluxDB2:
#   http://localhost:8086
#   user: admin, pass: 9GZf)=TgNud
#   org = my_org
#   bucket = grott
#   Operator API token: wSHdOJFvr9EhWLl4O3AGKtaLQbqiVyKJOKI-UbD46FPrq0du3UU4RaoVLWicY-bwuZmmk7qaJqhO2b_bX4v6pg==
#   Automated setup: see https://hub.docker.com/_/influxdb "Automated Setup"

# Docker restart policies: https://blog.knoldus.com/restart-policies-in-a-docker-container/

# BLOCKER: https://github.com/johanmeijer/grott/issues/103

services:
  grott:
    image: ledidobe/grott:2.8.2
    environment:
      gnomqtt: "true"
      ginflux: "true"
      ginflux2: "true"
      gifip: "influxdb.dc"
      gifport: 8086
      giftoken: "xyz"
      giforg: "my_org"
      gifbucket: "grott"
    ports:
      - "5279:5279"
    volumes:
      - "./grottconf.py:/app/grottconf.py"
    restart: unless-stopped
  influxdb.dc:
    image: influxdb:2.7.1
    volumes:
      - "./db:/var/lib/influxdb2"
    ports:
      - "8086:8086"
    environment:
      DOCKER_INFLUXDB_INIT_MODE: "setup"
      DOCKER_INFLUXDB_INIT_USERNAME: "admin"
      DOCKER_INFLUXDB_INIT_PASSWORD: "admin"
      DOCKER_INFLUXDB_INIT_ORG: "my_org"
      DOCKER_INFLUXDB_INIT_BUCKET: "grott"
      DOCKER_INFLUXDB_INIT_ADMIN_TOKEN: "xyz"
```

patcher-ms, 2023-10-04:
I am having the same issue. Any chance this could be supported by Grott itself? Patching `grottconf.py` works but it's not great