johanmeijer/grott #104: Noise in Proxy mode

Timwalet, 2022-03-04:
Hey guys,

I installed this service on my raspberry pi this week, because the connection between the dataloggers and the Growatt server was very unstable, and i was losing a lot of data.

The installation went fine and the first day i had a good result, but the previous two days a see a lot of noise in the data.
Let me show that with an example:
[image]
Here you see data from a non existing inverter, with extremely high numbers.

My setup has 2x  2000-s inverters from growatt, combined with 14 panels. 
The data is logged by two of those loggers: Growatt ShineLink LanBox
Both are set to the proxy server by the default portnumber 5279.
MQTT is separating the data based on the inverter ID.

In the screenshot underneath you see a correct data-example of one of the inverters:
[image]

and here is the other:
[image]

Because of this topic:
https://github.com/johanmeijer/grott/issues/15 
i tried setting the offset to different values, but without result.
The noise keeps coming, and the above correct inverters are sending incorrect data as a result.

i am in no means an expert in Python, more like a beginner, but i was expecting that this peace of code in Grottdata.py:
            if definedkey["pvstatus"] == 0 or definedkey["pvstatus"] == 1:
should filter out this noice, but as you can see, it isnt.

It is likely that my hardware had a problem?
Or is it possible to modify the code, so this noise will disappear?

Thanks in advance.

johanmeijer, 2022-03-05:
I have not seen this before. 

It really does seems to be  a hardware or communication error. Normally if the record values are at the wrong place (not the right layout definition used) then the serial number is still ok. But this is also already not readable so this is really  meshed up data. 

The header seems to be ok so probably it already went wrong in de communication between the datalogger and the inverter (or by creating/scrambling the content part of the record) .
 Can you send me the Growatt original data part in text so I can test with it. 

If it is a communication error I can try to see if the CRC (record check bytes) are ok, If not we can then discard the record. If this is oke (so the datalogger packed and sent rubbish inverter information). Then we have a challenge to detect if it is a wrong record. 

Grott latest automatic detection does not filter on pvstatus anymore. With the new inverter like SPH. SPF TL3 there are more statussen possible then 0 or 1 so it would be hard to filter on that one. But we can try to use it: everything with a pvstatus > a certain value, eg 20) but there is no guarantee that  everything lower lower then that is oke. We can also look at grid voltage or some other value. 

How does your configuration look like? Do you have 2 inverters with a shinelan box and RF connection or to separate shinelan / wifi  dataloggers. 
Does both inverters generates the noise or only one? 

The approach or error detection will a little be depend on your configuration.  My first thoughts: 

With the shinelan box: see if there is something in between that can disturb the RF signal. Try to exchange both RF dongles (does the failure move also?). Is the RS232 connection (the dongle) connector ok? Take it out and put it in again. 

With the shinewifi/wshine lan. Is the RS232 connection ok (re-install the dongles). Is the wifi connection ok (shinewife)? Is the LAN cable oke (shinelan)?  

The growatt server will not have any problem while it does not recognize the the PVserial. It might generate faulty MQTT or Influxdb records (I do not know what you use). But you can also filter on this while it will contains a strange pvserial. PVOutput logging will show strange values.

TimW85, 2022-03-05:
Thanks Johan for your answer.
A few things happened today. First of all i was able to update my shinelinkLAN boxes to software version 1.0.8.7. (yesterday it was 1.0.7.6). I noticed that since then the record layout switched from T05NNNN to T06NNNN and the good thing is that since then, the noise didnt appear anymore. Im not yet convinced that the problem is gone, because the first day of installation, it worked fine as well, so i will check my logs regularly.

Besides that, today was the day that Growatt announced to update their servers. Maybe that could have something to do with that.

I wrote some lines of code to prevent Grott to send the noise to MQTT, because that was influancing my home assistants stats a lot. 
I just declared a new variable with a list of my inverters, 
I changed line 375 of the grottdata.py file to:
<if conf.nomqtt != True and definedkey["pvserial"] in conf.pvinverterlist:>

and in line 377 i added the pvserial to the topic string, so in my case (with multiple inverters) the data ends up in different topics:
<publish.single(conf.mqtttopic + '/' + definedkey["pvserial"]>
Maybe it will help someone with the same issue.

Then regarding your questions:
**How does your configuration look like? Do you have 2 inverters with a shinelan box and RF connection or to separate shinelan / wifi dataloggers.**
      I have two inverters, both having a RF dongle that communicated with there own ShinelanBox. The ShinelanBoxes are connected by wire to my LAN. the WIFI is disabled.
  

**Does both inverters generates the noise or only one?**
I didnt test this when the noise was there, but as soon as i will have the noise again, i will unpower one of the ShineLanBoxes.
What is interesting is that the noise could be gone of 30 minutes, but as soon as it happens it is there in a high quantity. It reminds me a bit of the frequency of the log messages that "Data less then minimum record length, data not processed."

In this picture you can see what i mean whit missing data:
[image]

It was quit sunny in the Netherlands yesterday, so im sure that those peaks and valleys are because of a data issue.

Looking at the same graph of today (also very sunny and no noise):
[image]

I think a lot of the data didnt end up at the Growatt servers.

johanmeijer, 2022-03-21:
Hi @TimW85 Do you think this problem can be the same issue as in #107? 

It seems that sometimes (in #107 even 37% of the time) the shinelink / RF combination generates PVSerial errors. 

Is it totally gone after the firmware upgrade?

TimW85, 2022-03-21:
@johanmeijer  yes the "noise" is totally gone after the upgrade. Your interface works flowlessly for a couple of weeks now.
The 37%  of the time sounds familiar. I had the same frequency of noise, but the difference i notice is that in my case everything was wrong except the inverterID and in the #107 issue it seems that everything is correct, except for the inverterid. 

With that being said, i think you can add the growatt 2000S to the list of compatible devices and this thread can be closed.

closed 2022-03-21 by johanmeijer