johanmeijer/grott #105: no matching record layout found

ageurtse, 2022-03-05:
Hello, 

i have a growatt min 3600 TL/XH and the data isn't decrypted.

Could someone help me out

here is the output from grott
`- Growatt packet received:
                  <socket.socket fd=572, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.20', 58956), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x0f\x00\x06\x02\x41\x51\x03\x09\x33\x2c\x4f\x23\x33\x22\x77\x35\x3a\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61
                 \x74\x74\x3b\x72\x6e\x77\xf8\x74\x74\x47\x16\x6f\x77\x46\x64\x74\x47\xfe\xcf
                 \x74\x89\x35\x38\x76\x5c\x5f\x77\x3b\x35\x16\x26\x72\x75\x77\x60\x74\x74\x44
                 \x9a\x6f\x4b\x61\x48\x74\x1d\x72\x35\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
                 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x64\x8d\x76\x75\x40\x94
                 \x6f\x74\x61\x71\x74\x4f\x72\x55\x77\x6f\x74\x72\x40\x42\x65\x22\x73\xfa\x60
                 \x59\x75\x5f\x7d\x34\x66\xfa\x53\x6c\x68\x47\x6b\x21\x66\xc9\x66\x71\x70\xc2
                 \x7d\x96\x55\xe1\x7c\xfa\x61\x46\x74\x75\x72\x5d\x77\x53\x74\x46\x47\x40\x6f
                 \x45\x61\x46\x74\x75\x72\x5d\x77\x69\x74\x7c\x4e\x90\x6f\x77\x20\x38\x16\x26
                 \x42\x5a\x47\x56\x44\x45\x75\x44\x6e\x46\x61\x74\x74\x47\x61\xf3\x77\x53\x7d
                 \xc0\x4e\x90\x67\x33\x69\x62\x74\x53\x72\x6a\x7e\x0e\x7c\x88\x60\x62\x48\x67
                 \x46\x64\x53\x57\x55\x7f\x50\x71\x74\xe2\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
                 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x75\x4f\x72\x6f\x78\x60\x74\x50
                 \x47\x72\x6f\x77\x61\x74\x7f\xff\x7e\x5b\x77\x61\x35\x30\x0d\x42\x2d\x3d\x59
                 \x44\x44\x70\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6e\x83\x61\x74\x75\xd7\x72\x6e\x77\x61\x61\x08\x47
                 \x72\x6d\x2f\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x51\xda\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x10\x47\x78\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x23\x72\x0b\x77\x60\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x37\x2f\x39\x2c\x2c\x1f
                 \x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x61\x74\x74\x47\x72\x6f\x77\x61\x54
                 \x54\x67\x52\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xa0\xd2
         - Grott automatic protocol detection
         - Grott data record length 585
         - layout   :  T065103X
         - no matching record layout found, try generic
         - Record layout used :  T065103X
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 000f0006024151034e41433842475630475500000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000020000
                 007c00010099000000640000271000008ca003e8414c312e30005a416261001a0001000003e80
                 03c003c005a005a00000000000000000000000000000000000100640000000020202050562049
                 6e7665727465722020000013ec020107e6000300050008003a000e000607300a55128e141e073
                 00a55128e141e07300a55128e141e07a309e21293138d00320032003200320032003200320032
                 003200320008000809e20000414c62613035303730313236013100000000139c003209b409e20
                 844081600140005096f08fc2710271027102710271027100096000001e400ff4e2000ff4e2000
                 ff4e2000ff4e20010800000f0100240000000000000bb80c34000041444a30424a38303037000
                 0000000000000000000000000000000000000000001f40000019000010000157c000002580000
                 000000000000000016a8000000000000000000000064000a00000000000000000000000000000
                 00000000064006400010000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000005858585858585858585858585858585800000000000000002020202000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000d2bd
         - Grott data ack record or data record not defined no processing done`

ageurtse, 2022-03-05:
after adding invtype=t13 in the generic section of the grott.ini file i get the following output

         - Growatt packet received:
                  <socket.socket fd=548, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.20', 64234), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x80\x00\x06\x02\x41\x51\x03\x09\x33\x2c\x4f\x23\x33\x22\x77\x35\x3a\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61
                 \x74\x74\x3b\x72\x6e\x77\xf8\x74\x74\x47\x16\x6f\x77\x46\x64\x74\x47\xfe\xcf
                 \x74\x89\x35\x38\x76\x5c\x5f\x77\x3b\x35\x16\x26\x72\x75\x77\x60\x74\x74\x44
                 \x9a\x6f\x4b\x61\x48\x74\x1d\x72\x35\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
                 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x64\x8d\x76\x75\x40\x94
                 \x6f\x74\x61\x71\x74\x4e\x72\x49\x77\x75\x74\x72\x40\x42\x65\x22\x73\xfa\x60
                 \x59\x75\x5f\x7d\x34\x66\xfa\x53\x6c\x68\x47\x6b\x21\x66\xc9\x66\x71\x70\xc2
                 \x7d\x96\x55\xe1\x7c\xfa\x61\x46\x74\x75\x72\x5d\x77\x53\x74\x46\x47\x40\x6f
                 \x45\x61\x46\x74\x75\x72\x5d\x77\x69\x74\x7c\x4e\x90\x6f\x77\x20\x38\x16\x26
                 \x42\x5a\x47\x56\x44\x45\x75\x44\x6e\x46\x61\x74\x74\x47\x61\xf3\x77\x53\x7d
                 \xc0\x4e\x90\x67\x33\x69\x62\x74\x53\x72\x6a\x7e\x0e\x7c\x88\x60\x62\x48\x67
                 \x46\x64\x53\x57\x55\x7f\x50\x71\x74\xe2\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
                 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x75\x4f\x72\x6f\x78\x60\x74\x50
                 \x47\x72\x6f\x77\x61\x74\x7f\xff\x7e\x5b\x77\x61\x35\x30\x0d\x42\x2d\x3d\x59
                 \x44\x44\x70\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6e\x83\x61\x74\x75\xd7\x72\x6e\x77\x61\x61\x08\x47
                 \x72\x6d\x2f\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x51\xda\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x10\x47\x78\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x23\x72\x0b\x77\x60\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x37\x2f\x39\x2c\x2c\x1f
                 \x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x61\x74\x74\x47\x72\x6f\x77\x61\x54
                 \x54\x67\x52\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x24\xd7
         - Grott automatic protocol detection
         - Grott data record length 585
         - layout   :  T065103XTL3
         - no matching record layout found, try generic
         - Record layout used :  T065103XTL3
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 00800006024151034e41433842475630475500000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000020000
                 007c00010099000000640000271000008ca003e8414c312e30005a416261001a0001000003e80
                 03c003c005a005a00000000000000000000000000000000000100640000000020202050562049
                 6e7665727465722020000013ec020107e600030005000900260014000607300a55128e141e073
                 00a55128e141e07300a55128e141e07a309e21293138d00320032003200320032003200320032
                 003200320008000809e20000414c62613035303730313236013100000000139c003209b409e20
                 844081600140005096f08fc2710271027102710271027100096000001e400ff4e2000ff4e2000
                 ff4e2000ff4e20010800000f0100240000000000000bb80c34000041444a30424a38303037000
                 0000000000000000000000000000000000000000001f40000019000010000157c000002580000
                 000000000000000016a8000000000000000000000064000a00000000000000000000000000000
                 00000000064006400010000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000005858585858585858585858585858585800000000000000002020202000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 0000000000056b8
         - Grott data ack record or data record not defined no processing done

ageurtse, 2022-03-05:
and a nother update, 

now i'm running the beta version 2.7-beta and it looks good.

and the data that is proccesd
        - Growatt packet received:
                  <socket.socket fd=532, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.20', 64426), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x08\x00\x06\x02\x41\x51\x04\x09\x33\x2c\x4f\x23\x33\x22\x77\x35\x3a\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x6a
                 \xcc\x78\x73\x72\x6e\x77\x61\x28\x44\x57\xdf\x6f\x40\x61\x74\x28\x77\x71\x94
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x28\x77\x72\x6f\x77\x61\x74\x74\x1f\xac\x7c\xff\x68\x58
                 \x74\x25\x72\x6f\x2e\x64\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x4e\x5e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x33\x47\x61\x75\xfe\x33\x72\x6f\x77\x78\x74\x74\x47\x92\x6f\x77\x61\x90\x74
                 \x47\x72\x75\x77\x61\x74\x90\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x5d\x72\x6f\x77\x85\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x75\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75
                 \x33\x46\x35\x6f\x77\x61\x74\x75\x0f\x62\xda\x77\x61\x3a\x54\x47\x4d\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x66\x4f\x68\x4c\x74\x5d\x72
                 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x1b\x42\x6f\x77\x61\x6e\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x8b
                 \x77\x61\x74\x6e\x47\x72\x6f\x93\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x7f\xd6\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x75\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x60\xdd\x74\x2f\x73
                 \xc6\x79\x71\x52\x86\x47\x7b\x6f\x70\x61\x76\x74\x47\x7a\xe4
         - Grott automatic protocol detection
         - Grott data record length 585
         - layout   :  T065104XTL3
         - no matching record layout found, try generic
         - Record layout used :  T06NNNNXTL3
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 00080006024151044e41433842475630475500000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000020bb8
                 0c34000100005c3010ad003700005c3003fb00000000000000000000000000000000000000000
                 00000005c3000000000000058de1388092c006200005905000000000000000000000000000000
                 00092c00000000000000000000000000005c3000018a7400000019000000e0000000e40000001
                 a000000e400000000000000000000000000000000000000000000000000000000000000000000
                 001a000000e400000000000000000000001a00000000fffa00000000000000000000014701470
                 0000000014810b500004e20003f000000000000000000000000000009380938001a0000000000
                 00003c0000000000000000000000005c300000001a0c350cb1000000000000000000000000000
                 000000000000000000000000000e40000001a000000e400000000000000000000000000000000
                 00000000000000000000000000000000000000002710000010a10000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 000000000000000000000000000100010000000000000000000001a9006801a90e1026f200090
                 00700020000088b
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06NNNNXTL3

         - Grott data record date/time processing started
         - no or no valid time/date found, grott server time will be used (buffer records not sent!)
[... 34 lines cut ...]
                 -  pvipmtemperature     :  32.7
                 -  pvboottemperature    :  0.0
[... cut ...]
                         "pvenergytotal": 26, "totworktime": 228, "epv1today": 0, "epv1total": 0,
                         "epv2today": 0, "epv2total": 0, "epvtotal": 0, "pvtemperature": 327,
                         "pvipmtemperature": 327, "pvboottemperature": 0, "uwBatVolt_DSP": 328,
                         "pbusvolt": 4277, "nbusvolt": 0}}
         - Grott MQTT topic used : energy/growatt
         - MQTT message message sent
         - Grott Send data to PVOutput disabled
         - Grott Send data to Influx disabled
         - Grott extension processing disabled

johanmeijer, 2022-03-05:
Hallo @ageurtse,   
A T06103TLX record is an announce record and does not need processing. More interesting is what happens with a T0615104TL3X. That are the real data records and Grott will process them (f all goes well). 

Direct after startup the "03" record will be send and after that (if the growatt server acknowledge this 03) the inverter will start sending the "04" (or "50" if records were buffered). Be aware there are a lot of other records in between (Ping, configuration etc) those also need no processing by Grott.

ageurtse, 2022-03-05:
well it is connected to the growatt server, but it looks like there aren't any updates. 
i had installed the pv pannels now for 2 days and the growatt app is realy disapointing and not realy working, updates are not every 5 minutes or 1 minute like i setup .

if i use your server script it seem all workin well.

johanmeijer, 2022-03-05:
Strange. There must be a definition error the at the growatt site I think. 

Although I also understand that the Growatt site has some problems lately ( I have not noticed that yet).

ageurtse, 2022-03-05:
i think that i install grottserver and grot on my homeassistant base system (Debian) that is a stable platform, so for now i skip the app. maybe sometime later i try it again.

johanmeijer, 2022-03-05:
Yes that is ok. That are ping records Grott does not process them. As long as you see "04"or "50" records being processed with messages send to MQTT it is ok.

ageurtse, 2022-03-05:
so beside the buggy connection with server.growatt.com there is nothing wrong. 
Thanks for the help and it looks like this could be closed.

closed 2022-03-05 by ageurtse