johanmeijer/grott #107: InfluxDB 2.0 "unable to parse" error

bkleef, 2022-03-09:
In looks like the character `i` is appended after each JSON (integer) value? Like `epv1today=70i`. This happens in Grott version 2.6.1g.

```
...
	 - Growatt data decrypted V2
	 - Grott influxdb jsonmsg:
			 [{'measurement': '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', 'time':
			 '2022-03-09T15:24:18', 'fields': {'pvserial':
			 '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', 'recortype1': 3000,
			 'recortype2': 3124, 'pvstatus': 1, 'pvpowerin': 18998, 'pv1voltage': 3356,
			 'pv1current': 52, 'pv1watt': 17589, 'pv2voltage': 3082, 'pv2current': 4,
			 'pv2watt': 1409, 'pvpowerout': 18599, 'pvfrequentie': 4996, 'pvgridvoltage':
			 2336, 'pvgridcurrent': 79, 'pvgridpower': 18600, 'pvgridvoltage2': 0,
			 'pvgridcurrent2': 0, 'pvgridpower2': 0, 'pvgridvoltage3': 0,
			 'pvgridcurrent3': 0, 'pvgridpower3': 0, 'totworktime': 1006976,
			 'pvenergytoday': 170, 'pvenergytotal': 2085, 'epvtotal': 2160, 'epv1today':
			 70, 'epv1total': 1068, 'epv2today': 106, 'epv2total': 1092, 'pvtemperature':
			 348, 'pvipmtemperature': 372}}]
	 - Grott InfluxDB error
(400)
Reason: Bad Request
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': '2.1.1', 'X-Platform-Error-Code': 'invalid', 'Date': 'Wed, 09 Mar 2022 15:24:19 GMT', 'Content-Length': '694'})
HTTP response body: {"code":"invalid","message":"unable to parse 'epv1today=70i,epv1total=1068i,epv2today=106i,epv2total=1092i,epvtotal=2160i,pv1current=52i,pv1voltage=3356i,pv1watt=17589i,pv2current=4i,pv2voltage=3082i,pv2watt=1409i,pvenergytoday=170i,pvenergytotal=2085i,pvfrequentie=4996i,pvgridcurrent=79i,pvgridcurrent2=0i,pvgridcurrent3=0i,pvgridpower=18600i,pvgridpower2=0i,pvgridpower3=0i,pvgridvoltage=2336i,pvgridvoltage2=0i,pvgridvoltage3=0i,pvipmtemperature=372i,pvpowerin=18998i,pvpowerout=18599i,pvserial=\"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\",pvstatus=1i,pvtemperature=348i,recortype1=3000i,recortype2=3124i,totworktime=1006976i 1646839458000000000': invalid field format"}

Grott Influxdb write error, grott will be stopped
```

johanmeijer, 2022-03-16:
OK. Interesting. I will look at that. I do not have that problem. 

Do you use the docker or native python version? You might have to update the python influxDBClient library. This one is influxdb version depended.

bkleef, 2022-03-18:
@johanmeijer looks like this bug occurs when `pvserial` has no value: `-  pvserial             :`

[Here](https://www.diffchecker.com/VUhkEYJf) a message that works (left) and a few minutes later a message (right) where `pvserial` has no value and the error occurs.

Remarkably, the keys `measurement` and `pvserial` have the value `\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00` in the _Grott influxdb jsonmsg_ message if `pvserial` has no value.

johanmeijer, 2022-03-18:
This seems very strange your record does not contain the inverter serial number. That is something I have not seen before. How does your configuration looks like? Which inverter type and datalogger do you use?

For Influxdb the pvserial is used for measurement identification. I do understand this can cause same errors.

bkleef, 2022-03-21:
Configuration is still the same as [previously shared](https://github.com/johanmeijer/grott/issues/107#issuecomment-1069271601).

Inverter type: Growatt hybrid inverter MIN 6000TL-XH
Datalogger: Growatt Shinelink

[Here](https://drive.google.com/file/d/1_LbaBbMnk6sBj_u-OLsG-RKkVRbcP6on/view?usp=sharing) are more logs. Of the 245 times that `- pvserial` appears here, 150 have a value. In 37% of the cases, `pvserial` has no value.

johanmeijer, 2022-03-21:
Sorry I read over that on my phone. But I am at the PC again ;). Nice inverter do you also have a battery connected? 

Did you have try to update the firmware of the shinelink. Maybe this is really a bug? Or maybe a total reset might help. 

If I look at your log (first look) I see also a lot of T065103X recorded. The "03" records are called "announce" records and announces the record to the growatt server. This is normally only than after a restart of Grott, a start of the inverter in the morning of after connection loss. And is only send once, not that often as in your log. This might point out to a bad connection with the growatt server (or maybe the RF connection between inverter and shinelink?). 

If we want to analyse the full protocol then we need to specify gminrecl = 1 this will show all trafic. This will generate even more log records. But we see then if the growatt server is acking the "03" records. The ack of this record is the start of a normal communication sequence. 

It is a strange error. I have not seen it before. I am also interested why you do not have a time in the data (04) record but I have seen that before with a shinelink. 

This is a "special" problem. I hope we can solve this. Of course we can insert an inverter id (specified ginverterid = ) if there is none detected. This will be a small change in Grott. But it feels like a dirty fix to solve growatt bugs. 

Are you willing to test this without the growatt server (using de beta Grottserver see discussions #98)? We can make a docker container for that also. 

I will also analyse the log more if I can see other strange things. If you want we can also communicatie in an other way (Mail, Chat, Zoom/Teams), this might be a difficult proces.

johanmeijer, 2022-03-21:
@bkleef maybe this error is the same as mentioned in #104 ?

johanmeijer, 2022-03-21:
I really think this is a connection error (between RF module and shinelink) or a firmware error in shinelink or RF module. 

The bulk "03" announce records do not contain an inverter ID either and will not be accepted by the growatt server. That is why there a lot of this records in your log. The shinelink will resend a new "03"  until it is accepted. 
In your file a count approximately 5500 "03" records (daily I have 2 or 3) and as you said only 245 "04" records (of which only 150 have the right information). 
Now I can understand growatt has communication problems if all Shinelink modules are doing this. 

Is there a possibility the RF communication is disturbed by some other signal or is the distance between inverter/shinelan and to big? Although I suppose that not only the PVserial should contain strange data.

johanmeijer, 2022-03-21:
@bkleef see reaction on #104. Maybe you can perform a firmware upgrade?

bkleef, 2022-03-21:
Thank you very much for your investigation. Quick reply, firmware version is 7.0.0.1 (hardware version is V1.0). Do you think this is the latest version?

johanmeijer, 2022-03-21:
I really do not know (if have a shinewifi). You can see this at the datalogger pages at the growatt server website. 

[image]

johanmeijer, 2022-03-30:
@bkleef  Have you already try to update the Firmware and did this show any improvements

bkleef, 2022-03-30:
@johanmeijer I have contacted Growatt support. Today they have tried 3 times to update the box firmware without success. They will look into it internally.
The RF stick firmware is up to date.

The Shinelan-X is now also on the same floor and placed 2 meters away from the inverter. This makes no difference.
I'm thinking of buying the Growatt ShineWifi-X, maybe this will solve everything?

I also set `gminrecl=1` and `ginverterid=AHJ0BJ801H`. Log files can be downloaded [here](https://drive.google.com/file/d/1lW9fDLVSeoWxaHNlNU8MKKzo0iCc8Gl4/view?usp=sharing).

johanmeijer, 2022-03-31:
It is really strange. Seems a hardware/ firmware error to me. To be honest I expected the RF stick would give the problem (because that is interfacing with the inverter and retrieves the Inverterid). But if this one is up to date then the change will be low I think. 

Just to be sure. What is your data transfer interval. I can not determine this from your log is seems to be very irregular. Maybe if you set this interval to low the datalogger can not keep up with this?

Using an other datalogger (shiniwifi) might solve the problem indeed (if is not inverter related).

Do you want me to implement a bypass (insert inverterid when this is not the datarecord)? I will also commit your pull request soon.

johanmeijer, 2022-06-10:
> I just received an update from Growatt support:
>
> Er is overal het vermoedden dat het aan de server ligt. Er is onderhoudt
> gepland voor morgen. Dit onderhoud duurt waarschijnlijk tot na het weekend,
> daarna zou alles stabieler moeten zijn en zal de update tijd hopelijk
> verkorten.

I am curious if it will be solved.

rany2, 2022-06-10:
@johanmeijer What's the issue? I'm personally been using InfluxDB v2 for a few weeks without any issues.

johanmeijer, 2022-06-10:
A lot of reconnects with the growatt server (losing connection and reinit again)

closed 2023-08-13 by johanmeijer