johanmeijer/grott #110: influxdb2 only recording values for 1 datalogger

ronosan, 2022-03-16:
Hi,

I installed grott v2.6.1g and influxdb2 and got it working properly for 1 datalogger (i can see data in influxdb). However, my 2nd datalogger isn't showing up in influxdb despite a measurement has been created automatically by grott. This data logger is sending messages correctly to grott (running in proxy mode, same instance used by the working datalogger) and i can see the data is routed to Growatt servers correctly and displays in the Growatt phone app. Grott logs also show an influxdb json was created correctly for the troublesome datalogger, successfully sent to influxdb, but mysteriously missing with no data in influxdb?

Sample grott output for datalogger missing influxdb data:
`	 - Growatt packet received:
		  <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.123', 5279), raddr=('192.168.2.102', 51575)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.123', 61004), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x01\xbe\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x47\x58\x46\x43\x77\x43\x5e\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x37\x23\x25\x46\x4d\x74\x47\x5e\x42\x51\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x77\x77\x64\x56\x75\x73\x75\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x3f\xb8\x49\x11\x6f\x6b\x61\x74\x5e\xb7\x79\x69
		 \x77\x7a\x74\x74\x67\xae\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x0d\x35\x7c\xf1\x68\x0d
		 \x74\x17\x72\x6f\x3d\xec\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x4e\x0b\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x25\x30\x65\xd5\x82\xde\x72\x6f\x77\xe5\x74\x77\x54\x2c\x6f\x74\x7e\x99\x74
		 \x47\x72\x22\x77\x60\xb2\xdc\x47\x72\x6f\x4e\x61\x75\x2d\x02\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x40\x55\x6f\x69\x61\x74\x74\x47\x66\x0e\x77\x61\x75
		 \x2f\x47\x72\x6f\x77\x61\x74\x75\xdf\x7d\xf7\x77\x61\x3a\x54\x47\x54\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xb4\x69
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 01be0006024101045847443039323730313100000000000000000000000000000000000000005
		 8544432393335313530000000000000000000000000000000000000000016031011071c020bb8
		 0c34000100004bcc0e63001c00002af00b06001b000020dc00000000000000000000000000000
		 000000000000000000000004a4713860979005000004a8d000000000000000000000000000000
		 00097900000000000000000000000000004a4704a1f699000000840003135e00031fed0000004
		 d0001c6a800000039000159450000000000000000000000000000000000000000000000000000
		 000000000000000000000000000000000000000000000727001e0000000014610000015b00000
		 000000001980f9800004e20002600000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000c350cb1000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000c606
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
[... 183 lines cut ...]
			 "epv2total": 6800, "pvtemperature": 341, "pvipmtemperature": 404}}
	 - MQTT message message sent
	 - Grott Send data to PVOutput disabled 
	 - Grott InfluxDB publihing started
	 - Timezone local specified default timezone used
	 - Grott original time :  2022-03-16T17:15:05 adjusted UTC time for influx :  2022-03-16T07:15:05
	 - Grott influxdb jsonmsg: 
			 [{'measurement': 'HMG4B03349', 'time': '2022-03-16T07:15:05', 'fields':
			 {'pvserial': 'HMG4B03349', 'recortype1': 3000, 'recortype2': 3124,
			 'pvstatus': 1, 'pvpowerin': 7014, 'pv1voltage': 4170, 'pv1current': 7,
			 'pv1watt': 3743, 'pv2voltage': 1673, 'pv2current': 17, 'pv2watt': 3271,
			 'pvpowerout': 6874, 'pvfrequentie': 4997, 'pvgridvoltage': 2424,
			 'pvgridcurrent': 29, 'pvgridpower': 6908, 'pvgridvoltage2': 0,
			 'pvgridcurrent2': 0, 'pvgridpower2': 0, 'pvgridvoltage3': 0,
			 'pvgridcurrent3': 0, 'pvgridpower3': 0, 'totworktime': 8936827,
			 'pvenergytoday': 133, 'pvenergytotal': 22790, 'epvtotal': 23260,
			 'epv1today': 93, 'epv1total': 16460, 'epv2today': 41, 'epv2total': 6800,
			 'pvtemperature': 341, 'pvipmtemperature': 404}}]
	 - Grott write to influxdb v2
	 - Grott extension processing disabled 
`
sample screenshot from influxdb:
[image]

Any idea what's going on? I can't see any errors anywhere :/

johanmeijer, 2022-03-16:
Hmm I do not see any any error either. What if you filter on only 1 measurement? Can you look in table or csv format if there are
measurements bor both?

ronosan, 2022-03-16:
filtering on 1 measurement for the correct datalogger:
[image]

filtering on 1 measurement for the 'missing data' datalogger:
[image]

Upon closer inspection, those few data points for the 'missing data' datalogger, appears to coincide with me attempting to troubleshoot this problem by restarting grott/influxdb. I've tried the latest grott master branch, and branch 2.5.4 but to no success.

Has anyone else managed to successfully run grott + influxdb with more than 1 datalogger? At the moment i don't think it's hardware or network issue between my dataloggers to grott (because the correct data is still forwarded to growatt servers). Could it be some sort of compatibility with the API between grott and influxdb? 

BTW I'm running influxdb2-2.1.1

johanmeijer, 2022-03-16:
This might be a network connection error but then I do not expect that the measurements in the growatt website or phone app would be right.

Do you see any errors in the grott log? Do you see all influxdb writes for both inverters in the grott log?

ronosan, 2022-03-16:
I checked the growatt phone app and it has accurate data points at 5 minute intervals. I also see every single packet logged by grott as received from both dataloggers. The biggest mystery is why the datapoints for 1 logger feeds into influxdb correctly whilst the other vanishes. Influxdb spits out irrelevant info messages (mostly compaction/snapshot related) and no errors/fatal messages.

There are absolutely no errors at all in the grott log running in verbose mode.

InfluxDB and grott are running on the same PC.

ronosan, 2022-03-17:
i think there's an inherent problem with influxdb. I spent the day setting up telegraf agent to tap into grott's mqtt stream dataset and plumb that into influxdb. Same problem with the metric for the 2nd datalogger is missing once delivered to influxdb. Extremely frustrating.

Upon further reading into influxdb, i'm coming to the conclusion that it's a terribly unreliable database as they have bigger problems since v1.8 where chunks of existing data can suddenly disappear due to corruption. I think i need to explore an alternative as a datastore...

closed 2022-03-17 by ronosan

johanmeijer, 2022-03-17:
Ok. Have not problems with it until now. If you know Which database you want use (e.g. mysql, postgres,mongodb, Prometheus) we can try to build in support (create an extension) for that one as well.

I have to admit I primarily use influxdb 1.x because that was the standard for raspberry pi. I have influxdb 2.x running in a docker container for testing (support) purposes.