johanmeijer/grott #111: Wrong date time for part of the day

andersonshatch, 2022-03-18:
I have just enabled InfluxDBv2 this week and noticed that date/time handling of the inverter's messages seems to be incorrect for part of the day.
At sunrise and sunset, the time calculated from messages received from the inverter appears correct, logs indicate the messages are not buffered, and parsed time from the message is within a few seconds or few minutes of the time the message is received.

But, from approximately 13:45 to 17:28 each day (I looked at the logs from a couple days, it was roughly the same time period) the time is quite off:
```
2022-03-18T13:36:56.833452667Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:35:55", "buffered": "no",
2022-03-18T13:37:56.961578650Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:36:50", "buffered": "no",
2022-03-18T13:38:57.046554246Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:37:45", "buffered": "no",
2022-03-18T13:39:57.155157141Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:38:41", "buffered": "no",
2022-03-18T13:40:57.361030987Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:39:36", "buffered": "no",
2022-03-18T13:41:57.476243957Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:40:31", "buffered": "no",
2022-03-18T13:42:57.573980129Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:41:26", "buffered": "no",
2022-03-18T13:43:57.692316172Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:42:22", "buffered": "no",
2022-03-18T13:44:57.789517176Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T13:43:17", "buffered": "no",
2022-03-18T13:45:57.897120882Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:34:43", "buffered": "no",
2022-03-18T13:46:58.100547933Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:35:38", "buffered": "no",
2022-03-18T13:47:58.224055441Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:36:33", "buffered": "no",
2022-03-18T13:48:58.339871368Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:37:29", "buffered": "no",
2022-03-18T13:49:58.726893056Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:38:24", "buffered": "no",
2022-03-18T13:50:58.876113297Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:39:19", "buffered": "no",
2022-03-18T13:51:59.015783212Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:40:15", "buffered": "no",
2022-03-18T13:52:59.127975551Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:41:10", "buffered": "no",
2022-03-18T13:53:59.231502502Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T06:42:06", "buffered": "no",
...
2022-03-18T17:21:49.403454319Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T09:51:24", "buffered": "no",
2022-03-18T17:22:49.570157294Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T09:52:19", "buffered": "no",
2022-03-18T17:23:49.708396451Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T09:53:15", "buffered": "no",
2022-03-18T17:24:50.960492385Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T09:54:10", "buffered": "no",
2022-03-18T17:25:51.042817533Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T09:55:06", "buffered": "no",
2022-03-18T17:26:51.135718020Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T09:56:01", "buffered": "no",
2022-03-18T17:27:18.440846953Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T09:56:26", "buffered": "no",
2022-03-18T17:27:53.465315487Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T09:56:26", "buffered": "no",
2022-03-18T17:28:53.574827769Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T17:28:44", "buffered": "no",
2022-03-18T17:29:53.684685340Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T17:29:37", "buffered": "no",
2022-03-18T17:30:53.988419499Z                   {"device": "JTG8BM50BP", "time": "2022-03-18T17:30:31", "buffered": "no",
```
(Left timestamp is from `docker-compose logs -t` and right timestamp is the first part of the MQTT message.
This results in bad data being stored in Influx, but live consumers e.g. Home Assistant via MQTT where the date time is not used, maintain proper history by just trusting that the latest result is from "now".

Happy to provide more log information to help debug this, but not sure what is pertinent...

Using:
- grott version `2.7.2`
- ShineWiFi-X
- docker-compose with env: `TZ=Europe/London`

johanmeijer, 2022-03-19:
Grott uses date / time that is inserted by the inverter/datalogger. If that is not available it uses the server (docker container) time.

You can see this in this log messages:

          - Grott data record date/time processing started
	 - no or no valid time/date found, grott server time will be used (buffer records not sent!)

I have not noticed any delays in Grott in processing the data. Can you send a copy of the log with a wrong record started from growatt package received till the write to inluxdb? Maybe we can detect where the delay is comming from.

It might be a delay caused by the growatt server. I think a new record is being sent after confirmation of the previous record is received by growatt.

andersonshatch, 2022-03-19:
Here's an example section from yesterday, the reading should be from around 15:18 per the container time, but was stored in influx at 07:59:

<details>

```
2022-03-18T15:18:26.878296443Z 	 - Growatt packet received:
2022-03-18T15:18:26.878360132Z 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.22.0.2', 57188), raddr=('47.91.67.66', 5279)>
2022-03-18T15:18:26.878674308Z 	 - Growatt original Data:
2022-03-18T15:18:26.879381911Z 		 \x02\x3b\x00\x06\x03\x3f\x01\x04\x1f\x35\x2b\x42\x23\x3c\x3c\x75\x3f\x28\x77
2022-03-18T15:18:26.879416539Z 		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2022-03-18T15:18:26.879458220Z 		 \x25\x23\x26\x4c\x36\x0a\x47\x5f\x35\x31\x74\x74\x47\x72\x6f\x77\x61\x74\x74
2022-03-18T15:18:26.879571519Z 		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x77\x77\x66\x40\x49\x77\x74\x6a
2022-03-18T15:18:26.879590520Z 		 \xcc\x78\x73\x72\x6e\x77\x61\x25\xa9\x40\x46\x6f\x4e\x61\x74\x5d\x66\x7a\x9c
2022-03-18T15:18:26.879617844Z 		 \x77\x4c\x74\x74\x6f\xce\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2022-03-18T15:18:26.879630244Z 		 \x72\x6f\x77\x61\x74\x25\x9a\x72\x6f\x77\x61\x74\x74\x16\x40\x7c\xf9\x68\xfb
2022-03-18T15:18:26.879655310Z 		 \x74\x12\x72\x6f\x26\x5e\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
2022-03-18T15:18:26.879667607Z 		 \x61\x74\x74\x4e\xfd\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2022-03-18T15:18:26.879690798Z 		 \x3e\xaa\x61\x7d\x4b\x74\x72\x6f\x77\xef\x74\x74\x45\x9c\x6f\x77\x63\x89\x74
2022-03-18T15:18:26.879748592Z 		 \x47\x72\x40\x77\x61\x75\x6e\x47\x72\x6f\x14\x61\x74\x75\xa4\x72\x6f\x77\x61
2022-03-18T15:18:26.879768942Z 		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
2022-03-18T15:18:26.879774309Z 		 \x77\x61\x74\x74\xd5\x72\x6f\x75\x9c\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2022-03-18T15:18:26.879777577Z 		 \x72\xfd\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75
2022-03-18T15:18:26.879790954Z 		 \xf7\x46\xf1\x6f\x77\x61\x74\x75\x30\x7c\xf3\x77\x61\x3a\x54\x47\x4b\x6f\x77
2022-03-18T15:18:26.879794361Z 		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x67\x58\x69\x5b\x74\xd5\x72
2022-03-18T15:18:26.879797300Z 		 \x6f\x77\x61\x74\x74\x47\x66\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
2022-03-18T15:18:26.879823219Z 		 \x16\xaf\x6f\x77\x61\xe6\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
2022-03-18T15:18:26.879829378Z 		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x71\x6f
2022-03-18T15:18:26.879832410Z 		 \x77\x61\x74\xe6\x47\x72\x6c\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2022-03-18T15:18:26.879835469Z 		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
2022-03-18T15:18:26.879838354Z 		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x61\xd9\x61\x74\x74\x47\x72\x6f\x77
2022-03-18T15:18:26.879841368Z 		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2022-03-18T15:18:26.879844286Z 		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
2022-03-18T15:18:26.879847268Z 		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
2022-03-18T15:18:26.879994251Z 		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
2022-03-18T15:18:26.880002875Z 		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2022-03-18T15:18:26.880006313Z 		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
2022-03-18T15:18:26.880009870Z 		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
2022-03-18T15:18:26.880013049Z 		 \x61\x75\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\xcf\x74\xac\x73
2022-03-18T15:18:26.880016364Z 		 \x18\x79\x71\x52\x90\x47\x7b\x6f\x70\x61\x76\x74\x47\x7e\xdd\x7a\x4f\x74\x74
2022-03-18T15:18:26.880065645Z 		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
2022-03-18T15:18:26.880069501Z 		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
2022-03-18T15:18:26.880072515Z 		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2022-03-18T15:18:26.880075513Z 		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
2022-03-18T15:18:26.880078507Z 		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
2022-03-18T15:18:26.880081471Z 		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2022-03-18T15:18:26.880243940Z 		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
2022-03-18T15:18:26.880300663Z 		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
2022-03-18T15:18:26.880342401Z 		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
2022-03-18T15:18:26.880348329Z 		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2022-03-18T15:18:26.880351427Z 		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
2022-03-18T15:18:26.880354305Z 		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
2022-03-18T15:18:26.880357364Z 		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2022-03-18T15:18:26.880360356Z 		 \x6f\xb3\xb3
2022-03-18T15:18:26.880363074Z 	 - Grott automatic protocol detection
[... cut ...]
2022-03-18T15:18:26.885122340Z 			 'pvgridpower2': 0, 'pvgridvoltage3': 0, 'pvgridcurrent3': 0, 'pvgridpower3':
2022-03-18T15:18:26.885124953Z 			 0, 'totworktime': 606003, 'pvenergytoday': 142, 'pvenergytotal': 750,
2022-03-18T15:18:26.885127505Z 			 'epvtotal': 765, 'epv1today': 47, 'epv1total': 282, 'epv2today': 99,
2022-03-18T15:18:26.885129975Z 			 'epv2total': 483, 'pvtemperature': 387, 'pvipmtemperature': 375}}]
2022-03-18T15:18:26.885132507Z 	 - Grott write to influxdb v2
2022-03-18T15:18:26.939054926Z 	 - Grott extension processing disabled 
```
</details>

johanmeijer, 2022-03-19:
Very strange. This is the time written in the data record by the inverter/datalogger. What type of inverter/datalogger do you have?

johanmeijer, 2022-03-19:
Sorry accidently closed

johanmeijer, 2022-03-19:
Be aware specifying gtime = server means also no buffered (history) records will be processed anymore.

andersonshatch, 2022-03-19:
Thanks for looking this over!
Have added `gtime=server` just now, will report back tomorrow if it worked. So far from grepping logs I have not had any buffered data sent over, but I suppose that might be a downside if I need to reboot the machine running this.

andersonshatch, 2022-03-21:
That definitely fixed things for yesterday's data, and then of course this morning there was an issue with DNS meaning Grott kept restarting because it couldn't resolve InfluxDB, so I did miss some data that would've been buffered :(
I came across another project that set the time on the inverter if it was too different from "now", I.e. your computer time, maybe will explore something like that

johanmeijer, 2022-03-21:
@andersonshatch 
Ok. So there is a bypass now. It is a pity you lose the buffered records then. 

Yes the protocol allows updating the time of the datalogger. Normally this is done multiple times a day by the growatt server (the first time direct after start-up in the morning). You can see this in the Grott log (if you use gblockcmd = "True", the time command is not blocked). 

     - Grott: Configure command not blocked :  Time
 
This can also be forced by stopping/starting Grott while the initialization will be redone. 

I am am working at a feature within Grott to sent commands (including time) but seeing the fact you have to interrupt/interfere with the communication between the inverter( datalogger) andthe growatt server this is not that easy. That is why it probably only will work when using Grott as a server but this is only in beta now (see #98 ). 

There is also an API to "talk" with the growatt server. I think that it will also be possible to send commands via this interface. I have not looked into that yet. 

An other way to set time is via the Growatt website and use the datalogger (not inverter!) register settings. This is part of the advanced settings and you have to set register 31. I have not tried this either (no need). But I would start with reading and then try if you can copy the format. 

Thinking about why the datalogger time seems to "drift" in the afternoon. What I understood is that the growatt server has some issues lately (because of to much traffic). Maybe the time command are delayed and sent to (arrive at) the datalogger to  late. Can you determine if this is the case in the Grott log or sent the grott log of one day (zip)?

andersonshatch, 2022-03-28:
> Can you determine if this is the case in the Grott log or sent the grott log of one day (zip)?

Apologies for the delay, I haven't had a chance to look at this more closely yet, though, here's yesterday's log output
[grott.txt.zip](https://github.com/johanmeijer/grott/files/8359521/grott.txt.zip)

johanmeijer, 2022-03-31:
@andersonshatch really strange. Your clock is really drifting a lot. After a set (every time after a restart and a normally a clock set command a couple of times a day) the time is losing track very fast. 
This must almost be a Hardware or firmware problem.

pmp02, 2023-06-29:
Hi @johanmeijer and @andersonshatch ,
Sorry for joining in an old topic like this, but just installed Grott few days ago (via docker) and I encounter the same (I guess) problem with the time, which is pretty inconsistent.

I also have a ShineWifi-X data logger and I'm also using the CSV Extension to easier look at the data.
Have you have a better recommendation (except the gtime=server)?
Below the times from Grott server (via CSV):

```
2023-06-28T05:46:48
2023-06-28T05:51:00
2023-06-28T05:55:13
2023-06-28T05:59:28
2023-06-28T06:03:42
2023-06-28T06:07:55
2023-06-28T06:12:07
2023-06-28T06:16:22
2023-06-28T06:20:34
2023-06-28T06:24:41
2023-06-28T06:28:56
2023-06-28T06:33:11
2023-06-28T06:47:34
2023-06-28T06:41:29
2023-06-28T06:57:35
2023-06-28T07:02:35
2023-06-28T06:53:59
2023-06-28T07:12:51
2023-06-28T07:17:42
2023-06-28T07:22:34
2023-06-28T07:27:41
2023-06-28T07:32:30
2023-06-28T07:19:00
2023-06-28T07:23:09
2023-06-28T07:47:37
2023-06-28T07:31:30
2023-06-28T07:57:34
2023-06-28T07:39:52
2023-06-28T08:07:36
2023-06-28T07:48:18
2023-06-28T07:52:27
2023-06-28T07:56:34
2023-06-28T08:00:51
2023-06-28T08:05:07
2023-06-28T08:09:22
2023-06-28T08:13:36
2023-06-28T08:17:49
2023-06-28T08:17:50
2023-06-28T08:21:58
2023-06-28T08:26:12
2023-06-28T08:30:28
2023-06-28T08:34:41
2023-06-28T08:38:58
2023-06-28T08:43:05
2023-06-28T08:47:23
2023-06-28T08:51:35
2023-06-28T08:55:43
2023-06-28T09:38:43
2023-06-28T09:43:44
2023-06-28T09:48:46
2023-06-28T09:53:47
2023-06-28T09:58:44
2023-06-28T09:20:46
2023-06-28T10:08:45
2023-06-28T09:29:06
2023-06-28T10:18:46
2023-06-28T09:37:38
2023-06-28T10:28:51
2023-06-28T10:33:46
2023-06-28T10:38:46
2023-06-28T10:43:48
2023-06-28T10:48:39
2023-06-28T10:53:40
2023-06-28T10:58:44
2023-06-28T11:03:40
2023-06-28T11:08:33
2023-06-28T11:13:39
2023-06-28T10:23:10
2023-06-28T11:23:37
2023-06-28T11:28:38
2023-06-28T11:33:42
2023-06-28T11:38:33
2023-06-28T10:43:56
2023-06-28T11:48:38
2023-06-28T11:53:37
2023-06-28T11:58:36
2023-06-28T11:00:39
2023-06-28T12:08:35
2023-06-28T12:13:37
2023-06-28T11:13:10
2023-06-28T12:23:39
2023-06-28T11:21:27
2023-06-28T12:33:34
2023-06-28T11:29:48
2023-06-28T11:34:00
2023-06-28T11:38:12
2023-06-28T11:42:26
2023-06-28T05:48:45
2023-06-28T05:53:08
2023-06-28T05:57:18
2023-06-28T06:01:36
2023-06-28T06:05:50
2023-06-28T06:05:51
2023-06-28T06:10:06
2023-06-28T06:14:27
2023-06-28T06:18:32
2023-06-28T06:22:48
2023-06-28T06:27:02
2023-06-28T06:31:15
2023-06-28T06:35:29
2023-06-28T06:39:43
2023-06-28T06:44:07
2023-06-28T06:48:17
2023-06-28T06:52:37
2023-06-28T06:56:46
2023-06-28T07:01:06
2023-06-28T07:05:19
2023-06-28T07:09:37
2023-06-28T07:13:47
2023-06-28T07:18:04
2023-06-28T07:22:21
2023-06-28T07:26:36
2023-06-28T07:30:48
2023-06-28T07:35:03
2023-06-28T07:39:21
2023-06-28T07:43:34
2023-06-28T07:47:48
2023-06-28T07:47:48
2023-06-28T07:51:58
2023-06-28T07:56:14
2023-06-28T08:00:34
2023-06-28T08:04:50
2023-06-28T08:08:55
2023-06-28T08:13:17
2023-06-28T08:17:30
2023-06-28T08:21:48
2023-06-28T08:26:01
2023-06-28T08:30:22
2023-06-28T08:34:38
2023-06-28T08:38:56
2023-06-28T08:43:12
2023-06-28T08:47:28
2023-06-28T08:51:40
2023-06-28T08:55:59
2023-06-28T09:00:14
2023-06-28T09:04:28
2023-06-28T09:08:36
2023-06-28T09:12:50
2023-06-28T09:17:08
2023-06-28T10:06:40
2023-06-28T16:31:04
2023-06-28T16:39:17
2023-06-28T16:43:37
2023-06-28T16:47:46
2023-06-28T16:52:03
2023-06-28T16:56:16
2023-06-28T17:00:22
2023-06-28T17:04:45
2023-06-28T17:08:51
2023-06-28T17:13:19
2023-06-28T17:17:21
2023-06-28T17:21:33
2023-06-28T17:25:43
2023-06-28T18:25:59
2023-06-28T17:34:08
2023-06-28T18:36:02
2023-06-28T17:42:21
2023-06-28T18:45:57
2023-06-28T17:50:45
2023-06-28T17:54:57
2023-06-28T19:01:04
2023-06-28T19:06:00
2023-06-28T19:10:55
2023-06-28T19:16:09
2023-06-28T19:20:54
2023-06-28T19:25:51
2023-06-28T19:30:55
2023-06-28T19:35:45
2023-06-28T19:40:48
2023-06-28T19:45:37
2023-06-28T18:40:29
2023-06-28T19:55:36
2023-06-28T12:51:11
2023-06-28T12:55:10
2023-06-28T13:01:04
2023-06-28T13:06:10
2023-06-28T13:07:27
2023-06-28T13:16:05
2023-06-28T13:21:01
2023-06-28T13:26:01
2023-06-28T13:31:03
2023-06-28T13:36:03
2023-06-28T13:41:15
2023-06-28T13:36:35
2023-06-28T13:40:47
2023-06-28T13:45:05
2023-06-28T13:49:19
2023-06-28T13:53:26
2023-06-28T13:57:40
2023-06-28T14:01:54

```

Thanks in advance!

johanmeijer, 2023-06-29:
> Hi @johanmeijer and @andersonshatch , Sorry for joining in an old topic like this, but just installed Grott few days ago (via docker) and I encounter the same (I guess) problem with the time, which [...]

This does not look to bad I think. By default the inverter will sent out data every 5 minutes. In your case it is between 4-5 minutes. If you want you can change the update frequency (by setting a register in the datalogger). But the  growatt server is not always able to keep track with higher update frequencies. If you want use that you might want to consider to use grottserver also.

pmp02, 2023-06-29:
Thanks for the reply, but I'm not sure I follow you :).
I already use grott server, this data si from the grott installation, in proxy mode. Grott is receiving the data from the datalogger and then it forwards it to Growatt servers.

Nevertheless, the problem is with the time from grott. If you look at the time series above, after some data sets, the time gets "reset" or it's miscalculated somehow...

I'll post below again with some highlights (sorry for the long post).
It looks like after 2023-06-28T11:42:26, none of the times are correct (I think).

2023-06-28T05:46:48
2023-06-28T05:51:00
2023-06-28T05:55:13
2023-06-28T05:59:28
2023-06-28T06:03:42
2023-06-28T06:07:55
2023-06-28T06:12:07
2023-06-28T06:16:22
2023-06-28T06:20:34
2023-06-28T06:24:41
2023-06-28T06:28:56
2023-06-28T06:33:11
2023-06-28T06:47:34
2023-06-28T06:41:29
2023-06-28T06:57:35
2023-06-28T07:02:35
2023-06-28T06:53:59
2023-06-28T07:12:51
2023-06-28T07:17:42
2023-06-28T07:22:34
2023-06-28T07:27:41
2023-06-28T07:32:30
2023-06-28T07:19:00
2023-06-28T07:23:09
2023-06-28T07:47:37
2023-06-28T07:31:30
2023-06-28T07:57:34
**2023-06-28T07:39:52
2023-06-28T08:07:36
2023-06-28T07:48:18**
2023-06-28T07:52:27
2023-06-28T07:56:34
2023-06-28T08:00:51
2023-06-28T08:05:07
2023-06-28T08:09:22
2023-06-28T08:13:36
2023-06-28T08:17:49
2023-06-28T08:17:50
2023-06-28T08:21:58
2023-06-28T08:26:12
2023-06-28T08:30:28
2023-06-28T08:34:41
2023-06-28T08:38:58
2023-06-28T08:43:05
2023-06-28T08:47:23
2023-06-28T08:51:35
2023-06-28T08:55:43
2023-06-28T09:38:43
2023-06-28T09:43:44
2023-06-28T09:48:46
2023-06-28T09:53:47
2023-06-28T09:58:44
**2023-06-28T09:20:46
2023-06-28T10:08:45
2023-06-28T09:29:06
2023-06-28T10:18:46
2023-06-28T09:37:38
2023-06-28T10:28:51**
2023-06-28T10:33:46
2023-06-28T10:38:46
2023-06-28T10:43:48
2023-06-28T10:48:39
2023-06-28T10:53:40
2023-06-28T10:58:44
2023-06-28T11:03:40
2023-06-28T11:08:33
2023-06-28T11:13:39
2023-06-28T10:23:10
2023-06-28T11:23:37
2023-06-28T11:28:38
2023-06-28T11:33:42
2023-06-28T11:38:33
2023-06-28T10:43:56
2023-06-28T11:48:38
2023-06-28T11:53:37
2023-06-28T11:58:36
2023-06-28T11:00:39
2023-06-28T12:08:35
2023-06-28T12:13:37
2023-06-28T11:13:10
2023-06-28T12:23:39
2023-06-28T11:21:27
2023-06-28T12:33:34
2023-06-28T11:29:48
2023-06-28T11:34:00
**2023-06-28T11:38:12
2023-06-28T11:42:26
2023-06-28T05:48:45
2023-06-28T05:53:08**
2023-06-28T05:57:18
2023-06-28T06:01:36
2023-06-28T06:05:50
2023-06-28T06:05:51
2023-06-28T06:10:06
2023-06-28T06:14:27
2023-06-28T06:18:32
2023-06-28T06:22:48
2023-06-28T06:27:02
2023-06-28T06:31:15
2023-06-28T06:35:29
2023-06-28T06:39:43
2023-06-28T06:44:07
2023-06-28T06:48:17
2023-06-28T06:52:37
2023-06-28T06:56:46
2023-06-28T07:01:06
2023-06-28T07:05:19
2023-06-28T07:09:37
2023-06-28T07:13:47
2023-06-28T07:18:04
2023-06-28T07:22:21
2023-06-28T07:26:36
2023-06-28T07:30:48
2023-06-28T07:35:03
2023-06-28T07:39:21
2023-06-28T07:43:34
2023-06-28T07:47:48
2023-06-28T07:47:48
2023-06-28T07:51:58
2023-06-28T07:56:14
2023-06-28T08:00:34
2023-06-28T08:04:50
2023-06-28T08:08:55
2023-06-28T08:13:17
2023-06-28T08:17:30
2023-06-28T08:21:48
2023-06-28T08:26:01
2023-06-28T08:30:22
2023-06-28T08:34:38
2023-06-28T08:38:56
2023-06-28T08:43:12
2023-06-28T08:47:28
2023-06-28T08:51:40
2023-06-28T08:55:59
2023-06-28T09:00:14
2023-06-28T09:04:28
2023-06-28T09:08:36
**2023-06-28T09:12:50
2023-06-28T09:17:08
2023-06-28T10:06:40
2023-06-28T16:31:04
2023-06-28T16:39:17
2023-06-28T16:43:37
2023-06-28T16:47:46
2023-06-28T16:52:03
2023-06-28T16:56:16
2023-06-28T17:00:22
2023-06-28T17:04:45
2023-06-28T17:08:51**
2023-06-28T17:13:19
2023-06-28T17:17:21
2023-06-28T17:21:33
2023-06-28T17:25:43
2023-06-28T18:25:59
2023-06-28T17:34:08
2023-06-28T18:36:02
2023-06-28T17:42:21
2023-06-28T18:45:57
2023-06-28T17:50:45
2023-06-28T17:54:57
2023-06-28T19:01:04
2023-06-28T19:06:00
2023-06-28T19:10:55
2023-06-28T19:16:09
2023-06-28T19:20:54
2023-06-28T19:25:51
2023-06-28T19:30:55
2023-06-28T19:35:45
2023-06-28T19:40:48
2023-06-28T19:45:37
2023-06-28T18:40:29
2023-06-28T19:55:36
2023-06-28T12:51:11
2023-06-28T12:55:10
2023-06-28T13:01:04
2023-06-28T13:06:10
2023-06-28T13:07:27
2023-06-28T13:16:05
2023-06-28T13:21:01
2023-06-28T13:26:01
2023-06-28T13:31:03
2023-06-28T13:36:03
2023-06-28T13:41:15
2023-06-28T13:36:35
2023-06-28T13:40:47
2023-06-28T13:45:05
2023-06-28T13:49:19
2023-06-28T13:53:26
2023-06-28T13:57:40
2023-06-28T14:01:54

pmp02, 2023-06-30:
Thanks @johanmeijer for the quick response and for looking into this!

Didn't noticed the grottserver component, thanks, I'll look into that. Also I'll try to schedule an automatic restart of the container periodically and see if it has any positive effects.

Regarding the grott logs, this is a sample piece:

```
grott  |         - Growatt original Data:
grott  |                 \x00\x39\x00\x06\x01\x0e\x01\x20\x1f\x35\x2b\x41\x22\x33\x23\x75\x43\x2d\x77
grott  |                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott  |                 \x3d\x3c\x29\x4c\x37\x04\x33\x5f\x46\x26\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott  |                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x72\x6a\x4e\x48\x60\x77\xa9
grott  |                 \x74\x74\x47\xfc\x6f\x77\x68\x2e\x74\x47\x7b\x3b\x77\x61\x7d\x2a\x47\x72\x6f
grott  |                 \x35\x61\x74\x74\x03\x72\x6f\x77\x27\x8b\x8b\x83\x79\x90\x88\xa0\x6d\x8b\xb8
grott  |                 \xcc\xa5\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x8d\x90\x85\xf2\x8b
grott  |                 \x8b\xb0\xd5\x90\x88\x9b\x7d\x8b\xb8\x8e\x57\x88\x9e\x88\x57\xb8\x8d\x93\x69
grott  |                 \x9e\x8b\x37\xaa\x72\x6f\x77\x61\x8b\x8b\xa3\x31\x90\x88\x9d\x5c\x74\x47\x73
grott  |                 \x9c\x77\x61\x64\x59\x47\x72\x7f\x47\x61\x74\x64\x72\x72\x6f\x5d\x01\x74\x74
grott  |                 \xce\x7d\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
grott  |                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
grott  |                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott  |                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
grott  |                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x5a\x67
grott  |         - Grott automatic protocol detection
grott  |         - Grott data record length 278
grott  |         - layout   :  T060120
grott  |         - Record layout used :  T060120
grott  |         - Growatt data decrypted V2
grott  |         - Grott Growatt data decrypted
grott  |         - Growatt plain data:
grott  |                 00390006010e01205847443643475732314200000000000000000000000000000000000000005
grott  |                 24b4838434341303147000000000000000000000000000000000000000017061e093a0f00c800
grott  |                 00008e0000095a000009540000095e000000420000004400000046ffffc40bffffc119ffffbec
grott  |                 a000000000000000000000000fffff293fffff7a7fffffa09fffffc38fffffc23fffffc1effff
grott  |                 43ed00000000ffffe443fffffc28000001f30000102d000010300000103500002a600000890f0
grott  |                 00000000000000000000000000000000000000000000000000000000000000000000000000000
grott  |                 00000000000000000000000000000000000000000000000000000000000000000000000000000
grott  |                 00000000000003510
grott  |         - Growatt new layout processing
grott  |                 - decrypt       :  True
grott  |                 - offset        :  6
grott  |                 - record layout :  T060120
grott  | 
grott  |         - Grott data record date/time processing started
grott  |         - date-time:  2023-06-30T09:58:15
grott  |         - Grott values retrieved:
grott  |                 -  datalogserial        :  XGD6CGW21B
grott  |                 -  pvserial             :  RKH8CCA01G
grott  |                 -  voltage_l1           :  239.4
grott  |                 -  voltage_l2           :  238.8
grott  |                 -  voltage_l3           :  239.8
grott  |                 -  Current_l1           :  6.6
grott  |                 -  Current_l2           :  6.8
grott  |                 -  Current_l3           :  7.0
grott  |                 -  act_power_l1         :  -1534.9
grott  |                 -  act_power_l2         :  -1610.3
grott  |                 -  act_power_l3         :  -1669.4
grott  |                 -  app_power_l1         :  0.0
grott  |                 -  app_power_l2         :  0.0
grott  |                 -  app_power_l3         :  0.0
grott  |                 -  react_power_l1       :  -343.7
grott  |                 -  react_power_l2       :  -213.7
grott  |                 -  react_power_l3       :  -152.7
grott  |                 -  powerfactor_l1       :  -1.0
grott  |                 -  powerfactor_l2       :  -1.0
grott  |                 -  powerfactor_l3       :  -1.0
grott  |                 -  pos_rev_act_power    :  -4814.7
grott  |                 -  pos_act_power        :  -4814.7
grott  |                 -  rev_act_power        :  -4814.7
grott  |                 -  app_power            :  0.0
[... 15 lines cut ...]
grott  |                         "react_power_l2": -2137, "react_power_l3": -1527, "powerfactor_l1": -968,
grott  |                         "powerfactor_l2": -989, "powerfactor_l3": -994, "pos_rev_act_power": -48147,
grott  |                         "pos_act_power": -48147, "rev_act_power": -48147, "app_power": 0,
grott  |                         "react_power": -7101, "powerfactor": -984, "frequency": 499, "L1-2_voltage":
grott  |                         4141, "L2-3_voltage": 4144, "L3-1_voltage": 4149, "pos_act_energy": 10848,
grott  |                         "rev_act_energy": 35087}}
grott  |         - Grott MQTT topic used : energy/growatt
grott  |         - MQTT message message sent
[... cut ...]
grott  |         - Grott Send data to Influx disabled 
grott  |         - Grott extension processing started :  grotcsv

```

The logs look identical (except the actual values parsed) when the dates are correct and when not.
I've enabled the PVOutput as well and see how it goes, although in the web site I only get the Voltage (power and energy are always zero... but guess that's something else).

If it has any relevance, in the Growatt website, my inverter it's shown as MAX/MID/MAC and in Grott config, I've configured it as `invtype=max`.

johanmeijer, 2023-07-26:
For a MAX/MID/MAC I think you have to use invtype=tl3. 

This will not help with the time issues of course.

pmp02, 2023-07-27:
Actually, it seems it did :). Thanks for the hint, indeed I was using the wrong type on inverter in settings. Things look waaay better with the correct type (tl3).

Thanks @johanmeijer !

closed 2023-08-13 by johanmeijer