johanmeijer/grott #115: Time issue influxdb / Summertime

johanmeijer, 2022-03-28:
There seams to be some time issues with influxdb after summertime is set this weekend (thanks @anarki1337 for notice it). 
The time of the record sent to influxdb is one hour off the GMT/UTC time. 

I really do not have found the exact reason for this but stopping/starting Grott did repair it for me. 
Be aware I use the setting timezone=europe/amsterdam in the .ini (gtimezone environmental for docker). 

I might be necessary to stop/start Grott to reload new timezone information (with summertime information in it). 

If you do not use timezone parameter the Server time settings are used. This also seems to give problems.  At this moment,  I really do not know why.

So workaround: Specify timezone= (see list: https://en.wikipedia.org/wiki/List_of_tz_database_time_zones) and restart Grott. 

You should see something like this in the grott log: 

      - Grott original time :  2022-03-28T18:18:59 adjusted UTC time for influx :  2022-03-28T16:18:59
 
If you do not have timezone= specified you will see this: 

    - Timezone local specified default timezone used
    - Grott original time :  2022-03-28T18:15:14 adjusted UTC time for influx :  2022-03-28T17:15:14
 
Please let me know if you have the same issue and if this solve the issue. 

I am looking into a solution for setting the right time without specifying the timezone and of course to eliminate the restart for the next summer/winter time change (that will be  a challenge while I can not recreate the problem anymore).

johanmeijer, 2022-03-28:
Update: If you have timezone= is specified. A restart does not seems to be necessary, al seems to work ok!. 

Only when you have not specified timezone= (gtimezone =) you might encounter summertime issues.

rany2, 2022-06-06:
I have fixed this issue in https://github.com/rany2/grott-1/commit/4165adcff2b2b23000f5b65dc24ca15081ac39d8.