johanmeijer/grott #116: Portainer Stacks Setup

mjdyson, 2022-04-08:
Hello, I'm trying to re-setup Grott using Portainer - a docker manager GUI. It has a concept of Stacks, which work like Docker compose.

For whatever reason, I cannot seem to get the data into InfluxDB.

I wonder if anyone could point me towards an answer?

The Docker-Compose-like file I've been using attempts to include an image of grafana and influxDB. 

`version: '3.3'
services:
    grott:
        network_mode: host
        image: ledidobe/grott
        container_name: grott
        restart: unless-stopped
        ports:
            - "5279:5279"
        environment:
            - ginvtype=sph
            - gblockcmd=true 
            - gnomqtt=false
            - gmqttip=192.168.30.30
            - gmqttauth=false
            - gmqttuser=homeassistant
            - gmqttpassword=MySecurePassword
            - gpvoutput=false
            - gpvapikey=yourapikey 
            - gpvinverterid1
            - gpvsystemid1=12345
            - ginflux=true
            - ginflux2=true
            - gifip=localhost
            - gifport=8086
            - giftoken=MyToken
            - giforg=grott_org
            - gifbucket=grott_bucket
            - gextension=false
            - gextname=grottext
            - 'gextvar={"ip":"192.168.30.109","port":"8000"}'
            - TZ=Europe/London
    grafana:
      image: grafana/grafana-oss
      container_name: grafana
      restart: always
      ports:
        - 3000:3000
    influxdb:
      image: influxdb
      container_name: influxdb
      restart: always
      ports:
        - 8086:8086`

mjdyson, 2022-04-08:
The logs of the grott container show the following

```Grott Growatt logging monitor : 2.6.1g

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
	nomqtt:      	 False
	inverterid:  	 automatic
	pvoutput:    	 False
	blockcmd:    	 False
	noipf:       	 False

Grott process configuration file

Grott process environmental variables

	Pulled 'ginvtype=sph' from the environment

	Pulled 'gblockcmd=true' from the environment

	Pulled 'gnomqtt=false' from the environment

	Pulled 'gmqttip=192.168.30.30' from the environment

	Pulled 'gmqttauth=false' from the environment

	Pulled 'gmqttuser=homeassistant' from the environment

	Pulled 'gmqttpassword=xxxxx' from the environment

	Pulled 'gpvoutput=false' from the environment

	Pulled 'gpvapikey=yourapikey' from the environment

	Pulled 'gpvsystemid1=12345' from the environment

	Pulled 'ginflux=true' from the environment

	Pulled 'ginflux2=true' from the environment

Grott InfluxDB server IP address env invalid

	Pulled 'gifport=8086' from the environment

	Pulled 'giforg=grott_org' from the environment

	Pulled 'gifbucket=grott' from the environment

	Pulled 'giftoken=xxxxxx' from the environment

	Pulled 'gextension=false' from the environment

	Pulled 'gextname=grottext' from the environment

	Pulled 'gextvar={"ip":"192.168.30.109","port":"8000"}' from the environment

Grott override settings if set in commandline

Grott process json layout files

Grott layout records loaded
T02NNNN  : 

|{'decrypt': {'value': 'False'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 210, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'gfcif': {'value': 214, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'dcif': {'value': 218, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vpvfault': {'value': 222, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vacfault': {'value': 226, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'facfault': {'value': 230, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'tmpfault': {'value': 234, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'faultcode': {'value': 238, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 246, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 250, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}, 'rac': {'value': 318, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractoday': {'value': 326, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractotal': {'value': 334, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}}
T05NNNN  : 

|{'decrypt': {'value': 'True'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'n [... cut ...]
	pvapikey:    	 yourapikey
	pvinverters: 	 1
	pvsystemid:  	 12345
_Influxdb:
	influx:      	 True
	influx2:     	 True
	database:    	 grottdb
	ip:          	 0.0.0.0
	port:        	 8086
	user:        	 grott
	password:    	 **secret**
	organization:	 grott_org
	bucket:      	 grott
	token:       	 **secret**
_Extension:
	extension:   	 False
	extname:     	 grottext
	extvar:      	 {'ip': '192.168.30.109', 'port': '8000'}

Grott proxy mode started
Hostname : yocal
IP :  192.168.30.109 , port :  5279 
```

mjdyson, 2022-04-08:
Closed issue as it looks like a problem with the docker / network. I'll install natively.

closed 2022-04-08 by mjdyson

darac, 2023-10-08:
You probably don't want to be sending influx data to "localhost". From the point of view of docker, each container is its own host. You want to either send to the docker host or you can take advantage of docker's "internal networking" by setting the influx hostname to be the container name (i.e. "influxdb")