johanmeijer/grott #118: How to get the Lifetime storaged charged value?

gabrielpc1190, 2022-04-19:
I have an SPF 3000 TL LVM-24p.
Grott works really well and I have realtime data on my Home Assistant. I love it!

But I'm missing a value. 
The Growatt integration for Home Assistant provides a value called "Lifetime storaged charged"
I use this value to feed my Home Assistant Energy Dashboard, on the section that calculates how much energy goes from solar into the battery, is the only value that I use from Growatt Integration, because it's not available from Grott.
For how much energy comes out of the battery I'm using "ebatDischarTotal" and it's perfect.

How can I get this value? Since I think the inverter must be delivering it to Growatt servers, it has to go trought grott first.
I already checked all the values on [this file ](https://github.com/johanmeijer/grott/blob/master/examples/Record%20Layout/t050104spf.json)and none of those is the one I'm missing.

johanmeijer, 2022-04-20:
I do not have all value expose yet for the SPF (on the todo list). But if look in the documentation I can not find it either. 

These are the fields I see in relation to charge/discharge: 

[image]

You can find al fields in this document (defined in input reg, starts at page 12). 
https://docplayer.net/199800433-Growatt-offgrid-spf5000-modbus-rs485-rtu-protocol.html 

Sometimes the growatt site calculated fields but to be honest I do not see how to calculate this either. I expected something like Bat charge total - AC charge Total or something like that. 

Maybe you see it. If I have to change (add fields to) the record layout please let me know.

johanmeijer, 2023-08-13:
Is this still a valid issue?