johanmeijer/grott #123: Help with 6000TL3-X

Schnittcher, 2022-06-05:
At first, thanks for this very nice project.

I have a 6000TL3-X.
I used the docker container. (Version 2.6.1g)

Grott settings:
```
_Generic:
version:       2.6.1g
verbose:       True
trace:         False
config file:   grott.ini
minrecl:       100
decrypt:       True
compat:        False
invtype:       tl3
include_all:   False
blockcmd:      False
noipf:         False
time:          auto
sendbuf:       True
timezone:      local
valueoffset:   6
offset:        6
inverterid:    automatic
mode:          proxy
grottip        default
grottport      5279
```

I get a lot of messages like this:
Growatt packet received:
```
<socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.7', 5279), raddr=('172.17.0.1', 47780)>
Data less then minimum record length, data not processed
```

But after a while, I get this:
```
Grott values retrieved:
datalogserial        :  XGD5BLB2QJ
pvserial             :  RFH3BHU05Y
pvstatus             :  1
pvpowerin            :  236.8
pv1voltage           :  249.4
pv1current           :  0.4
pv1watt              :  99.7
pv2voltage           :  252.3
pv2current           :  0.3
pv2watt              :  75.6
pvpowerout           :  233.3
pvfrequentie         :  50.0
pvgridvoltage        :  234.4
pvgridcurrent        :  0.6
pvgridpower          :  140.6
pvgridvoltage2       :  233.8
pvgridcurrent2       :  0.6
pvgridpower2         :  140.2
pvgridvoltage3       :  232.4
pvgridcurrent3       :  0.6
pvgridpower3         :  139.4
Vac_RS               :  405.5
Vac_ST               :  404.1
Vac_TR               :  403.7
pvenergytoday        :  14.7
pvenergytotal        :  15.7
totworktime          :  16.6
epv1today            :  8.0
epv1total            :  8.4
epv2today            :  6.8
epv2total            :  7.2
epvtotal             :  15.6
pvtemperature        :  59.8
pvipmtemperature     :  33.8
pvboottemperature    :  32.8
uwBatVolt_DSP        :  38.0
pbusvolt             :  3126
nbusvolt             :  3110
```

And after this message  I get this again:
```
Growatt packet received:
<socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.7', 5279), raddr=('172.17.0.1', 47780)>
Data less then minimum record length, data not processed
```
Is this correct? Or are my settings wrong?

Thanks!

regards,
Kai

rany2, 2022-06-05:
Your settings are correct, this is to be expected. By default grott doesn't process messages under 100 bytes because they are usually ping requests. In all cases the data gets forwarded to Growatt servers so they could respond to them

Schnittcher, 2022-06-05:
Thanks for the fast answer.
I have still one question. Can I set any settings via grott?
For example register 03?

rany2, 2022-06-05:
@Schnittcher You could not in stable but this is being planned with grottserver. The goal is that it will be integrated into grott itself eventually. 

You could use grottserver in the 2.7(Beta) branch and have grott use that as the 'Growatt server'. Grottserver includes an HTTP server that allows you to set or get any register you want. For example:

```
$ # To get a value
$ curl 'http://localhost:5782/inverter?inverter=xxxx&command=register&register=1'
{"value": 3}
$ # To set a value
$ curl 'http://localhost:5782/inverter?inverter=xxxx&command=register&register=1&value=0' -XPUT
OK
```

rany2, 2022-06-05:
If you need the Growatt application to continue providing information (without allowing you to set anything), you could use the version in my pull request. https://github.com/johanmeijer/grott/pull/122

Schnittcher, 2022-06-05:
So I can´t set anything with grott?

rany2, 2022-06-05:
No, with the stable branch all it does is collect data to publish to MQTT, PVOutput, or InfluxDB and allow you to block commands.

Schnittcher, 2022-06-05:
Very nice with the webserver.
I will try this! Thanks!

Schnittcher, 2022-06-05:
@rany2, is it possible to set the data update interval to 1 minute or smaller?

rany2, 2022-06-05:
@Schnittcher Yep, change register 4 for the datalogger to the minute value you want (it accepts decimal too). The lowest I managed to get it was 0.1, anything below that didn't work.

```
$ curl 'http://localhost:5782/datalogger?command=register&datalogger=Dxxxxx&register=4&value=0.1' -XPUT
```

Schnittcher, 2022-06-05:
Thanks. Do you have a List for alle register and values?

rany2, 2022-06-05:
I have some idea of them:

* Register 8: serial number
* Register 17: growatt server IP
* Register 18: growatt server port
* Register 21: firmware version
* Register 31: datalogger time
* Register 56: wifi SSID
* Register 57: wifi pwd
* Register 76: wifi signal strength in dBm
* Register 81: firmware URL

rany2, 2022-06-05:
@Schnittcher I SHOULD HAVE WARNED YOU, datalogger and inverter register is different. What I sent above is for datalogger only which isn't well documented and is probably different between models!!!

Schnittcher, 2022-06-05:
I only search the register for the update interval. I try it via server.growatt.com.
But I get "Read Param Failure".

rany2, 2022-06-05:
Did you block commands in Grott?

johanmeijer, 2022-06-05:
@Schnittcher if you do want to use the grottserver yet (I am still working on it) , it is also possible to set the datalogger register 4 via the Growatt website. 

Be aware if you set the interval to fast you will get problems with pvoutput allowing only allowing 60 api calls per minute (including your queries for looking at it). If you donate at the pvoutput site, 120 API's calls are possible. 

I will update the wiki (but developing/improving grott/grottserver has a higher priority) with the registers (both Inverter and datapower). @rany2 and others please keep on publishing the values you found, that will help me with the documentation. 

There is no risk at using the 2.7.2 (beta) docker version. This version is alo rather stable but does not contain the grottserver yet. At some point in time I will publish version 3.x with everything integrated.

Schnittcher, 2022-06-05:
No. This option is false.

rany2, 2022-06-05:
@Schnittcher You should try it multiple times. Sometimes the request gets lost

Schnittcher, 2022-06-05:
I only want the data via MQTT.

Schnittcher, 2022-06-05:
The growatt is now offline. So I will try it tomorrow.

johanmeijer, 2022-06-05:
> 

My experience is that the datalogger setting works most of the time. But  you really have to set the datalogger register and not the inverter. That is a totally seperate menu at the growatt server website (and also in grottserver)

Schnittcher, 2022-06-06:
So now ist works. Thanks. :)
I will also test the beta version soon.
Now I have another question, is it possible to set the logger to standby at night?

johanmeijer, 2022-06-06:
At this moment that is not possible. Why do you want to do that?

You can stop Gtott but the logger will buffer the datarecords and send them at the moment the connection is established.

Grott can ignore the buffered records ( sendbuf = False in the grott.ini or -e gsendbuf="false" for docker).

Schnittcher, 2022-06-08:
I want it for tests. So that I can also test with grott in the evening
But that is not important.

Today the cloud (server.growatt.com) was offline and I didn´t get any data via grott.
And and the data from the morning has been lost. Is this a config problem?

rany2, 2022-06-08:
@Schnittcher No it's to be expected because stable grott relies on Growatt servers for communication. This is no longer required in beta because you could run your own "server.growatt.com" server

Schnittcher, 2022-06-08:
Thanks. Then I will install the beta today.
And this is why  I want the inverter in standby. To test this at evening.

johanmeijer, 2022-06-08:
> I want it for tests. So that I can also test with grott in the evening But that is not important.
> 
> Today the cloud (server.growatt.com) was offline and I didn´t get any data via grott. And and the data from the morning has been lost. Is this a config problem?

That happens not that often that the growatt server is not available (at least not in the years I use it, from the Netherlands). 

You should not loose data. Normally the dataloger buffers the records and will sent them the moment the connection is established again. 

Grott ignores (does not proces) the buffered records if 

1. you specify sendbuf = False in the grott.ini
2. The buffered record doesn't contain or have corrupted time/date information (seems to happen with some dataloggers). 

These records are always sent to the growatt server (probably these can also do nothing with missing or corrupted date/time info). 

The grott log will give more information on the processing of the buffered records. 

So if you still have the log of this morning we can look at it.

Schnittcher, 2022-06-13:
Now I installed the beta version.
Is there a guide for the config?

johanmeijer, 2022-06-13:
Are you sure the inverter data is already processed by Grott? If the there has no been data sent to Grott Grott will not recognize the inverter. 

You can see more information with: 
http://nasip:5782/inverter or http://nasip:5782/datalogger

Most of the configuration is described at the wiki: 
     https://github.com/johanmeijer/grott/wiki
(and in the exampes/grott.ini)

In principle you do not have to configure a lot. if u use MQTT: 

    docker run -d -p 5279:5279 -p 5782:5782 --name grott -e gmqttip="your.ip.address.number" ginvtype="tl3"  ledidobe/grottserver:beta 

(the thing that might not described will is the ginvtype parm for your inverter (see readme for more info on that one). 

I would also add a timezone environmental: 

    docker run -d -p 5279:5279 -p 5782:5782 --name grott -e gmqttip="your.ip.address.number" ginvtype="tl3"  -e TZ=Europe/Amsterdam ledidobe/grottserver:beta 

I you want to use influxdb and/or pvoutput you have to set environmentals for that as well. 

For pvoutput: 

    docker run -d -p 5279:5279 -p 5782:5782 --name grott -e gmqttip="your.ip.address.number" ginvtype="tl3"   -e gpvoutput="True" -e gpvapikey="apikey" -e gpvsystemid="systemid" -e TZ=Europe/Amsterdam ledidobe/grottserver:beta 

But everything starts with routing the data to Grott!

rany2, 2022-06-13:
Minor note, a `-e` is missing in the docker command for the `ginvtype` setting.

```diff

Schnittcher, 2022-06-14:
> Are you sure the inverter data is already processed by Grott? If the there has no been data sent to Grott Grott will not recognize the inverter.

Yes  I see the data in the docker log on synology.
The url http://NASIP:5782/inverter is very slow an  I get only `{}`

johanmeijer, 2022-06-15:
I do not see any strange in your grott.ini (in can be must simplex while most is default, but that does not  change the working).

I really need the log to see what is going wrong. Can you provide more information?

closed 2023-08-13 by johanmeijer