johanmeijer/grott #126: Gr

berndh1979, 2022-06-17:
Nice work on this grott proxy. Really appreciate the effort put into this. 
There is one issue that I have. It seems that new records are being send, but the record keep having the same time stamp. I read that this timestamp is set by the datalogger/inverter, but the data that is send to the growatt server is not showing this same timestamp. They nicely show a parabole over the full day.  

All the date/time related timestamp in the logging below are identical. The next record that is send by the datalogger again has this same timestamp of 2022-07-17T08:24:42. 

is there an issue with my setup?

Jun 17 12:36:12 pi grott[15539]: #011 - Growatt new layout processing
Jun 17 12:36:12 pi grott[15539]: #011#011 - decrypt       :  True
Jun 17 12:36:12 pi grott[15539]: #011#011 - offset        :  6
Jun 17 12:36:12 pi grott[15539]: #011#011 - record layout :  T06NNNNX
Jun 17 12:36:12 pi grott[15539]: #011 - Grott data record date/time processing started
Jun 17 12:36:12 pi grott[15539]: #011 - date-time:  **2022-06-17T08:24:42**
Jun 17 12:36:12 pi grott[15539]: #011 - Grott values retrieved:
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvserial             :  AEJ0BGA00N
Jun 17 12:36:12 pi grott[15539]: #011#011 -  recortype1           :  3000
Jun 17 12:36:12 pi grott[15539]: #011#011 -  recortype2           :  3124
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvstatus             :  1
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvpowerin            :  4213.6
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pv1voltage           :  169.3
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pv1current           :  11.8
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pv1watt              :  2002.6
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pv2voltage           :  195.2
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pv2current           :  11.3
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pv2watt              :  2211.0
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvpowerout           :  3970.8
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvfrequentie         :  50.0
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridvoltage        :  232.2
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridcurrent        :  17.2
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridpower          :  3977.3
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridvoltage2       :  0.0
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridcurrent2       :  0.0
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridpower2         :  0.0
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridvoltage3       :  0.0
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridcurrent3       :  0.0
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvgridpower3         :  0.0
Jun 17 12:36:12 pi grott[15539]: #011#011 -  totworktime          :  512.1
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvenergytoday        :  11.4
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvenergytotal        :  725.8
Jun 17 12:36:12 pi grott[15539]: #011#011 -  epvtotal             :  774.5
Jun 17 12:36:12 pi grott[15539]: #011#011 -  epv1today            :  5.7
Jun 17 12:36:12 pi grott[15539]: #011#011 -  epv1total            :  362.6
Jun 17 12:36:12 pi grott[15539]: #011#011 -  epv2today            :  6.4
Jun 17 12:36:12 pi grott[15539]: #011#011 -  epv2total            :  411.9
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvtemperature        :  61.8
Jun 17 12:36:12 pi grott[15539]: #011#011 -  pvipmtemperature     :  54.0
Jun 17 12:36:12 pi grott[15539]: #011 - MQTT jsonmsg:
Jun 17 12:36:12 pi grott[15539]: #011#011#011 {"device": "AEJ0BGA00N", "time": "**2022-06-17T08:24:42**", "buffered": "no",
Jun 17 12:36:12 pi grott[15539]: #011#011#011 "values": {"recortype1": 3000, "recortype2": 3124, "pvstatus": 1,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 "pvpowerin": 42136, "pv1voltage": 1693, "pv1current": 118, "pv1watt": 20026,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 "pv2voltage": 1952, "pv2current": 113, "pv2watt": 22110, "pvpowerout":
Jun 17 12:36:12 pi grott[15539]: #011#011#011 39708, "pvfrequentie": 4998, "pvgridvoltage": 2322, "pvgridcurrent": 172,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 "pvgridpower": 39773, "pvgridvoltage2": 0, "pvgridcurrent2": 0,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3":
Jun 17 12:36:12 pi grott[15539]: #011#011#011 0, "totworktime": 3687087, "pvenergytoday": 114, "pvenergytotal": 7258,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 "epvtotal": 7745, "epv1today": 57, "epv1total": 3626, "epv2today": 64,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 "epv2total": 4119, "pvtemperature": 618, "pvipmtemperature": 540}}
Jun 17 12:36:12 pi grott[15539]: #011 - MQTT message message sent
Jun 17 12:36:12 pi grott[15539]: #011 - Grott Send data to PVOutput disabled
Jun 17 12:36:12 pi grott[15539]: #011 - Grott InfluxDB publihing started
Jun 17 12:36:12 pi grott[15539]: #011 - **Grott original time :  2022-06-17T08:24:42 adjusted UTC time for influx :  2022-06-17T06:24:42**
Jun 17 12:36:12 pi grott[15539]: #011 - Grott influxdb jsonmsg:
Jun 17 12:36:12 pi grott[15539]: #011#011#011 [{'measurement': 'AEJ0BGA00N', 'time': '**2022-06-17T06:24:42**', 'fields':
Jun 17 12:36:12 pi grott[15539]: #011#011#011 {'pvserial': 'AEJ0BGA00N', 'recortype1': 3000, 'recortype2': 3124,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 'pvstatus': 1, 'pvpowerin': 42136, 'pv1voltage': 1693, 'pv1current': 118,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 'pv1watt': 20026, 'pv2voltage': 1952, 'pv2current': 113, 'pv2watt': 22110,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 'pvpowerout': 39708, 'pvfrequentie': 4998, 'pvgridvoltage': 2322,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 'pvgridcurrent': 172, 'pvgridpower': 39773, 'pvgridvoltage2': 0,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 'pvgridcurrent2': 0, 'pvgridpower2': 0, 'pvgridvoltage3': 0,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 'pvgridcurrent3': 0, 'pvgridpower3': 0, 'totworktime': 3687087,
Jun 17 12:36:12 pi grott[15539]: #011#011#011 'pvenergytoday': 114, 'pvenergytotal': 7258, 'epvtotal': 7745, 'epv1today':
Jun 17 12:36:12 pi grott[15539]: #011#011#011 57, 'epv1total': 3626, 'epv2today': 64, 'epv2total': 4119, 'pvtemperature':
Jun 17 12:36:12 pi grott[15539]: #011#011#011 618, 'pvipmtemperature': 540}}]
Jun 17 12:36:12 pi grott[15539]: #011 - Grott write to influxdb v1
Jun 17 12:36:12 pi grott[15539]: #011 - Grott extension processing disabled
Jun 17 12:36:12 pi grott[15539]: #011 - Growatt packet received:
Jun 17 12:36:12 pi grott[15539]: #011#011  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 5279), raddr=('10.0.0.6', 17431)>
Jun 17 12:36:12 pi grott[15539]: #011 - Data less then minimum record length, data not processed
Jun 17 12:38:18 pi grott[15539]: #011 - Growatt packet received:
Jun 17 12:38:18 pi grott[15539]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 38530), raddr=('47.91.67.66', 5279)>
Jun 17 12:38:18 pi grott[15539]: #011 - Data less then minimum record length, data not processed
Jun 17 12:38:18 pi grott[15539]: #011 - Growatt packet received:
Jun 17 12:38:18 pi grott[15539]: #011#011  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 5279), raddr=('10.0.0.6', 17431)>
Jun 17 12:38:18 pi grott[15539]: #011 - Data less then minimum record length, data not processed

berndh1979, 2022-06-17:
Seems that after a restart of the grott service it recovered...

Jun 17 12:41:13 pi grott[15539]: #011#011#011 {"device": "AEJ0BGA00N", "time": "2022-06-17T12:35:40", "buffered": "no",
Jun 17 12:41:13 pi grott[15539]: #011#011#011 "values": {"recortype1": 3000, "recortype2": 3124, "pvstatus": 1,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 "pvpowerin": 40100, "pv1voltage": 1682, "pv1current": 112, "pv1watt": 19033,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 "pv2voltage": 1907, "pv2current": 109, "pv2watt": 21067, "pvpowerout":
Jun 17 12:41:13 pi grott[15539]: #011#011#011 37783, "pvfrequentie": 4999, "pvgridvoltage": 2350, "pvgridcurrent": 160,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 "pvgridpower": 37815, "pvgridvoltage2": 0, "pvgridcurrent2": 0,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3":
Jun 17 12:41:13 pi grott[15539]: #011#011#011 0, "totworktime": 3687757, "pvenergytoday": 118, "pvenergytotal": 7262,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 "epvtotal": 7750, "epv1today": 59, "epv1total": 3628, "epv2today": 67,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 "epv2total": 4122, "pvtemperature": 623, "pvipmtemperature": 545}}
Jun 17 12:41:13 pi grott[15539]: #011 - MQTT message message sent
Jun 17 12:41:13 pi grott[15539]: #011 - Grott Send data to PVOutput disabled
Jun 17 12:41:13 pi grott[15539]: #011 - Grott InfluxDB publihing started
Jun 17 12:41:13 pi grott[15539]: #011 - Grott original time :  2022-06-17T12:35:40 adjusted UTC time for influx :  2022-06-17T10:35:40
Jun 17 12:41:13 pi grott[15539]: #011 - Grott influxdb jsonmsg:
Jun 17 12:41:13 pi grott[15539]: #011#011#011 [{'measurement': 'AEJ0BGA00N', 'time': '2022-06-17T10:35:40', 'fields':
Jun 17 12:41:13 pi grott[15539]: #011#011#011 {'pvserial': 'AEJ0BGA00N', 'recortype1': 3000, 'recortype2': 3124,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 'pvstatus': 1, 'pvpowerin': 40100, 'pv1voltage': 1682, 'pv1current': 112,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 'pv1watt': 19033, 'pv2voltage': 1907, 'pv2current': 109, 'pv2watt': 21067,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 'pvpowerout': 37783, 'pvfrequentie': 4999, 'pvgridvoltage': 2350,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 'pvgridcurrent': 160, 'pvgridpower': 37815, 'pvgridvoltage2': 0,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 'pvgridcurrent2': 0, 'pvgridpower2': 0, 'pvgridvoltage3': 0,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 'pvgridcurrent3': 0, 'pvgridpower3': 0, 'totworktime': 3687757,
Jun 17 12:41:13 pi grott[15539]: #011#011#011 'pvenergytoday': 118, 'pvenergytotal': 7262, 'epvtotal': 7750, 'epv1today':
Jun 17 12:41:13 pi grott[15539]: #011#011#011 59, 'epv1total': 3628, 'epv2today': 67, 'epv2total': 4122, 'pvtemperature':
Jun 17 12:41:13 pi grott[15539]: #011#011#011 623, 'pvipmtemperature': 545}}]
Jun 17 12:41:13 pi grott[15539]: #011 - Grott write to influxdb v1

berndh1979, 2022-06-17:
Here you can see that 5min later the record values gets updated but the time stamp is still from previous record sample

**Jun 17 12:46:13** pi grott[15539]: #011 - Grott values retrieved:
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvserial             :  AEJ0BGA00N
Jun 17 12:46:13 pi grott[15539]: #011#011 -  recortype1           :  3000
Jun 17 12:46:13 pi grott[15539]: #011#011 -  recortype2           :  3124
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvstatus             :  1
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvpowerin            :  4202.4
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pv1voltage           :  166.6
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pv1current           :  11.9
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pv1watt              :  1994.8
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pv2voltage           :  194.2
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pv2current           :  11.3
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pv2watt              :  2207.6
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvpowerout           :  3956.2
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvfrequentie         :  50.0
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridvoltage        :  234.7
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridcurrent        :  16.9
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridpower          :  3962.2
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridvoltage2       :  0.0
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridcurrent2       :  0.0
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridpower2         :  0.0
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridvoltage3       :  0.0
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridcurrent3       :  0.0
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvgridpower3         :  0.0
Jun 17 12:46:13 pi grott[15539]: #011#011 -  totworktime          :  512.3
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvenergytoday        :  12.1
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvenergytotal        :  726.5
Jun 17 12:46:13 pi grott[15539]: #011#011 -  epvtotal             :  775.3
Jun 17 12:46:13 pi grott[15539]: #011#011 -  **epv1today            :  6.1**
Jun 17 12:46:13 pi grott[15539]: #011#011 -  epv1total            :  363.0
Jun 17 12:46:13 pi grott[15539]: #011#011 -  epv2today            :  6.8
Jun 17 12:46:13 pi grott[15539]: #011#011 -  epv2total            :  412.3
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvtemperature        :  62.7
Jun 17 12:46:13 pi grott[15539]: #011#011 -  pvipmtemperature     :  55.0
Jun 17 12:46:13 pi grott[15539]: #011 - MQTT jsonmsg:
Jun 17 12:46:13 pi grott[15539]: #011#011#011 {"device": "AEJ0BGA00N", "time": "**2022-06-17T12:35:40**", "buffered": "no",
Jun 17 12:46:13 pi grott[15539]: #011#011#011 "values": {"recortype1": 3000, "recortype2": 3124, "pvstatus": 1,
Jun 17 12:46:13 pi grott[15539]: #011#011#011 "pvpowerin": 42024, "pv1voltage": 1666, "pv1current": 119, "pv1watt": 19948,
Jun 17 12:46:13 pi grott[15539]: #011#011#011 "pv2voltage": 1942, "pv2current": 113, "pv2watt": 22076, "pvpowerout":
Jun 17 12:46:13 pi grott[15539]: #011#011#011 39562, "pvfrequentie": 4998, "pvgridvoltage": 2347, "pvgridcurrent": 169,
Jun 17 12:46:13 pi grott[15539]: #011#011#011 "pvgridpower": 39622, "pvgridvoltage2": 0, "pvgridcurrent2": 0,
Jun 17 12:46:13 pi grott[15539]: #011#011#011 "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3":
Jun 17 12:46:13 pi grott[15539]: #011#011#011 0, "totworktime": 3688353, "pvenergytoday": 121, "pvenergytotal": 7265,
Jun 17 12:46:13 pi grott[15539]: #011#011#011 "epvtotal": 7753, "epv1today": 61, "epv1total": 3630, "epv2today": 68,
Jun 17 12:46:13 pi grott[15539]: #011#011#011 "epv2total": 4123, "pvtemperature": 627, "pvipmtemperature": 550}}
Jun 17 12:46:13 pi grott[15539]: #011 - MQTT message message sent
Jun 17 12:46:13 pi grott[15539]: #011 - Grott Send data to PVOutput disabled
Jun 17 12:46:13 pi grott[15539]: #011 - Grott InfluxDB publihing started
Jun 17 12:46:13 pi grott[15539]: #011 - Grott original time :  2022-06-17T12:35:40 adjusted UTC time for influx :  2022-06-17T10:35:40
Jun 17 12:46:13 pi grott[15539]: #011 - Grott influxdb jsonmsg:

berndh1979, 2022-06-17:
> 

I'm using a Shine-X module connected to a MIN-4600TL-XH inverter. 
I see that when I restart grott service the timestamp gets updated from the record send by the data logger. It seems that the synchronization phase during the startup somehow triggers the datalogger to update the time stamp.

berndh1979, 2022-06-17:
btw, I also get these messages int he log file. Are these normal ACK/OK communcation going back-forth between grott and server.growatt and data logger?

Jun 17 14:11:01 pi grott[16394]: #011#011  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 5279), raddr=('10.0.0.6', 27986)>
Jun 17 14:11:01 pi grott[16394]: #011 - Data less then minimum record length, data not processed
Jun 17 14:11:01 pi grott[16394]: #011 - Growatt packet received:
Jun 17 14:11:01 pi grott[16394]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 38534), raddr=('47.91.67.66', 5279)>
Jun 17 14:11:01 pi grott[16394]: #011 - Data less then minimum record length, data not processed
Jun 17 14:11:02 pi grott[16394]: #011 - Growatt packet received:
Jun 17 14:11:02 pi grott[16394]: #011#011  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 5279), raddr=('10.0.0.6', 27986)>
Jun 17 14:11:02 pi grott[16394]: #011 - Data less then minimum record length, data not processed
Jun 17 14:11:02 pi grott[16394]: #011 - Growatt packet received:
Jun 17 14:11:02 pi grott[16394]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 38534), raddr=('47.91.67.66', 5279)>
Jun 17 14:11:02 pi grott[16394]: #011 - Data less then minimum record length, data not processed
Jun 17 14:11:05 pi grott[16394]: #011 - Growatt packet received:
Jun 17 14:11:05 pi grott[16394]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 38534), raddr=('47.91.67.66', 5279)>
Jun 17 14:11:05 pi grott[16394]: #011 - Data less then minimum record length, data not processed
Jun 17 14:13:41 pi grott[16394]: #011 - Growatt packet received:
Jun 17 14:13:41 pi grott[16394]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 38534), raddr=('47.91.67.66', 5279)>
Jun 17 14:13:41 pi grott[16394]: #011 - Data less then minimum record length, data not processed
Jun 17 14:13:41 pi grott[16394]: #011 - Growatt packet received:
Jun 17 14:13:41 pi grott[16394]: #011#011  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.253', 5279), raddr=('10.0.0.6', 27986)>
Jun 17 14:13:41 pi grott[16394]: #011 - Data less then minimum record length, data not processed

johanmeijer, 2022-06-17:
> 

Yes that is true during initialization Phase a date/time update command is sent from growatt server to the datalogger. 

You can also update the date/time at the data logger settings menu (advanced settings, read/set register 31)

johanmeijer, 2022-06-17:
> btw, I also get these messages int he log file. Are these normal ACK/OK communcation going back-forth between grott and server.growatt and data logger?
>
> Jun 17 14:11:01 pi grott[16394]: #11#011 [...]

Yes this is normal behavior. Grott suppresses messages that has a length <100 (default setting, can be changed in grott.ini). These messages are mostly ack/nack (ping) and config records (like the set time) but not of any interest for processing the data.

johanmeijer, 2022-06-17:
An firmware upgrade can also be performed from the data logger settings menu (I am no sure this will help).

berndh1979, 2022-06-17:
> An firmware upgrade can also be performed from the data logger settings menu (I am no sure this will help).
> 
> Maybe resetting your devices can help (as we say in Dutch maybe there a crooked bit somewhere).

Indeed. I did an upgrade of the datalogger (3.1.0.2 -> 3.1.0.5). Let's see if that helps.

berndh1979, 2022-06-17:
> Yes this is normal behavior. Grott suppresses messages that has a length <100 (default setting, can be changed in grott.ini). These messages are mostly ack/nack (ping) and config records (like the [...]

Thanks. confirms my thinking.

berndh1979, 2022-06-17:
> > An firmware upgrade can also be performed from the data logger settings menu (I am no sure this will help).
> > Maybe resetting your devices can help (as we say in Dutch maybe there a crooked bit somewhere).
> 
> Indeed. I did an upgrade of the datalogger (3.1.0.2 -> 3.1.0.5). Let's see if that helps.

Now it seems to work for the 3 first records that were processed. Time of record matches the actual time :)

berndh1979, 2022-06-17:
**Jun 17 15:49:30** pi grott[16394]: #011#011#011 {"device": "AEJ0BGA00N", "time": "**2022-06-17T15:47:56**", "buffered": "no",

I see that there is a growing delta between the timestamps. Is there a way that I can sync the time on the ShineX with the ntp server I have running within my network?

rany2, 2022-06-17:
@berndh1979 There is no way you could do such a thing. The way this dongle deals with time is terrible. It basically sends a command to set the time to a current date and there is no way to have it correct itself on its own (has no knowledge of precision).

Also it doesn't have any knowledge of timezones and keeps track of your local time (doesn't store UTC offset separately or have a tzinfo), so when the timezone changes (DST, change of locations, etc) it will go nuts for a bit.

berndh1979, 2022-06-17:
> The way this dongle deals with time is terrible

Hmm this dongle is EPS8266 based right? There should be a possibility to have time-sync, but it then probably needs a firmware flash. On the other hand, it is also not a big of a problem if the device is a bit off. Solar is not a real-time based device anyway :)

johanmeijer, 2022-06-17:
I do not think it would be easy to change the ESP program. It is a compiled bin file and we do not have the source code I am afraid. 

I am not sure why there are that many time problems with (some) dataloggers (maybe a bad batch). Probably there is a  additional RTC module (next to the ESP) that is faulty. 

What you can do is: 

1.  use time=sever parameter of grott as stated above 
The actual records will always have the right (server) time.  Your server (linux, docker container) is syncing with NTP I think
I think losing buffered data is not that of a big problem, I only have buffered data when I am playing (sorry testing) with Grott
2. Stop / start Grott on regular basis (automate with cron?)
3. Update time via the growatt site 
Via the menus (manual) but there is an API available that can gives command (I never have looked at that, but we try to find more information)
4. Use to grottserver emulating the growatt server
Grottserver provides an API that can give datalogger and inverter commands and allow setting time (you can script this and schedule with cron I think). 
Stop /starting Grottserver will also sent a time command to the datalogger (see 1)

At this moment if you use grottserver no data is being sent to the growatt server any more (that is the objective of emulating the grottserver to bypass the china servers). But it possible (maybe in one of next versions of grottserver, I am still doubting  about that) to also sent the data to the growatt server. 

I can also think about adding some code in grott (or grottserver) server in the future to sent a time update at regular basis to bypass the datalogger time problems.  

In your case I would go for option one for now!

rany2, 2022-06-17:
> But it possible (maybe in one of next versions of grottserver, I am still doubting about that) to also sent the data to the growatt server.

In my fork I have been able to send data to Growatt server (in read-only mode) by only sending ping records and data records. It works flawlessly.

johanmeijer, 2022-06-17:
Noticed.

johanmeijer, 2022-06-19:
@berndh1979 can we close this one.  I know you still have the drifting problem but that is something grott can not resolve You can only using the workarounds as described above to make it a little better I think.

berndh1979, 2022-06-19:
Sure. Please close this topic. The firmware update did indeed fix the issue. ThanksVerzonden vanaf mijn Galaxy

closed 2022-06-19 by johanmeijer