johanmeijer/grott #127: Question regarding ShineWifi communication

rany2, 2022-06-17:
It appears that my dongle creates multiple connections instead of just one connection for communication. The issue is that this situation is making it much harder for me to know if I should close them or not.

Ever since I added a 5 minute timeout (I know I could have made it 3min because that's the ping interval) on recv for sockets, I have had a much worse time having consistent information flow with the datalogger. Does anyone have any tips on how I could associate these multiple connections with a single datalogger so I don't close them erroneously and destroy all communication?

rany2, 2022-06-17:
Something to note is that these connections appear to end up doing absolutely nothing. They just connect and don't send/recv anything after the fact which is why it triggers the recv timeout.

rany2, 2022-06-17:
The reason why I am trying to close these connections is so that I don't end up reaching the maximum number of open connection. On Linux there is a limit to the amount of open file handlers you could have and with the current status quo I am running out after a few days.

rany2, 2022-06-17:
Is it possible for the server to initiate a ping request? I would like to ping the datalogger to see if it is still connected or not after I notice it is inactive for some time.

johanmeijer, 2022-06-17:
There is always a ping (ack/nack) communication going between datalogger and server (initiated by the datalogger). It should be possible to detect if there is no communication flow anymore.

The ping is not sent if there is a frequent data communication flow  (low interval time) I believe. But if there  is no communication flow for a longer time you can assume the connection is not used anymore.

rany2, 2022-06-17:
Thanks, so I guess there is nothing wrong with having a recv timeout. I might be having some kind of WiFi issues then, the connection with the WiFi isn't that good

johanmeijer, 2022-06-17:
It seems that the shinewifi start a new connection without closing the old one if he thinks the connection doesn't work probably in the log you see a ping (0116) and then announce  (0103 for grott) and maybe some 0119 config in between but then it stops and start a new connection. It looks like that ack on 0103 is not received anymore. 

I have seen logs with 10's(maybe even 100's) 0103 records a day. I have only one (occasionally 2). 

Maybe a network trace will give more clarity about what is happening.

rany2, 2022-06-17:
@johanmeijer I will do so when it happens again, I have a feeling this strange behavior where it was initiating connections and not closing the older ones happened because I ran out of open file handles.

After I ran out of file handles, it realized the server was not working so it maybe triggered some kind of "fix" as a workaround which in this case meant it never closed the older connections. 

It would make sense for me to run out faster than you because of multiple devices and having forwarding enabled. 

In any case it's working fine now even with the recv timeout commit enabled. Hopefully having recv timeout would help close the older connections. I guess we will have to see in the coming days...

johanmeijer, 2022-06-19:
Is your problem solved? 

I do not totally understand your recv data timeout issue. Is this because your now running with multiple threads and every thread has it own received buffer? 

I think in the original version you only look at the recv buffer is there is a trigger that there is something posted on the queue (there is a readable socket).

rany2, 2022-06-19:
@johanmeijer Well I didn't have to do any code changes to fix it. The datalogger was just not working properly, I even tried your version and it was still not working. I just waited a bit and it went back to normal.

Most likely when I ran out of file handles the datalogger became buggy because the server wasn't working properly, hopefully now with a recv timeout set this doesn't happen whatsoever. My timeout of 5 minutes seems to work fine, though 3 minutes should work fine too because that's the ping interval.

closed 2022-06-19 by rany2