johanmeijer/grott #128: No data?

knutmyr, 2022-06-19:
I'm trying to set this up on my two MID series inverters, but I don't get any data. If I run the grott.py script i get this

Grott Growatt logging monitor : 2.6.1g

Grott Command line parameters processed:
        verbose:         True
        config file:     grott.ini
        output file:     <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
        nomqtt:          False
        inverterid:      automatic
        pvoutput:        False
        blockcmd:        False
        noipf:           False

Grott process configuration file

Grott process environmental variables

Grott override settings if set in commandline

Grott process json layout files

Grott layout records loaded
T02NNNN  :
{'decrypt': {'value': 'False'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 210, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'gfcif': {'value': 214, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'dcif': {'value': 218, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vpvfault': {'value': 222, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vacfault': {'value': 226, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'facfault': {'value': 230, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'tmpfault': {'value': 234, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'faultcode': {'value': 238, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 246, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 250, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}, 'rac': {'value': 318, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractoday': {'value': 326, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractotal': {'value': 334, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}}
T05NNNN  :
{'decrypt': {'value': 'True'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'numx', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 210, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'gfcif': {'value': 214, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'dcif': {'value': 218, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vpvfault': {'value': 222, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vacfault': {'value': 226, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'facfault': {'value': 230, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'tmpfault': {'value': 234, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'faultcode': {'value': 238, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 246, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 250, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}, 'rac': {'value': 318, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractoday': {'value': 326, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractotal': {'value': 334, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}}
T06NNNN  :
{'decrypt': {'value': 'True'}, 'pvserial': {'value': 76, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 136, 'divide': 10}, 'recortype1': {'value': 150, 'length': 2, 'type': 'num', 'incl': 'yes'}, 'recortype2': {'value': 154, 'length': 2, 'type': 'num', 'incl': 'yes '}, 'pvstatus': {'value': 158, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 162, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 174, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 178, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 186, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 190, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 194, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value':  [... cut ...]
juni 18 09:37:31 sunnylogger grott[901]: Hostname : sunnylogger
juni 18 09:37:31 sunnylogger grott[901]: IP :  127.0.0.2 , port :  5279
juni 19 09:35:14 sunnylogger systemd[1]: Stopping Grott Growatt monitor...
juni 19 09:35:14 sunnylogger systemd[1]: grott.service: Main process exited, code=killed, status=15/TERM
juni 19 09:35:14 sunnylogger systemd[1]: grott.service: Succeeded.
juni 19 09:35:14 sunnylogger systemd[1]: Stopped Grott Growatt monitor.

Any ideas what I've done wrong, or does it not work with my invertes?

rany2, 2022-06-19:
> IP : 127.0.0.2 , port : 5279

This is a loopback address. You need to have it listen on your LAN (typically 192.168.*.*) and configure the inverter to use it.

knutmyr, 2022-06-19:
Traffic already routed trough this computer, and upload to growatt works.

What I've done is change settings in grott.ini file in my grott working dir. Now I have correct IP in joirnalctl, but still no data. My inverters push data every 1 min since I'm using data in home control.

rany2, 2022-06-19:
Verify your firewall settings then? Did you follow the instructions on how to have these datalogger use your computer instead of Growatt's servers?

knutmyr, 2022-06-19:
I made the changes to network settings in Linux. As i tried another solution first, but realized that my loggers are too new for that way. And my loggers are set to local ip.

rany2, 2022-06-19:
Are you sure you actually have `192.168.3.5` assigned to your computer? What is your current IP on the computer you are trying to run grott on?

knutmyr, 2022-06-19:
Yeah I'm sure it's correct,  run static assignments to all my devices, and use ssh to connect to it

johanmeijer, 2022-06-19:
From the log I see I can not conclude it will not work with your inverters.  (Mid TL3-X will work, with invtype = tl3).  

To be honest at this moment I do not understand your problem. Do you not receiving any data at all? 

You say you made changes in the Linux network settings. What do you mean by that. Running Grott in proxy mode does not require any Linux settings changes. You onlt have to change the destination address/port from the inverter / datalogger from the growatt server to the server you running Grott.

knutmyr, 2022-06-19:
I don't get anything when listening to mosquito topic or nothing on pvoutput.org.

johanmeijer, 2022-06-19:
How did you configure the remote server adres in your inverter (datalogger).

knutmyr, 2022-06-19:
[image]
Inverter logger settings

johanmeijer, 2022-06-19:
That seems to be oké. Strange you do not see any data in grott.

knutmyr, 2022-06-20:
Yes I did when trying the other solution. I'll revert them and see if it works then

johanmeijer, 2022-06-20:
It should be possible to run in proxy mode with multiple inverters (if they are the same model). 

I think you might have to specify invtype=tl3 in the grott.ini if you do not see the right data. 

I have test with sniffer mode for a long time (I do not have a server with ip-forwarding available any more) so I am not sure if everything works fine (I do not see any reason why not but you never know). 

PV Output with 2 inverters should work.

knutmyr, 2022-06-20:
I have right data now, in sniffer mode. Not getting upload to pvoutput now, will look into it after work. Tried to find Node-Red flow as mentioned in wiki, but haven't found it yet

ZonPV, 2022-06-20:
It should be here:
[https://github.com/johanmeijer/grott/tree/2.7-(Beta)/examples/nodered](url)

knutmyr, 2022-06-20:
> It should be here: [https://github.com/johanmeijer/grott/tree/2.7-(Beta)/examples/nodered](url)

Thanks :) Now just need to figure out how to split data from 2 inverters in node red and home assistant

johanmeijer, 2022-06-20:
I suppose you read this entry already: https://github.com/johanmeijer/grott/wiki/PVOutput.org-support 

I know it works for a couple of people.  If you give me the grott log around the pvoutput processing I will see if I can find the problem. 

In node red you can identify the inverter based on pvserial in the MQTT json message.

knutmyr, 2022-06-21:
This is my mqtt topics. will try to figure how to sort them from device ID

{
    "device": "GUF0A11025",
    "time": "2022-06-21T10:18:44",
    "buffered": "no",
    "values": {
        "pvstatus": 1,
        "pvpowerin": 15851,
        "pv1voltage": 3687,
        "pv1current": 21,
        "pv1watt": 7742,
        "pv2voltage": 3727,
        "pv2current": 21,
        "pv2watt": 7826,
        "pvpowerout": 15614,
        "pvfrequentie": 4995,
        "pvgridvoltage": 1368,
        "pvgridcurrent": 39,
        "pvgridpower": 5335,
        "pvgridvoltage2": 1376,
        "pvgridcurrent2": 38,
        "pvgridpower2": 5228,
        "pvgridvoltage3": 1372,
        "pvgridcurrent3": 38,
        "pvgridpower3": 5213,
        "Vac_RS": 2379,
        "Vac_ST": 2373,
        "Vac_TR": 2380,
        "pvenergytoday": 38,
        "pvenergytotal": 60941,
        "totworktime": 25854779,
        "epv1today": 22,
        "epv1total": 31830,
        "epv2today": 21,
        "epv2total": 31166,
        "epvtotal": 62996,
        "pvtemperature": 472,
        "pvipmtemperature": 356,
        "pvboottemperature": 339,
        "uwBatVolt_DSP": 0,
        "pbusvolt": 1974,
        "nbusvolt": 1992
    }
}
QoS: 0 - Retain: false
Message 20 received on energy/growatt at 10:18 AM:
{
    "device": "GVF0A15046",
    "time": "2022-06-21T10:18:41",
    "buffered": "no",
    "values": {
        "pvstatus": 1,
        "pvpowerin": 19503,
        "pv1voltage": 2966,
        "pv1current": 43,
        "pv1watt": 12753,
        "pv2voltage": 3375,
        "pv2current": 20,
        "pv2watt": 6750,
        "pvpowerout": 19052,
        "pvfrequentie": 4994,
        "pvgridvoltage": 1385,
        "pvgridcurrent": 46,
        "pvgridpower": 6371,
        "pvgridvoltage2": 1385,
        "pvgridcurrent2": 45,
        "pvgridpower2": 6232,
        "pvgridvoltage3": 1371,
        "pvgridcurrent3": 48,
        "pvgridpower3": 6580,
        "Vac_RS": 2405,
        "Vac_ST": 2390,
        "Vac_TR": 2380,
        "pvenergytoday": 50,
        "pvenergytotal": 89014,
        "totworktime": 27405990,
        "epv1today": 38,
        "epv1total": 59447,
        "epv2today": 17,
        "epv2total": 30821,
        "epvtotal": 90268,
        "pvtemperature": 492,
        "pvipmtemperature": 406,
        "pvboottemperature": 385,
        "uwBatVolt_DSP": 0,
        "pbusvolt": 1919,
        "nbusvolt": 1965
    }
}

knutmyr, 2022-06-22:
Thanks a lot. Publish to pvoutput via node-red, and get data to Home Assistant