johanmeijer/grott #130: Incorrect data for Growatt SPF 3500 ES

ha-bnst, 2022-06-30:
I have connected my spf 3500 es with grott running on a raspberry pi.
I have added invtype: sph to grott.ini

I am getting the following results, but the values are not correct:
 - Grott automatic protocol detection
         - Grott data record length 359
         - layout   :  T060150
         - no matching record layout found, try generic
         - Record layout used :  T06NNNN
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 00060006015f015044444430424a51304e3600000000000000000000000000000000000000005
                 4554a31424a37303357000000000000000000000000000000000000000016061e0d361e030000
                 002c00090db30000000000aa00000000000300000000198200002634000000000000000014520
                 04b1145099d1384099d1386000000be00bc0117000000000000000000990000001a0002000018
                 560000180800000000000000004e89002d00590a0c0000000000000001000000c500000000000
                 0000000000028000000b0000000000000006f0000005c000009b7000000001892000025260000
                 000000000000ffffff38000000000000002d000300000000005c000001d50000005a008600040
                 00100000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 000000000000000000000e3fb
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06NNNN

         - Grott data record date/time processing started
         - date-time:  2022-06-30T13:54:30
         - Grott values retrieved:
                 -  pvserial             :  TXXXXXXX
                 -  recortype1           :  0
                 -  recortype2           :  44
                 -  pvstatus             :  9
                 -  pvpowerin            :  22983475.2
                 -  pv1voltage           :  0.0
                 -  pv1current           :  17.0
                 -  pv1watt              :  0.0
                 -  pv2voltage           :  0.3
                 -  pv2current           :  0.0
                 -  pv2watt              :  653.0
                 -  pvpowerout           :  978.0
                 -  pvfrequentie         :  0.0
                 -  pvgridvoltage        :  0.0
                 -  pvgridcurrent        :  0.0
                 -  pvgridpower          :  520.2
                 -  pvgridvoltage2       :  7.5
                 -  pvgridcurrent2       :  442.1
                 -  pvgridpower2         :  16128909.2
                 -  pvgridvoltage3       :  246.1
                 -  pvgridcurrent3       :  499.8
                 -  pvgridpower3         :  19.0
                 -  pvenergytoday        :  1232104.7
                 -  pvenergytotal        :  0.0
                 -  totworktime          :  0.0
                 -  pvtemperature        :  15.3
                 -  pvipmtemperature     :  0.0
                 -  epv1today            :  0.1
                 -  epv1total            :  19.7
                 -  epv2today            :  0.0
                 -  epv2total            :  0.0
                 -  epvtotal             :  4.0

johanmeijer, 2022-06-30:
Did you specify invtype = spf in the grott.ini?

ha-bnst, 2022-06-30:
Yes this is the section from grott.ini. I used sph, as it is a hybrid inverter with batteries

[Generic]
invtype = sph

johanmeijer, 2022-06-30:
That is strange I expect that it should use layout T06NNNNSPF then. 

Do you have the grott log from grrott startup?

johanmeijer, 2022-06-30:
Sph. I think for spf (off grid) you shoul use spf not sph.

ha-bnst, 2022-07-01:
I have changed the type to spf and attached the output below:

Grott Growatt logging monitor : 2.6.1g

Grott Command line parameters processed:
        verbose:         True
        config file:     grott.ini
        output file:     <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
        nomqtt:          False
        inverterid:      automatic
        pvoutput:        False
        blockcmd:        False
        noipf:           False

Grott process configuration file

Grott process environmental variables

        Pulled 'gblockcmd=False' from the environment

        Pulled 'gnomqtt=False' from the environment

        Pulled 'gmqttip=192.168.1.9' from the environment

        Pulled 'gmqttauth=True' from the environment

        Pulled 'gmqttuser=homeassistant' from the environment

        Pulled 'gmqttpassword=ciebaekechievoowa4xeuFaesahfo1gooqu8AiNg2oGh0shaiShae0oGh4joez5f' from the environment

        Pulled 'gpvoutput=False' from the environment

        Pulled 'gpvapikey=yourapikey' from the environment

        Pulled 'gpvsystemid1=12345' from the environment

        Pulled 'ginflux=False' from the environment

        Pulled 'ginflux2=False' from the environment

        Pulled 'gifdbname=grottdb' from the environment

Grott InfluxDB server IP address env invalid

        Pulled 'gifport=8086' from the environment

        Pulled 'gifuser=grott' from the environment

        Pulled 'giforg=grottorg' from the environment

        Pulled 'gifbucket=grottdb' from the environment

        Pulled 'giftoken=influx_token' from the environment

        Pulled 'gextension=false' from the environment

        Pulled 'gextname=grottext' from the environment

Grott override settings if set in commandline

Grott process json layout files
t0060104sph.json

Grott layout records loaded
T02NNNN  :
{'decrypt': {'value': 'False'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 210, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'gfcif': {'value': 214, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'dcif': {'value': 218, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vpvfault': {'value': 222, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vacfault': {'value': 226, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'facfault': {'value': 230, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'tmpfault': {'value': 234, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'faultcode': {'value': 238, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 246, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 250, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}, 'rac': {'value': 318, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractoday': {'value': 326, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractotal': {'value': 334, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}}
T05NNNN  :
[... cut ...]
                         202, "epv2today": 0, "epv2total": 0, "epvtotal": 0}}
         - MQTT message message sent
         - Grott Send data to PVOutput disabled
         - Grott Send data to Influx disabled
         - Grott extension processing disabled

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.18.0.2', 5279), raddr=('192.168.4.51', 28986)>
         - Data less then minimum record length, data not processed

ha-bnst, 2022-07-01:
I have worked out the issue, it was not pulling the variables in from the ini file. All working now, thanks for your help.

closed 2022-07-01 by ha-bnst

johanmeijer, 2022-07-01:
:)