johanmeijer/grott #131: More Datalogger Register Commands

rany2, 2022-06-30:
This is for ShineWifi-S but might apply for the rest.

I was messing around with my datalogger and found some registers that might be of interest. While the inverter registers are well documented, datalogger ones are not. For this reason I decided to put some info here as some kind of "crowd-funded" effort.

Previously mentioned in https://github.com/johanmeijer/grott/issues/123:

* Register 4: reporting interval in minutes
* Register 8: serial number
* Register 17: growatt server IP
* Register 18: growatt server port
* Register 21: firmware version
* Register 31: datalogger time
* Register 56: wifi SSID
* Register 57: wifi pwd
* Register 76: wifi signal strength in dBm
* Register 80: firmware URL set when attempting update from site

Newly found:

* Register 32: only WRITE, when not zero it restarts the datalogger

johanmeijer, 2022-07-01:
Thanks!  I will start documenting it in the wiki soon.

osos, 2022-08-16:
`curl 'http://localhost:5782/datalogger?command=register&datalogger=[dataloggerserial]&register=16'`

Additional

- Register 16: MAC address of Wifi
- Register 19: growatt server sub.domain.tld - can be used instead of ip (register 17) to point to sub.domain.tld hosting the grott server

smallbitt, 2022-11-05:
On my Shinewifi-s I can't read or set any of those registers. I get an error: "Read Param Failure". Sadly I cannot even change the interval.

johanmeijer, 2022-11-06:
Do you use the blockcmd parameter?

smallbitt, 2022-11-06:
Oh yeah. It was blockcmd parameter set on "true". 
So does the interval register works only through grottserver or even directly to growatt server?

johanmeijer, 2022-11-06:
It also works with grottserver. See also #93. 

Be aware using the grottserver the interval can not set to low (I think not <2 minutes).

cyberdiamond, 2023-07-28:
I have been doing some testing, and now I have some more to add:

Register 5: Modbus range low
Register 6: Modbus address
Register 9: Hardware version
Register 11: FTP details
Register 12: DNS ip address
Register 20: Device type
Register 22: Version
Register 25: Subnet mask
Register 26: IP address

There are lots more with values that I am not sure of their purpose yet?

rany2, 2023-07-28:
@cyberdiamond These don't apply to SHINEWIFI-S, or at least the values appear uninitalized on my unit.

cyberdiamond, 2023-07-28:
I am using a Shinelan box so maybe some are specific to that?

cyberdiamond, 2023-07-28:
To be more exact for any one finding this in the future it is a ShineLink (which is a combination of ShineLanBox and ShineRFStick @ 433mhz)

cyberdiamond, 2023-07-30:
A couple of possiblities:

Register 28: Offline data number?
Register 30: Timezone (looks like set to China timezone)?

curl -X GET "http://127.0.0.1:5782/datalogger?datalogger=NAC5Axxxxx&command=register&register=30"
{"value": "LLT+8"}