johanmeijer/grott #134: trouble with docker

ageurtse, 2022-07-10:
I'm having a bit of trouble setting up the docker container. 

i managed to install the docker container through portainer. 
i altered the grott.ini file to my needs, from within the container. 

after (re)starting the containe the growatt inverter isn't connecting to the grottserver

i think it has somthing to do with the network settings. 

below the last part of the log. 
```
	bucket:      	 grottdb
	token:       	 **secret**
_Extension:
	extension:   	 False
	extname:     	 grottext
	extvar:      	 {'none': 'none'}
Grott proxy mode started
Hostname : 722dcc0cf5aa
IP :  172.30.33.9 , port :  5279 

```

when running from my linux box, command line i have 127.0.0.1 (locahost, which is 192.168.1.30)

need some help with this.

ZonPV, 2022-07-10:
Not quite clear what sort of software you try to run. If you want your own server than you have to open 2 ports in Docker: port 5279 and 5782 as well
Maybe its helpfull if you show all your settings in the ini-file.

ageurtse, 2022-07-10:
below is the whole ini file. 
where do you open those ports in docker

```
# Specify grott monitor configuration
# Defaults are described
# Remove # and update the value to enable the setting
# Last updated: 2021-01-09
# Version 2.4.0

[Generic]
# Specify verbose for extended messaging
verbose = True

# Specify minrecl for debugging purposes only (default = 100)
#minrecl = 100

# Specify mode (sniff or proxy)(> 2.1.0 proxy is default)
mode = proxy

# Specify port and IP address to listen to (only proxy), default port 5279, 0.0.0.0 ==> own ip address
#ip = 192.168.1.30
#port = 5781

# To blocks commands from outside (to channge inverter and shine devices settings) specify blockcmd = True,
# specify noipf = True if you still want be able to dest ip addres from growatt server
# Specify noipf = True if you still want be able to dest ip addres from growatt server (advice only to use
# this for a short time)
#blockcmd = True
#noipf = True

# Time = auto/server parameter enable/disable date/time retrieval from data record (server), default is
# auto: grott decides which time is used (data record if valid otherwise Server)
# If time = server Grott server time is alwas used
time = auto

# Sendbuf = True / False parameter to enable  / disable sending historical (buffered) data. Default is sendbuf = $
sendbuf = True

# Compat is True and valoffset needs to be set if offset / growatt protocol has been changed.
#compat = False
#valueoffset = 6

# Specify inverter id (not necessary in version >2.1.0 if compat = false!)
#inverterid = ABC1234567

# Decrypt is False if growatt communication is not encrypted (older inverters), (not necessary in version
# >2.1.0 if compat = false!)
decrypt = True

[Growatt]
# Ip address and port of Growatt server (specify only if IP has changed)

ip = 192.168.1.30
port = 5781

[MQTT]
# Mqtt parameters definitions
# Be aware nomqtt = True means no MQTT processing will be done!!!!!!

#nomqtt = False
ip = 192.168.1.30
port = 1883
topic= energy/growatt
auth = true
user = *****
password = *****

[PVOutput]
# PVOutput parameters definitions

#pvoutput = True

# Use this if you have one inverter
#apikey = yourapikey
#systemid = 12345

# Use this if you have multiple inverters
#systemid1 = 12345
#inverterid1 = inverter1
#systemid2 = 67890
#inverterid2 = inverter2
#systemid99 = 99999
#inverterid99 = inverter99
#inverterid99 = inverter99

[influx]
# Influxdb parameters definitions

#influx = False
#influx2 = False
#dbname = grottdb
#ip = localhost
#port = 8086
#user = grott
#password = growatt2020
#token  = "influx_token"
#org  = "grottorg"
#bucket = "grottdb"

[extension]
# grott extension parameters definitions

#extension = True
#extname = grottext
#extvar = {"var1": "var1_content", "var2": "var2_content"}

```

johanmeijer, 2022-07-10:
Same question as @ZonPV do you use grott or grottserver container and have you open the port 5279 ( and 5782 for grottserver) in docker (portainer).

ZonPV, 2022-07-10:
I see a difference in portnumbers the log of your first post (5279) and in your last post: the ini-file (5781)

Strange...

ZonPV, 2022-07-10:
What port do you use in the Wifi or LAN-stick?

ageurtse, 2022-07-10:
i use only grottserver container, and didn't open anything, don't know howto open those ports 
on the lan-stck i use port 5279

ZonPV, 2022-07-10:
This are my settings in Portainer:

[url=https://ibb.co/19vwm5s][img]https://i.ibb.co/vJwKjGs/Portainer.jpg[/img][/url]

ZonPV, 2022-07-10:
What kind of hardware do you use? And what is your OS?

ageurtse, 2022-07-10:
my host enviroment is a debian machine.
 i installed the container with this command ledidobe/grottserver:beta
but ther ports where n't setup, where could i add those manualy

ageurtse, 2022-07-10:
i find the solution. 

after editting the container, i can add both ports. then after recreating the container/image one have to alter the grott.ini (don't know how to alter the mqtt settings from within portainer) 
know it is up and running. 

For know it could be closed, trying to find out how to add mqtt settings from within portainer, so that they are keeped after an update.

johanmeijer, 2022-07-10:
I designed grott to be configured with environmentals to be used for containers. So you would have to configure grott.ini (every setting in the .ini has a g.... environmental).

You can also try to externalise grott.ini and make a docker bind to that file. I have to say that I experience that as complex (with the docker run command)  but maybe it is easy with portainer.

ageurtse, 2022-07-14:
Where do i have to enter from within portainer to get the settings, do i have to edit the container and add these to this line in portainer ?
[image]

> I designed grott to be configured with environmentals to be used for containers. So you would not have to configure grott.ini (every setting in the .ini has a g.... environmental).
> 
> You can also try to externalise grott.ini and make a docker bind to that file. I have to say that I experience that as complex (with the docker run command) but maybe it is easy with portainer.

ZonPV, 2022-07-23:
You wrote:

```
i find the solution.

after editting the container, i can add both ports.
```

Can you tell us how you did this action? 
I guess you can use this method too for adding other environmentsettings.

Another way is go into the CLI of Portainer and edit grott.ini. But than you have to install an editor in the container first.

ageurtse, 2022-07-24:
> Can you tell us how you did this action?
> I guess you can use this method too for adding other environmentsettings.

i edit the container from within portainer, in portainer there is a button colled duplicate/edit
[image]

then next under manual port publishing add the 2 ports
[image]

only thing i did't figure out is how to get the mqtt credential in the containr, now i installed through the CLI an editor en alterd the ini file. 
but i would like to keep it clean and would add this through the docker container/portainer. 

did you figure that part out ?

closed 2022-07-24 by ageurtse