johanmeijer/grott #135: Error in packet parser

erikieNL, 2022-07-12:
I am running Grott on my QNAP server to send data to my Home Assistant installation. I noticed the following error to happen:

Traceback (most recent call last):
  File "grott.py", line 35, in <module>
    proxy.main(conf)
  File "/app/grottproxy.py", line 85, in main
    self.on_recv(conf)
  File "/app/grottproxy.py", line 165, in on_recv
    procdata(conf,data)    
  File "/app/grottdata.py", line 191, in procdata
    definedkey[keyword] = int(result_string[conf.recorddict[layout][keyword]["value"]:conf.recorddict[layout][keyword]["value"]+(conf.recorddict[layout][keyword]["length"]*2)],16)                                     
ValueError: invalid literal for int() with base 16: ''

To resolve this more easily I will enable verbose mode again to capture the packet that causes this. When it happens again will post the packet. 
I have a growatt 3600MTL inverter and have the data interval set to 0.2 minutes.

erikieNL, 2022-07-12:
It happened 2 times now in a week time. It stops the docker. Now running with verbose on so I can get the package when it happens again.  QNAP does not restart a container when it stops so I can see the last data in the console screen

johanmeijer, 2022-07-12:
Do you use Grott or Grottserver?

erikieNL, 2022-07-13:
Using Grott to send data both to mqtt broker and to growatt itself. This way I can use my growatt app when I am not at home and still have all my historical data in the cloud and start building up my complete picture at home as I am also using my P1 data over there.

erikieNL, 2022-07-20:
got the error again this morning, here is full output from the console:
- Growatt original Data: 
\x00\x01\x00\x02\x00\xd9\x01\x04\x41\x48\x34\x34\x33\x37\x30\x34\x32\x37\x43 
\x53\x31\x34\x34\x30\x30\x30\x39\x32\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00 
\x2c\x00\x01\x00\x00\x00\x6f\x04\xd5\x00\x00\x00\x00\x00\x00\x06\x92\x00\x00 
\x00\x00\x00\x6f\x00\x00\x00\x5f\x13\x8a\x09\x0b\x00\x00\x00\x00\x00\x5f\x00 
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96 
\x00\x03\x69\xc5\x0c\xc2\xb4\xd5\x01\x84\x00\x00\x00\x00\x00 
- Grott automatic protocol detection 
- Grott data record length 110 
- layout : T020104 
- no matching record layout found, try generic 
- Record layout used : T02NNNN 
- Grott Growatt unencrypted data used 
- Growatt plain data: 
0001000200d901044148343433373034323743533134343030303932000000000000020000002 
c00010000006f04d5000000000000069200000000006f0000005f138a090b00000000005f0000 
000000000000000000000000000000000096000369c50cc2b4d501840000000000 
- Growatt new layout processing 
- decrypt : False 
- offset : 6 
- record layout : T02NNNN
Traceback (most recent call last): 
File "grott.py", line 35, in <module> 
proxy.main(conf) 
File "/app/grottproxy.py", line 85, in main 
self.on_recv(conf) 
File "/app/grottproxy.py", line 165, in on_recv 
procdata(conf,data) 
File "/app/grottdata.py", line 191, in procdata 
definedkey[keyword] = int(result_string[conf.recorddict[layout][keyword]["value"]:conf.recorddict[layout][keyword]["value"]+(conf.recorddict[layout][keyword]["length"]*2)],16) 
ValueError: invalid literal for int() with base 16: ''

johanmeijer, 2022-07-20:
Thank your for the information. I will look into it. 

It has been a long time since I have seen this format of the data record (T02xxxx means old type of record / unencrypted and without CRC error checking). I have to see how we can detect this is an invalid record. But we will manage it! 

It also not ok that Grott stops working after this so the needs to be fixed as well.

johanmeijer, 2022-08-07:
@erikieNL  I added record validation in the new beta version (2.7.5). This should prevent your error!

Your inverter (datalogger) does use CRC checking (only length validation will be performed) for other users: 

- To enable CRC checking for Grott an additional python library is needed (sudo pip3 install libscrc) 
- Without libscrc only validation on length will be performed.