johanmeijer/grott #137: grottserver docker container

ageurtse, 2022-07-24:
I don't see how to add parameters to the grottserver container. 
the command to start this container is /bin/sh -c ./grottwrap.sh from with in this file grottserver is started and grott itself. 
how do i pass the mqtt settings to this. 

now i edited through the cli the ini file. 

i use portainer as helper for my container managment.

johanmeijer, 2022-07-24:
You have to define environmental variables or externalize the grott.ini (defining a volume / file mapping). 

The last thing I do not know yet how to do that in Portainer. 

Defining environmental is easy in the env tab using adv settings: 
[image]

These are the MQTT settings you can change (from the wiki: https://github.com/johanmeijer/grott/wiki/Grott-Configuration):
[image]

Be aware you also have to do a port mapping!

[image]

johanmeijer, 2022-07-24:
Sorry you have to map two network ports for grottserver (5279/5782) but you already did that in issue #134

johanmeijer, 2022-07-24:
You externalize the grott.ini by defining a volume mapping: 

[image]

Be aware use the bind option!

ageurtse, 2022-08-16:
Thanks, this solved it