johanmeijer/grott #138: Idea: using custom DNS for ShineLan-X?

Lyve1981, 2022-08-01:
Another idea I am having how to point a ShineLan-X USB module to grott would be to use a DNS override. Lets say that the ShineLan-X IP is 192.168.1.10 and grott is 192.168.1.15, a configuration for unbound 1.6 could look like this:

```
server:
  access-control-view: 192.168.1.10/32 grott
view:
  name: "grott"
  local-data: "server.growatt.com. 90 IN A 192.168.1.15"
  view-first: yes
```
What this does is to replace the IP address of server.growatt.com only for the ShineLan-X USB module. Other clients in the same net would still be able to access server.growatt.com in regular fashion.

However: Did anyone ever try if this works? It can only work if ShineLan-X does **not** check the validity for the DNS entry for server.growatt.com. My inverter is getting installed this week so right now I cannot test it. If no one else knows, I'll report back after installation. Maybe this is worth being added to the wiki if this works?

johanmeijer, 2022-08-03:
I am not sure. I do not have a privat dns running to try.

 But having looked at the protocol and know that the wifishine is based on esp8266 I do not think it checks dns validity.

Lyve1981, 2022-08-04:
Just wanted to report that it works. I have overwritten the DNS entry for server.growatt.com, it now points to the local IP where grott is running instead and my ShineLan-X connects just fine.

For those using pfsense, these are my settings, they need to be put into the custom options field for the DNS Resolver:
[image]

192.168.20.24 is my ShineLan-X USB module
192.168.20.19 is grott

By using custom options, the DNS response is **only** modified for the ShineLan-X module. This way, I can still access the regular website at https://server.growatt.com from any other device

closed 2022-08-04 by Lyve1981