johanmeijer/grott #140: Is it possible to increase the interval?

skinkie, 2022-08-07:
On the wiki I notice that you are able to change the interval for a WiFi stick. https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server

As having the LAN variant, I don't see a user configurable option to increase the interval. Is there any option to do so?
[image]

johanmeijer, 2022-08-07:
Yes that is possible see discussions #93. 

https://github.com/johanmeijer/grott/discussions/93

It can be done via the growatt website or if you use grottserver then it can be done using the API.

skinkie, 2022-08-07:
@johanmeijer thanks!! Confirmed!

[image]

closed 2022-08-07 by skinkie

johanmeijer, 2022-08-07:
You can also specify interval < 1 minutes (e.g. 0.5) but the Growatt server might have problems to keep up with this.