johanmeijer/grott #142: mqtt output from a Growatt Hybrid inverter explained ?

osos, 2022-08-09:
Hi,

I am using a Hybrid Inverter SPH-3000 with a Smart-Meter and a battery.

Output on mqtt:
```
      "pvstatus": 5,
      "pvpowerin": 4000,
      "pv1voltage": 2855,
      "pv1current": 14,
      "pv1watt": 4092,
      "pv2voltage": 0,
      "pv2current": 0,
      "pv2watt": 0,
      "pvpowerout": 3962,
      "pvfrequentie": 5004,
      "pvgridvoltage": 2270,
      "pvgridcurrent": 20,
      "pvgridpower": 3987,
      "pvgridvoltage2": 0,
      "pvgridcurrent2": 0,
      "pvgridpower2": 0,
      "pvgridvoltage3": 0,
      "pvgridcurrent3": 0,
      "pvgridpower3": 0,
      "totworktime": 20876883,
      "eactoday": 17,
      "pvenergytoday": 17,
      "eactotal": 14947,
      "epvtotal": 15271,
      "epv1today": 15,
      "epv1total": 15257,
      "epv2today": 0,
      "epv2total": 0,
      "pvtemperature": 295,
      "pvipmtemperature": 270,
      "pvboosttemp": 279,
      "bat_dsp": 519,
      "eacharge_today": 4,
      "eacharge_total": 689,
      "batterytype": 1,
      "uwsysworkmode": 5,
      "systemfaultword0": 0,
      "systemfaultword1": 0,
      "systemfaultword2": 0,
      "systemfaultword3": 0,
      "systemfaultword4": 0,
      "systemfaultword5": 0,
      "systemfaultword6": 0,
      "systemfaultword7": 2048,
      "pdischarge1": 0,
      "p1charge1": 0,
      "vbat": 518,
      "SOC": 10,
      "pactouserr": 1688,
      "pactousertot": 1688,
      "pactogridr": 0,
      "pactogridtot": 0,
      "plocaloadr": 5600,
      "plocaloadtot": 5600,
      "spdspstatus": 5,
      "spbusvolt": 2935,
      "etouser_tod": 78,
      "etouser_tot": 16138,
      "etogrid_tod": 0,
      "etogrid_tot": 3308,
      "edischarge1_tod": 1,
      "edischarge1_tot": 2735,
      "eharge1_tod": 1,
      "eharge1_tot": 2954,
      "elocalload_tod": 92,
      "elocalload_tot": 27882
```

It is not all of the payload I can directly understand or find documented.

Could help out.

I would like to identify:

- Current Solar production (which is either used by home, exported to grid or charging the battery)
- Total Solar production (today and all-time)
- Current Home load (the smart-meter should measure this)
- Total Home consumption (today and all-time)
- Current charge / discharge of battery
- Current export to grid
- Total export to grid (today and all-time)
- Current import from grid
- Total import from grid (today and all-time)

jonnym1, 2022-08-09:
most of those values are calculated from the the values above, this is the formula the growatt website uses...

`Priority mode=	Priority
System State=	Device connection status != Lost and Device connection status != Malfunction (UwSysWorkMode) else (Device connection status)
Solar=	Ppv / 1000
PV Inverter= Power	Pex / 1000
Charging=	Pcharge1 / 1000
Discharging	=Pdischarge1 / 1000
SOC=	SOC
Import=	PacToUser / 1000
Export=	PacToGrid / 1000
Consumption=	(Ppv + Pdischarge1 + PacToUser - PacToGrid - Pcharge1) / 1000

Totals
Solar=	Today(Epvtoday)、Total (EpvTotal)
Inverter  generation=	Today(Eextoday)、Total (Eextotal)
Discharged	Today= (Edischarge1Today)、Total (Edischarge1Total)
Exported To Grid=	Today (new version (EsystemToday - EselfToday) else (EtoGridToday))、Total (new version (EsystemTotal - EselfTotal) else (EtoGridTotal))
Load Consumption=	Today (new version (ElocalLoadToday) else (TodaySolar + Edischarge1Today + EtoUserToday - EtoGridToday - Echarge1Today)、Total (new version (ElocalLoadTotal) else (TotalSolar + Edischarge1Total + EtoUserTotal - EtoGridTotal - Echarge1Total)
Solar Revenue=	Today (TodaySolar * Selling Price)、Total (TotalSolar * Selling Price)`

jonnym1, 2022-08-09:
That did'nt come out very clear, Try again...
[image]
[image]

osos, 2022-08-09:
Thank you.

The mapping from the document to the payload is, however, not very clear on all points.

Maybe a comment somewhere on each of the properties in the payload could be given.

ppv is that pvpowerin?
pex is that pvpowerout?

how to map eextoday to the payload?

Of course I am seeking to switch to HA to use these values in the Energy Dashboard instead of the values from Growatt.com.

jonnym1, 2022-08-09:
I believe eextoday is eactoday. but none of the figures are very accurate! it you've got real metering devices in your home the real measurements don't often match the measurement from the inverter, there are internal inefficencys and losses which are not accounted for! for example the inverter uses 50-60watts just to power itself and this is not reflected in the data.

johanmeijer, 2022-08-09:
@jonnym1, The values in the MQTT message are coming from this document: 

https://www.scribd.com/document/489508943/Growatt-Inverter-Modbus-RTU-Protocol-II-V1-20-English

From 4.2 input reg describes the possible data fields/registers I (grott) can retrieve from the data record. 
For SPG the register 0-124 and 1000 -1124 applies. Maybe not all data are in the MQTT message now but can be included by changing the layout (please let me know if you miss something). 

Indeed at the growatt site other values are calculated. Thanks to @osos for explaining that. Did you find this at the growatt site? Do you have a link to that?

jonnym1, 2022-08-09:
> Indeed at the growatt site other values are calculated. Thanks to @osos for explaining that. Did you find this at the growatt site? Do you have a link to that?

It just a screenshot, when you hover the mouse over the i symbols on the dashboard on the new site.

johanmeijer, 2022-08-09:
> > Indeed at the growatt site other values are calculated. Thanks to @osos for explaining that. Did you find this at the growatt site? Do you have a link to that?
> 
> It just a screenshot, when you hover the mouse over the i symbols on the dashboard on the new site.

Sorry I read it wrong you explained it ;)

jonnym1, 2022-08-09:
this value is worth adding, quite meaningful for a hybrid inverter, shows load first, battery first or grid first mode..
`"priority"        : {"value" :630, "length" : 2, "type" : "num", "divide" : 1},`

osos, 2022-08-09:
@johanmeijer, thank you for the pdf. Gave some good insights.

I would also appreciate the Battery-temperature, described on page 43. register 1040.

Here is my updated sensors-growatt.eng.yaml file. Feel free to use

```
# Grott - Home Assistant Growatt sensors 
# 
# This file exposes all sensors from Grott to HA, including dummy sensors for the type of the inverter and the type and serial number of the datalogger (Be aware, the dummy
# sensors have to be set manually) 

- platform: template
  sensors:
    growatt_inverter:
      unique_id: growatt_invertertype
      friendly_name: Growatt - Type
      # Please set the type of your inverter
      value_template: "SPH3600"
      icon_template: mdi:select-inverse

- platform: template
  sensors:
    growatt_datalogger_type:
      unique_id: growatt_dataloggertype
      friendly_name: Growatt - Datalogger type
      # Please set the type of your datalogger
      value_template: "ShineWIFI-S"
      icon_template: mdi:select-inverse

- platform: mqtt
  state_topic: energy/growatt
  unique_id: growatt_datalogger_serial
  name: Growatt - Datalogger serial number
  # Please set the serial number of your datalogger
  value_template: "{{ value_json['values']['datalogserial'] }}"
  icon: mdi:select-inverse

- platform: mqtt
  state_topic: energy/growatt
  value_template: "{{ value_json['device'] }}" 
  unique_id: growatt_serial
  name: Growatt - Serial number
  icon: mdi:select-inverse

- platform: mqtt
  state_topic: energy/growatt
  # If you like to have the date in another format, please change "timestamp_custom('%d-%m-%Y')"
  # For more information: https://docs.python.org/3/library/time.html#time.strftime
  value_template: "{{ as_timestamp(strptime(value_json['time'], '%Y-%m-%dT%H:%M:%S')) | timestamp_custom('%d-%m-%Y') }}" 
  unique_id: growatt_date
  name: Growatt - Date
  icon: mdi:calendar

- platform: mqtt
  state_topic: energy/growatt
  # If you like to have the date in another format, please change "timestamp_custom('%H:%M:%S')"
  # For more information: https://docs.python.org/3/library/time.html#time.strftime
  value_template: "{{ as_timestamp(strptime(value_json['time'], '%Y-%m-%dT%H:%M:%S')) | timestamp_custom('%H:%M:%S') }}" 
  unique_id: growatt_time
  name: Growatt - Time
  icon: mdi:clock-digital

- platform: mqtt
  state_topic: energy/growatt
  value_template: >
    {% if (value_json['values']['pvstatus'] | int == 0) %}
      Waiting
    {% elif (value_json['values']['pvstatus'] | int == 1) %}
      Normal
[... 293 lines cut ...]
  unique_id: growatt_export_to_grid
  device_class: power
  unit_of_measurement: "kW"
  name: Growatt - Export to grid (actual)
  icon: mdi:transmission-tower-import

- platform: mqtt
  state_topic: energy/growatt
  value_template: "{{ value_json['values']['etogrid_tod'] | float / 10 }}"
  unique_id: growatt_export_to_grid_today
  device_class: energy
  unit_of_measurement: "kWh"
  name: Growatt - Export to grid (today)
  icon: mdi:transmission-tower-import

- platform: mqtt
  state_topic: energy/growatt
  value_template: "{{ value_json['values']['etogrid_tot'] | float / 10 }}"
  unique_id: growatt_export_to_grid_total
  device_class: energy
  state_class: total_increasing
  unit_of_measurement: "kWh"
  name: Growatt - Export to grid (total)
  icon: mdi:transmission-tower-import
```

jonnym1, 2022-08-09:
This will give you the batt temp...
`            "battemp"           : {"value" :826, "length" : 2, "type" : "num", "divide" : 10}`

osos, 2022-08-09:
> This will give you the batt temp... ` "battemp" : {"value" :826, "length" : 2, "type" : "num", "divide" : 10}`

Sorry for being noob, but how shall I use that to get battemp?

jonnym1, 2022-08-09:
take t06NNNNXSPH.json from the examples/record layout folder and put it into grott's working directory, then edit the file adding above line.
Infact it's already in there, just needs uncommenting.

johanmeijer, 2022-08-09:
It is not that easy in a docker container (adding a file). 

But for now there is a way around it: 

You can get a more detailed MQTT message by specifying -e gincludeall = "True" in the docker run (or for non docker includeall = True in the grott.ini [Generic] section. 

The name will be #battemp (I do not know why I specified an #. Was absolutely logical at that  moment. I will change it in a next update). 

I will also add priority to the layout, that is not there yet.

osos, 2022-08-09:
@johanmeijer, sounds nice.

In docker I have mounted the updated t06NNNNXSPH.json file from docker-compose like this:

```
  grott:
    image: ledidobe/grott:beta
    container_name: grott
    restart: unless-stopped
    ports:
      - "5279:5279"
    volumes:
      - ./grott/t06NNNNXSPH.json:/app/t06NNNNXSPH.json
    environment:
      - ginvtype=sph
      - gblockcmd=False 
      - ...
```

osos, 2022-08-09:
Another thing is, that it would be tremendous nice to change the priority of the inverted between, load-first, grid-first, battery-first, hereunder set whether to charge by AC.

This would allow for automations from HA.

E.g. charge battery while power is cheap, use/sell power when it is expensive.

johanmeijer, 2022-08-09:
With grottserver (see discussion #98) it is possible to change inverter config setting with an API. I think it should be possible then to change the priority. 

I only do not know with register to set for the SPH. 

For the SPF it is register one (1). I have not found it yet in the documentation for the SPH (should be in the holding reg, section of the document mentioned above). If you are able to do it from the growatt site we can trace the command and find out which register it is.

closed 2022-08-10 by osos