johanmeijer/grott #143: docker image grottserver not forwarding data to growatt.com

osos, 2022-08-10:
Just switched from docker image grott:beta (https://hub.docker.com/r/ledidobe/grott) to grottserver:beta (https://hub.docker.com/r/ledidobe/grottserver).

While the configuration is the same, grott forwarded data to mqtt, server.growatt.com and pvoutput.org. Grottserver outputs to mqtt and pvoutput.org.

I still want data on growatt.com for while. How can I do that?

(On a site note, both the grott and grottserver docker images need updating)

johanmeijer, 2022-08-10:
At this moment that is not possible.

The basic idea of grottserver is that it will replace the growatt server and prevent sending data to china.
It is technical not impossible but not part of the basic design.

I do understand the requirement of having the api available and also sending the data to growatt. 
I will enable this in a next version (either in Grott or grottserver).

I know that the docker images need updates. I will do that in the comming days. Available hobby time is a rare resource, especially during the summer ;)

osos, 2022-08-29:
How easy can we enable the grottserver to forward to growatt servers as well.

I want to forward to growatt server to maintain the overview, and to be able to use growatt interface for "failsafe" configuration of the inverter.

However, I would very much like to be able to have an easy way to automate switching the inverter state between grid-first, battery-first (including enabling/disabling ac-charge) and load-first.

I will have Home-Assistant look into the power prices and set up automation to optimize whether to import/export power and best utilization of the battery from an economical aspect.

johanmeijer, 2022-08-29:
Hi @osos,

Forwarding the data to Growatt by grottserver is not that difficult to implement (and I will probably made this an option in one of the next releases).

What I did not intent to do was to enable a 2 way communication for Grottserver (so also allow command / data sent back from growatt server to the inverter / datalogger). 
That is in conflict with the intention of Grottserver not using / not being depended on the growatt website any more. 

The ability of giving commands in Grott in proxy mode seems to be of added value. Let me think about that. Maybe I going to implement that in a next version (Version 3). 

For now you have to use Grottserver and fall back to the old mode (changing grott.ini / enverionmentals) to go into the "failsafe" mode.