johanmeijer/grott #144: MQTT Mosquitto in Home Assistant not receiving any data

gijs007, 2022-08-10:
I'm new to this and probably doing something wrong, but can't figure out what.
Here is what I've done:

1. Installed MQTT Mosquitto broker in Home Assistant (default config)
2. Enabled the integration in Home Assistant (I can locally send and subscribe to data)
3. Created the default grott user in Home Assistant (for authentication with Musquitto)
4. Installed the Grott Docker container with: 
`sudo docker run -d -p 5279:5279 --restart on-failure -e gmqttip="homeassistant.IP" -e TZ="Europe/Amsterdam" -e gblockcmd="True" -e auth="True" ledidobe/grott:beta`

However I'm not seeing any data in the debug log of Musquitto either case. Network connectivity from the Docker host has been verified as working with telnet and nc.
Note: I've also tested this with the default auth disabled. 

Perhaps Grott only starts sending data over MQTT once it receives data from the Growatt Inverter?
Right now I've configured port forwarding on my Opnsense router, which reroutes all TCP/UDP data from my WiFi network with a destination of 47.91.67.66 (growatt server),  to the Grott IP and port 5279. 
This should reroute the traffic from my Growatt inverter to Grott.

Docker container output:
`	version:     	 2.7.2

	verbose:     	 True

	trace:       	 False

	config file: 	 grott.ini

	minrecl:     	 100

	decrypt:     	 True

	compat:      	 False

	invtype:     	 default

	include_all: 	 False

	blockcmd:    	 True

	noipf:       	 False

	time:        	 auto

	sendbuf:     	 True

	timezone:    	 local

	valueoffset: 	 6

	offset:      	 6

	inverterid:  	 automatic

	mode:        	 proxy

	grottip      	 default

	grottport    	 5279

_MQTT:

	nomqtt       	 False

	mqttip:      	 homeassistantip

	mqttport:    	 1883

	mqtttopic:   	 energy/growatt

	mqttmtopic:  	 False

	mqttmtopicname:	 energy/meter

	mqtttretain: 	 False

	mqtttauth:   	 False

	mqttuser:    	 grott

	mqttpsw:     	 **secret**

_Growatt server:

	growattip:   	 47.91.67.66

	growattport: 	 5279

_PVOutput:

	pvoutput:    	 False

	pvdisv1:     	 False

	pvtemp:      	 False

	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp

	pvapikey:    	 yourapikey

	pvinverters: 	 1

	pvsystemid:  	 systemid1

_Influxdb:

	influx:      	 False

	influx2:     	 False

	database:    	 grottdb

	ip:          	 localhost

	port:        	 8086

	user:        	 grott

	password:    	 **secret**

	organization:	 grottorg

	bucket:      	 grottdb

	token:       	 **secret**

_Extension:

	extension:   	 False

	extname:     	 grottext

	extvar:      	 {'none': 'none'}

Grott proxy mode started

Hostname : f87e8715791b`

johanmeijer, 2022-08-10:
Oops documentation error. It should be: -e gmqttauth="True".
I will update the wiki.

gijs007, 2022-08-10:
Thanks, that explains why it wasn't enabled.

Now it shows as enabled. However I'm still not seeing any connections on the broker end in the debug log.
I'm trying to determine where to troubleshoot (connection inverter > Grott or Grott > MQTT).
Does Grott make any MQTT connections, even when it's not receiving data from the inverter?

johanmeijer, 2022-08-10:
Grott connect and sent a message to mqttt if it received data from the inverter.

You should be able to see this in the log of the docker container.

gijs007, 2022-08-10:
Thanks, I tested a simple TCP connection from a PC and see the messages. Apparently I was missing a firewall rule.
I'm not receiving any data from the inverter yet. Does the inverter only send data when the panels produce electricity?

johanmeijer, 2022-08-11:
Mine does not send data. But is also off when the panels does not produce power. 

I am not sure with the always on inverters.

gijs007, 2022-08-11:
Makes sense, I waited for the inverter to start working.
However I still don't see any connections from the inverter to Grott in my firewall. 
However I do see a telnet (port 23) connection from Grott to the inverter, every six seconds. Still Grott docker console output is not showing any connection neither is MQTT.

Just to be sure: I'm currently redirecting TCP traffic from the inverter with destination IP 47.91.67.66 and port 5279 from my firewall (OPNsense) to the Grott IP. Did I overlook something?

johanmeijer, 2022-08-11:
That seems fine to me. Can you look in the grott (docker container) log? Docker logs command.

gijs007, 2022-08-11:
It shows the same as before, only difference being that mqtttauth is now enabled.

johanmeijer, 2022-08-11:
Hmm oké. Then still no data is received by grott. No I think about it that routing the growatt ip address with a firewall might not be a good idea. Grott itself is a proxy receives the data from the inverter and send it to growatt (with the address you rerouted again). 

So you created some kind of loop I think.

The easiest way is to configure your shinelan/wifi to send to the grott ip and delete the growatt firewall rule.

gijs007, 2022-08-14:
It's working now, turned out I had mixed up some internal IP addresses. So the Growatt wasn't being rerouted at all.
Thanks for the help :)

closed 2022-08-14 by gijs007