johanmeijer/grott #146: I am suddenly getting loads of buffered (old) data

osos, 2022-08-12:
Both with grott and grottserver having the same issue. Restarted the ShineWifi-S dongle etc.

Dont know how why there are suddenly buffered data. Are they buffered in grott or in the dongle or the inverter?

How to clear the buffered data

Logs:

```
grott               | 	 - Grott data record date/time processing started
grott               | 	 - date-time:  2022-08-12T16:14:07
grott               | 	 - Grott values retrieved:

...

grott               | 	 - Grott data record date/time processing started
grott               | 	 - date-time:  2022-08-12T00:38:04
grott               | 	 - Grott values retrieved:

...

grott               | 	 - Grott data record date/time processing started
grott               | 	 - date-time:  2022-08-12T00:40:05
grott               | 	 - Grott values retrieved:

...

```

johanmeijer, 2022-08-12:
Just let it run? It is the easiest way I think.

It also possible via the growatt website (and grottserver but I do not know how yet, which register to set).

johanmeijer, 2022-08-12:
Do you know why you buffered data. Do you have  lot off connection loss?

osos, 2022-08-12:
I tried to "Clear Data Logger Log" and restart the datalogger, but didn't solve it.

I have no idea why there was a buffer.

It seems connection must have been lost since yester 21.30-ish and I didn't notice until 15.30 today, restarted grott:beta docker image and the data started to get in again both new and buffered.

As I do not know how to filter the buffered data from HA, it is a mess, as HA handles all data as current state. The PVoutput seems to handled correct by PVoutput.org

johanmeijer, 2022-08-12:
By the way they are buffered in the dongle ( datalogger). I believe it can buffer days ( with the interval of 5 minutes).

closed 2022-08-13 by osos