johanmeijer/grott #151: MQTT topic with multiple inverters

JohnBerg60, 2022-08-14:
I have 2 inverters, linked to home assistant over mqtt.  

Both inverters publish over the same topic, as defined in grott.ini. HA cannot separate the data, because of the same topic.

I changed grottdata.py, from line 373 like this:

```
try: 
                mqtttopic = conf.mqtttopic + '/' + definedkey["pvserial"]
                publish.single(mqtttopic, payload=jsonmsg, qos=0, retain=False, hostname=conf.mqttip,port=conf.mqttport, client_id=conf.inverterid, keepalive=60, auth=conf.pubauth)
                if conf.verbose: print("\t - " + 'MQTT message message sent, topic:', mqtttopic) 
```

I don't know if this is the most elegant solution (problably not, a wildcard in the ini would make more sense), but for me, this quick and dirty fix works.   

my 2 cents ...

johanmeijer, 2022-08-15:
The Grott One MQTT topic supports multiple inverters. The MQTT message JSON format is setup so that difference in inverter can be recognized by the device "key" (example):  

     {"device": "inverterid1", "time": "2022-08-15T20:26:13", "buffered": "no",  
      "values": {"datalogserial": "loggerid", "pvserial": "inverterid",  "pvstatus": 0, "pvpowerin": 0, "pv1voltage": 1664,
      "pv1current":  0, "pv1watt": 0, "pv2voltage": 0, "pv2current": 0, "pv2watt": 0, "pvpowerout": 0, "pvfrequentie": 4997, 
      "pvgridvoltage": 2363, "pvgridcurrent": 0,"pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, 
      "pvgridvoltage3": 0, "pvgridcurrent3": 0,"pvgridpower3": 0, "pvenergytoday": 69, "pvenergytotal": 77420, 
      "totworktime":113228968, "pvtemperature": 287, pvipmtemperature": 0, "epv1today": 74, "epv1total": 81371,
      "epv2today": 0, "epv2total": 0, "epvtotal": 81371}}

I little bit depends on the consumer of the message if it can handles this format. I use Nodered and that works fine with two inverters and this MQTT message format. 

If HA can not separate on basis off this keyword then indeed having a topic per inverter can be a solution.  

Can you look if it is possible in HA (I do not use it) the select on device. If not we can make an enhancement to grott to support  a topic per device (be aware that will be an option that needs to be set in the .ini because of compatibility reasons). 

I will almost be the solution you described above I think (with some if then around it).

1onord, 2022-08-16:
I just tried it once but unfortunately it didn't work. If I use a topic for both inverters, it overwrites the values ​​that exist for both inverters and in Home Assistant the status jumps back and forth. If there was a possibility to have a separate topic for each inverter, the 2 Pi Zero would be omitted and the sticks could send the data directly to the server. Thumbs up for that.

PR #154 (2022-08-26): Add inverter serial to MQTT topic
Fixes #151
closed 2022-08-26 by johanmeijer