johanmeijer/grott #155: Wrong record layout results in high totals

pinpoint87, 2022-08-28:
Hi,

Recently started using Grott, only issue i'm getting is that sometimes at random intervals I get spikes of 100k+ kWh. 
If i am reading the log correctly its coming from the second rail (which is not connected).
Is there a way to filter this these messages out?

Inverter / Logger: 3600MTLS / ShineLink-X

Message with invalid data; (all other messages are received with layout T06NNNN);
> - Grott automatic protocol detection
> - Grott data record length 479
> - layout   :  T065004X
> - no matching record layout found, try generic
> - Record layout used :  T06NNNNX
> - Growatt data decrypted V2
> - Grott Growatt data decrypted
> - Growatt plain data:
>      00d40006015f50044e41433038343132423300000000000000000000000000000000000000003
>      9555a473832363031410000000000000000000000000000000000000000000000000000030000
>      002c000100001d720c67001700001d6c024400000000000600001cdd138a08e3002000001cec0
>      000000000000000000000000000000000000011000186d406f8e2e50167000000000000000000
>      0000000000000001870e0000000000002d00594e20000000000000001100018a6600000000000
>      0000000018a6600000000000000000000000000000000001d0000000000000000000000000000
>      0000000000000000000000000000000000000000000000000000000000000000005a008600191
>      60514121928000000191604130f1b21000000191604130e0d0f0373001916031a09300b008700
>      1915040f120c05000000000000000000000000000000000000000000000000000000000000000
>      0000000000000000000001a6972ba77677454466466444d332d07371a473d0315330628050e03
>      15330628050e0315330628055c0c74917269774175624e4143303834313242330000000000000
>      000000000000000000000000000a22861a37441724f76777d476b3536372401441b050e031533
>      0628050e0315330628050e031533061a31
> - Growatt new layout processing
>      - decrypt       :  True
>      - offset        :  6
>      - record layout :  T06NNNNX
> - Grott values retrieved:
>      -  pvserial             :  Yes
>      -  recortype1           :  0
>      -  recortype2           :  44
>      -  pvstatus             :  1
>      -  pvpowerin            :  753.8
>      -  pv1voltage           :  317.5
>      -  pv1current           :  2.3
>      -  pv1watt              :  753.2
>      -  pv2voltage           :  58.0
>      -  pv2current           :  0.0
>      -  pv2watt              :  0.6
>      -  pvpowerout           :  0.0
>      -  pvfrequentie         :  0.0
>      -  pvgridvoltage        :  0.0
>      -  pvgridcurrent        :  1.7
>      -  pvgridpower          :  10005.2
>      -  pvgridvoltage2       :  178.4
>      -  pvgridcurrent2       :  5808.5
>      -  pvgridpower2         :  2352742.4
>      -  pvgridvoltage3       :  0.0
>      -  pvgridcurrent3       :  0.0
>      -  pvgridpower3         :  0.0
>      -  totworktime          :  182044.4
>      -  pvenergytoday        :  0.0
>      -  pvenergytotal        :  111411.3
>      -  epvtotal             :  232194048.0
>      -  epv1today            :  0.0
>      -  epv1total            :  0.1
>      -  epv2today            :  232194048.0
>      -  epv2total            :  0.0
>      -  pvtemperature        :  13.4
>      -  pvipmtemperature     :  644.0

rany2, 2022-08-28:
The data your server is getting is corrupted in some way. The length field in that modbus packet is 351(excl. header of 8bytes) but grott reports to have ndata of 479 (incl header). Solution is to fix how grott handles read operations and do something like this https://github.com/rany2/grott/blob/master/grottproxy.py#L144-L153 but right now I can't contribute anything because developer is only using GitHub to host code and hasn't switched to git.

rany2, 2022-08-28:
@johanmeijer If you were going to do something like this and aren't using socketserver, do what I did in https://github.com/rany2/grott/blob/master/grottproxy.py#L172-L188. (tl;dr use `socket.MSG_WAITALL` in your recv call)

johanmeijer, 2022-08-28:
@pinpoint87 which Grott version are you using? Invalid data records (this one is longer than the growatt data record)should not be processed in the Beta Version (2.7.5). 

If you are using the beta version I have to test what went wrong.

@rany2 Yep this is the length for the total databuffer received. That is how it was initial programmed and works fine most of the time for a couple of years now. It is used to detect from which type of inverter the data is coming. That was for me the only way to detect this at that point in time (in the mean while I do understand the data record more.
I will include a different way of detecting the record type (not based on length but on content) in one of the future releases. 
But there a lot more improvements I want to make in the future. This will take some time. 

@rany2 please do not try to  point out each time how good you are in programming (and I am apparently not from your point of view). I think Grott works fine for a lot of people and for me it works for experimenting and learning a lot of new things. I really like to read your tips for improvements but please let me adopt changes in my way and speed.

rany2, 2022-08-28:
@johanmeijer I'm sorry I really didn't mean to sound rude, I just had this issue before and wanted to let you know how I solved it. It took me a long time to figure it out and I didn't want to waste your time trying to solve something I already came across. I'll just be quiet, my apologies...

pinpoint87, 2022-08-29:
@johanmeijer, I am running the :latest tag (v2.6.1g)

johanmeijer, 2022-08-29:
@pinpoint87 If you use the beta version this problem should be solved. This version validates the records before processing. 

Please let me know if it works!

pinpoint87, 2022-08-30:
Just pulled 2.7.6 i'll let you know if the issue still exists at the end of the week

seaser74, 2022-08-30:
For whats it worth i had exactly the same issue as discribed above.  I was using grottrpi image, just pulled the 2.7.6 and installed libscrc.
I will let you know if this works later this week.

johanmeijer, 2022-08-30:
Yes please let me know if 2.7.6 solve the problems! Thanks.

seaser74, 2022-09-01:
Sorry, can't get it to start.. sorry:
`Starting grott ... done
Attaching to grott
grott        | exec /usr/local/bin/python: exec format error`

I think the container is looking for python in the wrong place in the " /usr/local/bin" directory, and i don't know how to point the container to the correct path "/usr/bin/"  where python is located.

johanmeijer, 2022-09-01:
Uh I am little bit lost. Do you use the docker container or the standalone Version? 
On which server you try to run it (x86-linux, rpi, nas?)

seaser74, 2022-09-01:
i'm using the docker container on a Rpi.

johanmeijer, 2022-09-01:
Ah oké you should download the grottserverrpi container. But I have not created the 2.7.6 container for rpi yet.

I have will create it the comming days.

johanmeijer, 2022-09-01:
Other question do you want to use grott or grottserver?

seaser74, 2022-09-02:
aha, that clarifies some things!
At the moment i am using Grottrpi:latest, because i was in the assumption grottserver was in a Beta stage??
And to be honest, i don't really care if my solar data goes to our chinees frends. as long as i can forward the Growatt data to my home assistant i am good!
But on the other hand, if it is easier for you to work with the grottserver, then i will be using the server.

johanmeijer, 2022-09-02:
No I will create both Grottrpi and grottserverrpi. I have some challenges with building the containers but it will be there at some point this weekend.

johanmeijer, 2022-09-02:
I added the 2.7.6 RPI containers to docker!

- ledidobe/grottrpi:2.7.6 or ledidobe/grottrpi:beta 
- ledidobe/grottserverrpi:2.7.6 or ledidobe/grottserverrpi:beta  

Please let me know if it works.

seaser74, 2022-09-03:
I am now have a working grottrpi:2.7.6 container, i will let you know if the spikes are gone. 
Thank you very much for the great work! highly appreciated!

seaser74, 2022-09-04:
Sorry Johan, I still get these spikes 2 or 3 times a day. As i am starting to doubt if this is the same problem as pinpoint87,  I enclosed a piece of the log file.
If you need more of the logs, just let me know.

> - Grott automatic protocol detection,
	 - Grott data record length 225,
	 - layout   :  T055004,
	 - no matching record layout found, try generic,
	 - Record layout used :  T05NNNN,
	 - Growatt data decrypted V2,
	 - Grott Growatt data decrypted,
	 - Growatt plain data:,
		 0002000500d9500409332c4759404275402e262a36467f465b4752317447726f7761767447724,
		 3776174744772623661747447726f7761747447726f776174744761e87f99747447726f776174,
		 7447726f7761747447726f7761747447726f77634d3c41a89ce161877447726f7761747447726,
		 f7761747447726f7a2e747447726f5a612d3a67726f7761747447726f7539a67447726f776174,
		 74477037a561747447726f7761747447726f7761747447726f7761747447726f7761757447726,
		 f889e747447161a4761747447726f7a29747447726f7761747447726f776140fa,
	 - Growatt new layout processing,
		 - decrypt       :  True,
		 - offset        :  6,
		 - record layout :  T05NNNN,
,
	 - Grott server date/time used,
	 - Grott values retrieved:,
		 -  datalogserial        :  	3,GY@Bu@.,
		 -  pvserial             :  &*6FF[GR1,
		 -  pvstatus             :  30561,
		 -  pvpowerin            :  195377752.2,
		 -  pv1voltage           :  2514.2,
		 -  pv1current           :  2494.8,
		 -  pv1watt              :  195083940.7,
		 -  pv2voltage           :  3056.1,
		 -  pv2current           :  2981.2,
		 -  pv2watt              :  119868197.5,
		 -  pvpowerout           :  163502189.5,
		 -  pvfrequentie         :  250.6,
		 -  pvgridvoltage        :  3266.5,
		 -  pvgridcurrent        :  2981.2,
		 -  pvgridpower          :  119868197.5,
		 -  pvgridvoltage2       :  2494.8,
		 -  pvgridcurrent2       :  2976.7,
		 -  pvgridpower2         :  191990768.1,
		 -  pvgridvoltage3       :  2981.2,
		 -  pvgridcurrent3       :  1829.0,
		 -  pvgridpower3         :  187009470.8,
		 -  pvenergytoday        :  195083940.7,
		 -  pvenergytotal        :  200299654.0,
		 -  totworktime          :  152995.7,
		 -  pvtemperature        :  2496.7,
		 -  pvipmtemperature     :  2929.5,
		 -  epv1today            :  195377752.2,
		 -  epv1total            :  186995344.6,
		 -  epv2today            :  195083940.7,
		 -  epv2total            :  200287550.8,
		 -  epvtotal             :  119853661.3,

johanmeijer, 2022-09-04:
That is not good!. 

It does not seem to be the same fault as @pinpoint87 I think. In his case the data record was corrupted. In your case the datarecord is not corrupted but the values are wrong I think. These values already put in the data record before it is sent (package, encrypted and secured with CRC error check code) to Grott. 

We have to found out a way how we can detect that this is the wrong content en skip the record.  There is already a filter for a invalid energy meter record, I do not mind to add one for your inverter.

Can you send me a valid record so I can compare? What type of inverter do you have?   

I think it is already possible to filter on inverter status (seems a little bit weird) or maybe another field.

rany2, 2022-09-04:
It is worth checking if an update for the datalogger is available. Upgrading did wonders for reliability in my case. (Check server.growatt.com, go to datalogger settings and upgrade datalogger)

pinpoint87, 2022-09-04:
@johanmeijer, my issue seemed to be resolved by using v2.7.6, no spikes anymore.

seaser74, 2022-09-04:
Okay, so if I understand correct, porbably the datalogger is sending corrupt data? I updated the datalogger, and we will see what happens.(thanks for the tip @rany2! )

Enclosed a valid record from the log. 
Also I see these " Data less then minimum..." log entrys quite often. :

> 
	 - Growatt packet received:,
		  <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.18.0.5', 5279), raddr=('192.168.178.75', 1036)>,
	 - Data less then minimum record length, data not processed,
,
	 - Growatt packet received:,
		  <socket.socket fd=8, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.18.0.5', 57544), raddr=('47.91.67.66', 5279)>,
	 - Growatt original Data:,
		 \x00\x03\x00\x05\x00\xd9\x50\x04\x09\x33\x2c\x47\x59\x40\x42\x75\x40\x2e\x26,
		 \x2a\x36\x46\x7f\x46\x5b\x47\x52\x31\x74\x47\x72\x6f\x77\x61\x76\x74\x47\x72,
		 \x43\x77\x61\x74\x74\x47\x72\x62\x4c\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74,
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x61\xe8\x7f\x8f\x74\x74\x47\x72\x6f\x77\x61,
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f,
		 \x77\x63\x4d\x3c\x41\xa8\x9c\xe1\x61\x87\x74\x47\x72\x6f\x77\x61\x74\x74\x47,
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x7a\x28\x74\x74\x47\x72\x6f\x5a\x61\x2d,
		 \x3a\x67\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x39\xa6\x74\x47\x72\x6f\x77,
		 \x61\x74\x74\x47\x70\x37\xa5\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72,
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75\x74,
		 \x47\x72\x6f\x88\x9e\x74\x74\x47\x16\x1a\x47\x61\x74\x74\x47\x72\x6f\x7a\x29,
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x0f\x5e,
	 - Grott automatic protocol detection,
	 - Grott data record length 225,
	 - layout   :  T055004,
	 - no matching record layout found, try generic,
	 - Record layout used :  T05NNNN,
	 - Growatt data decrypted V2,
	 - Grott Growatt data decrypted,
	 - Growatt plain data:,
		 0003000500d950044e414330383436323241514b4232383434303345000000000000020000002,
		 c0000000000000d3b000000000000000000000000000000000000138708ee0000000000000000,
		 0000000000000000000000000000000000000002394806daf39600f3000000000000000000000,
		 0000000000000000d4900000000002d00594e200000000000000000000258d200000000000000,
		 00000258d20000000000000000000000000000000000000000000000000000000000010000000,
		 0ffff0000006475300000000000000d4800000000000000000000000000007b2a,
	 - Growatt new layout processing,
		 - decrypt       :  True,
		 - offset        :  6,
		 - record layout :  T05NNNN,
,
	 - Grott server date/time used,
	 - Grott values retrieved:,
		 -  datalogserial        :  NAC084622A,
		 -  pvserial             :  QKB284403E,
		 -  pvstatus             :  0,
		 -  pvpowerin            :  0.0,
		 -  pv1voltage           :  338.7,
		 -  pv1current           :  0.0,
		 -  pv1watt              :  0.0,
		 -  pv2voltage           :  0.0,
		 -  pv2current           :  0.0,
		 -  pv2watt              :  0.0,
		 -  pvpowerout           :  0.0,
		 -  pvfrequentie         :  50.0,
		 -  pvgridvoltage        :  228.6,
		 -  pvgridcurrent        :  0.0,
		 -  pvgridpower          :  0.0,
		 -  pvgridvoltage2       :  0.0,
		 -  pvgridcurrent2       :  0.0,
		 -  pvgridpower2         :  0.0,
		 -  pvgridvoltage3       :  0.0,
		 -  pvgridcurrent3       :  0.0,
		 -  pvgridpower3         :  0.0,
		 -  pvenergytoday        :  0.0,
		 -  pvenergytotal        :  14573.6,
		 -  totworktime          :  15974.0,
		 -  pvtemperature        :  24.3,
		 -  pvipmtemperature     :  0.0,
		 -  epv1today            :  0.0,
		 -  epv1total            :  15381.0,
		 -  epv2today            :  0.0,
		 -  epv2total            :  0.0,
		 -  epvtotal             :  15381.0,

rany2, 2022-09-04:
Data less than minimum is normal, it's usually just ping requests from the datalogger to make sure the connection is still alive. You could disable it by changing the minrecl config option to 0. That will allow you to see what the data the datalogger sent was.

seaser74, 2022-09-07:
it looks like updating the datalogger solved the problem, 2 days without any spikes!
Thunk you all very much!

johanmeijer, 2022-09-08:
@pinpoint87 @seaser74 can we close this issue?

pinpoint87, 2022-09-08:
@johanmeijer ok by me.
I didn't mark it as resolved last week because @seaser74 was still waiting for his results.

closed 2022-09-09 by pinpoint87