johanmeijer/grott #156: Alerts in Server API

timwegenerDE, 2022-09-03:
Hi there,

the only reason I’m still forwarding the inverter to growatt is an potential alert. So far I dind‘t get any but who knows…

Is there any possibility to get alerts from the new server API?

Regards,
Tim

johanmeijer, 2022-09-03:
What kind of alerts do you expect and how do you receive these?

At this moment grott does not sent alerts. But it wil be not to difficult to create an extension for that.

But I am very curious what alert growatt can sent. Is there somewhere a description I can read?

timwegenerDE, 2022-09-03:
In the ShinePhone App, see attached.
I don‘t know what alerts to expect there and if it could detect something critical.

A real push
[image]
 / mail or something like that would be nice to have.

timwegenerDE, 2022-09-03:
I found something in the Growatt Server API: http://www.growatt.pl/wp-content/uploads/2020/01/Growatt-Server-API-Guide.pdf (4.3.6 and 4.3.11)

I also found some error codes which could be displayed on the inverter itself in case of an error: https://solaranalytica.com/growatt-error-codes/
These error codes doesn‘t match the error codes from the API examples.

Unfortunately there is no error I would like to provoke in order to see where the errorcode is found in the transmitted data.

However it would be great to get any error(code) from your API here if something is faulty.

rany2, 2022-09-03:
Isn't this just `faultBit` and `warningBit`? I could share a PDF detailing what each bit corresponds to.

timwegenerDE, 2022-09-03:
Maybe?! That would be great.

Is there already something in your api to get them?

rany2, 2022-09-03:
Those values are returned in MQTT and Influx.

timwegenerDE, 2022-09-03:
Currently I‘m just sending the data to PVOuput and just found out about the new grott server api today. So I was hoping to get these informations there.

rany2, 2022-09-03:
It's not in grottserver API because the data regarding warning/faults is transmitted along with all the regular records (like load consumption, pv watt, grid watt, battery soc, etc). Grottserver API only has to do with getting and setting datalogger/inverter registers, setting the inverter's date/time, maintaining the connection with inverter, and acknowledging the receipt of the data records. Everything else is responsibility of grottdata component. I believe if this was to change then grott will have to be redesigned quite dramatically.

In any case using MQTT could help you check if you have any errors and if you're tech savvy you could create a grott extension that could have a custom action if a warning/fault bit was set.

timwegenerDE, 2022-09-04:
Got it, thanks for your explanation.

I think in this case a grott extension would be the way to go. Unfortunately my Python skills are VERY limited and I‘m not even sure what should happen in an alarm- /warning- event. My first thought was to poll the API (if these information would be found there) and setting a virtual contact sensor in my smart home. However I think knowing of potential problems with the growatt is an essential feature for everyone and should be implemented more generic. The most generic would be an e-mail, but if every data transmission sends an errorcode that would spam a lot…

johanmeijer, 2022-09-04:
We can think about an extension (or addition to Grott) if useful or API extension. 

I am not totally sure in the 3 years I have to Growatt inverters I never have a fault code (but maybe to comes with the years). 

In case of a API how do you expect to sent the query? If you now only using PVOutput you should have a more "intelligent" platform to do that. 

If we create an extension would a message to e.g. Telegram being sufficient? Mail is always difficult because of security challenges. More and More mail app checks the originator of the mail (and with a reason). It might be difficult to get Grott running on your own environment being trusted. 

What you can do at this moment use nodered. Nodered can read the MQTT message and filter the Fault codes and give alerts (to almost every app/platform you want). 

It will a challenge to translate the error codes (Hex/Binary) in the data record to the growatt error codes I think.

timwegenerDE, 2022-09-05:
So far I didn‘t get any error neither, but sometime this will happen for sure :-P

For me personal a simple http get would be sufficient with json / xml return. Maybe something like: 
lastMessageHadError: yes/no
errorsLast24Hours: {<Array of Errorcodes>}

I understand that mail is a bit outdated and also hard to configure (SPF and so on), but for me it would be the easiest way. However, maybe we will get some feedback here from other users what they would like? Telegram is nothing Iam using but it seems to be the new messaging standard.

A translation for the errorcodes would be nice, but not necessary for me in the first step. The most important for me is to know that there is ANY error.

rany2, 2022-09-05:
If you use influx you could easily create a statement that tells you if an error occured in last n time. I think something like

```
from (bucket: "grott" )
|> range (start: -24h) 
|> filter (fn : (r) => r["_measurement"] "xxx")
|> filter (fn: (r) => r["_field '"] "faultBit" or r["_field"] "warningBit")
|> aggregateWindow (every: y.windowPeriod, fn: max, create Empty: false)
|> yield (name : "max")
```

If one of the values influx returns is greater than 0, then a warning/fault occured. I don't think there is a need for this to be part of grott, especially because of the increase in complexity. I don't think it's feasible anyway

timwegenerDE, 2022-09-06:
I have installed influxdb and and getting the data from grott. Also I testet your query, of course without a result. First of all I can't find thie two fields faultBit and warningBit? Also I can't find them in the grott log json?

rany2, 2022-09-06:
Did you change `xxx` with your inverter ID?

timwegenerDE, 2022-09-06:
Sure.

rany2, 2022-09-06:
What inverter do you use? I guess that your inverter has a different field for faults

timwegenerDE, 2022-09-06:
Growatt 3000S

That's what in the Growatt log json message to influx:
`[{'measurement':'X','time':'2022-09-06T11:35:57','fields':
{'datalogserial':'X','pvserial':'X','pvstatus':1,
'pvpowerin':X,'pv1voltage':X,'pv1current':X,'pv1watt':X,
'pv2voltage':X,'pv2current':X,'pv2watt':0,'pvpowerout':X,
'pvfrequentie':X,'pvgridvoltage':X,'pvgridcurrent':X,
'pvgridpower':X,'pvgridvoltage2':X,'pvgridcurrent2':X,
'pvgridpower2':X,'pvgridvoltage3':X,'pvgridcurrent3':X,'pvgridpower3':
X,'pvenergytoday':X,'pvenergytotal':X,'totworktime':X,
'pvtemperature':X,'pvipmtemperature':X,'epv1today':X,'epv1total':`

rany2, 2022-09-06:
Mhm, not sure which field is being used by the application for alerts but it has to be one of them. Maybe try ones with `fault` in the name?

timwegenerDE, 2022-09-06:
Sorry, updated in comment: but no "fault" field.

johanmeijer, 2022-09-06:
@timwegenerDE which layout is being used by Grott(eg T06nnnn). I can determine if a fault code is available in the record (and add to the layout if necessary).

timwegenerDE, 2022-09-06:
I see some different messages in the log:
	 - layout   :  T065104 _<or T065103 or T065129 or T065104 maybe even more?>_
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNN

johanmeijer, 2022-09-06:
You can specify includeall = True in grott.ini [growatt] section and you wil get more info in the mqtt message (and inluxdb).

timwegenerDE, 2022-09-07:
After setting the docker env. var `gincludeall:True` there are some new fields in the record:
- vpvfault
- vacfault
- facfault
- tmpfault
- faultcode

The faultcode sound promising for the alert I think? 

Now my biggest problem is dealing with influxdb and how to get these alert in any of my other environments; But that's not your business ;-)

I think it's a bit (to) complicated for "just" getting an potential errorcode.
Maybe you could think about a easier way inside of grott.

However, thank you very much.

closed 2022-09-07 by timwegenerDE