johanmeijer/grott #157: PVoutput error: Bad request 400: Energy value [2147483647] too high for system size

rybber78, 2022-09-08:
New install. Will not upload to PVoutput because of this error. System size is 9360W in PVoutput.

Grott Docker log:

  
      
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47

		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74

		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x36\x0e\x47\x72\x6f\x77

		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72

		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x1c\x35

	 - Grott automatic protocol detection

	 - Grott data record length 585

	 - layout   :  T060104X

	 - no matching record layout found, try generic

	 - Record layout used :  T06NNNNX

	 - Growatt data decrypted V2

	 - Grott Growatt data decrypted

	 - Growatt plain data:

		 00300006024101045847443643424430395a00000000000000000000000000000000000000005

		 24b483543413730424100000000000000000000000000000000000000001609080a1226020000

		 007c00010000436d1117000900000f610df30023000030d200000000000000000000000000000

		 00000000000000000000000000000000000000000000000000000000000000000000000426b13

		 8d093e00180000162e09140018000015c909160018000015ce0ff00fc70fbc000000120000005

		 70000f93500000005000000250000000a0000002e000000000000000000000000000000000000

		 000000000000000000000000000000000000000000000000000000000000000000530277018e0

		 1830000019f0c4c0c344e2000000001b206000000000000000000000000000000000000000f00

		 1e0000000000000000000000000000000000000000007d00f9000000000000000000000000000

		 00000000000000000000000000000000000000000000000000000000000000000000000000000

		 00000000000000000000000000000000000000000000000000000000000000000000000000000

		 0000000000000000000000000000000000000000000000000000000000000000000045749b600

		 00000000160009000700160009000800000000000029db0086000b007a0000000000000000000

		 00000000000000000000000000000000000000000000000000000000000000000000000000000

		 000000000000427a0000000000000000000000000000000000000000000000000000000000000

		 000000000006e5a

	 - Growatt new layout processing

		 - decrypt       :  True

		 - offset        :  6

		 - record layout :  T06NNNNX

	 - Grott data record date/time processing started

	 - date-time:  2022-09-08T10:18:38

	 - Grott values retrieved:

		 -  pvserial             :  RKH5CA70BA

		 -  recortype1           :  0

		 -  recortype2           :  124

		 -  pvstatus             :  1

		 -  pvpowerin            :  1726.1

		 -  pv1voltage           :  437.5

		 -  pv1current           :  0.9

		 -  pv1watt              :  393.7

		 -  pv2voltage           :  357.1

		 -  pv2current           :  3.5

		 -  pv2watt              :  1249.8

		 -  pvpowerout           :  0.0

		 -  pvfrequentie         :  0.0

		 -  pvgridvoltage        :  0.0

		 -  pvgridcurrent        :  0.0

		 -  pvgridpower          :  0.0

		 -  pvgridvoltage2       :  0.0

		 -  pvgridcurrent2       :  0.0

		 -  pvgridpower2         :  0.0

		 -  pvgridvoltage3       :  0.0

		 -  pvgridcurrent3       :  0.0

		 -  pvgridpower3         :  111431361.3

		 -  totworktime          :  218.5

		 -  pvenergytoday        :  36582603.2

		 -  pvenergytotal        :  26470393.2

		 -  epvtotal             :  1.8

		 -  epv1today            :  8.7

		 -  epv1total            :  6379.7

		 -  epv2today            :  0.5

		 -  epv2total            :  3.7

		 -  pvtemperature        :  63.1

		 -  pvipmtemperature     :  41.5

	 - MQTT jsonmsg: 

			 {"device": "RKH5CA70BA", "time": "2022-09-08T10:18:38", "buffered": "no",

			 "values": {"recortype1": 0, "recortype2": 124, "pvstatus": 1, "pvpowerin":

			 17261, "pv1voltage": 4375, "pv1current": 9, "pv1watt": 3937, "pv2voltage":

			 3571, "pv2current": 35, "pv2watt": 12498, "pvpowerout": 0, "pvfrequentie":

			 0, "pvgridvoltage": 0, "pvgridcurrent": 0, "pvgridpower": 0,

			 "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0,

			 "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 1114313613,

			 "totworktime": 1572864, "pvenergytoday": 365826032, "pvenergytotal":

			 264703932, "epvtotal": 18, "epv1today": 87, "epv1total": 63797, "epv2today":

			 5, "epv2total": 37, "pvtemperature": 631, "pvipmtemperature": 415}}

	 - MQTT message message sent

	 - Grott send data to PVOutput systemid:  92521 for inverter:  RKH5CA70BA

		 -  {'X-Pvoutput-Apikey': '85599034902fda2c509de518301e51510ae63b06', 'X-Pvoutput-SystemId': '92521'}

		 -  {'d': '20220908', 't': '10:18', 'v1': 36582603200, 'v2': 0.0, 'v6': 0.0}

	 - Grott PVOutput response: 

		 -  Bad request 400: Energy value [2147483647] too high for system size [9360]

	 - Grott Send data to Influx disabled 

	 - Grott extension processing disabled 

	 - Growatt packet received:

		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.18.0.28', 5279), raddr=('10.0.1.136', 12248)>

	 - Data less then minimum record length, data not processed

	 - Growatt packet received:

		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.18.0.28', 39962), raddr=('47.91.67.66', 5279)>

	 - Data less then minimum record length, data not processed

	 - Growatt packet received:

		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.18.0.28', 5279), raddr=('10.0.1.136', 12248)>

	 - Data less then minimum record length, data not processed

johanmeijer, 2022-09-08:
Indeed a high value. What type of inverter do you have?

johanmeijer, 2022-09-08:
I also think that is the issue. That is the reason why I askedfor inverter type  

You have probably to specify invtype = sph (or spf) in grott.ini or -e ginvtype= "sph" in docker run command.

rybber78, 2022-09-08:
Yes, adding the following to docker-compose.yml fixed it:
      - ginvtype=sph
 

But I have another problem. Timestamps are wrong. I have Europe/Amsterdam in docker compose. GMt+1 in PVoutput settings, but  there is a 6 hour difference from what is sent by Grott?

{'d': '20220908', 't': '07:44', 'v1': 12000, 'v2': 3016.3, 'v6': 235.5}

(local time is 13:44)

johanmeijer, 2022-09-08:
Is it possible these are buffered records and the inverter try to catch up?

In you record above the time was 10:18 seems to be okay for me.

As far I can see the time in the data record (from the inverter) is being used.

johanmeijer, 2022-09-08:
You can detect if it is a buffered record if in the mqtt Json message buffered : yes is shown.

Challenge with pvoutput is that  if there are to much buffered records being sent you reach very fat the limit of 60 message / hour. 

That probably only once. But if it stays the you can disable buffer processing by grott by specifing gsendbuf = "False".