johanmeijer/grott #159: Home Assistant new MQTT format, sensors remain unknown?

gijs007, 2022-09-09:
Home Assistant is changing the configuration file format for the MQTT sensors. Instead of using the sensor platform key, we need to use the MQTT sensor platform key. As per: https://www.home-assistant.io/integrations/sensor.mqtt/

I've rewritten the old config file from: https://github.com/johanmeijer/grott/wiki/Home-Assistant-configuration-examples

Configuration.yaml contains:
`mqtt: !include_dir_merge_list mqtt/`

Created the following directory and file: mqtt/growatt_nl.yaml.
With the following content:

`# Grott - Home Assistant Growatt sensors 
# 
# In dit yaml bestand worden alle sensoren die door Grott worden aangeboden, beschikbaar gesteld in HA, inclusief een dummy sensor om ook het type van de inverter
# in HA beschikbaar te maken (Let op, dit moet handmatig gevuld worden). 

sensor:
  - state_topic: energy/growatt
    value_template: "{{ value_json['device'] }}" 
    unique_id: growatt_serial
    name: Growatt - Serienr
    icon: mdi:select-inverse

  - state_topic: energy/growatt
    # Als je de datum in een ander formaat wil tonen, pas dan dit deel aan "timestamp_custom('%d-%m-%Y')"
    # Voor meer informatie, zie: https://docs.python.org/3/library/time.html#time.strftime
    value_template: "{{ as_timestamp(strptime(value_json['time'], '%Y-%m-%dT%H:%M:%S')) | timestamp_custom('%d-%m-%Y') }}" 
    unique_id: growatt_date
    name: Growatt - Datum
    icon: mdi:calendar

  - state_topic: energy/growatt
    # Als je de datum in een ander formaat wil tonen, pas dan dit deel aan "timestamp_custom('%H:%M:%S')"
    # Voor meer informatie, zie: https://docs.python.org/3/library/time.html#time.strftime
    value_template: "{{ as_timestamp(strptime(value_json['time'], '%Y-%m-%dT%H:%M:%S')) | timestamp_custom('%H:%M:%S') }}" 
    unique_id: growatt_time
    name: Growatt - Tijd
    icon: mdi:clock-digital

  - state_topic: energy/growatt
    value_template: >
      {% if (value_json['values']['pvstatus'] | int == 0) %}
        Wachtend
      {% elif (value_json['values']['pvstatus'] | int == 1) %}
        Normaal
      {% elif (value_json['values']['pvstatus'] | int == 2) %}
        Fout
      {% else %}
        Onbekend
      {% endif %}
    unique_id: growatt_status
    name: Growatt - Status
    icon: mdi:power-settings

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pv1watt'] | int / 10 }}" 
    unique_id: growatt_string1_watt
    device_class: power
    unit_of_measurement: "W"
    name: Growatt - String 1 (Watt)

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pv1voltage'] | int / 10 }}" 
    unique_id: growatt_string1_voltage
    device_class: voltage
    unit_of_measurement: "V"
    name: Growatt - String 1 (Voltage)

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pv1current'] | int / 10 }}" 
    unique_id: growatt_string1_current
    device_class: current
    unit_of_measurement: "A"
    name: Growatt - String 1 (Stroom)

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pv2watt'] | int / 10 }}" 
    unique_id: growatt_string2_watt
    device_class: power
    unit_of_measurement: "W"
    name: Growatt - String 2 (Watt)

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pv2voltage'] | int / 10 }}" 
    unique_id: growatt_string2_voltage
    device_class: voltage
    unit_of_measurement: "V"
    name: Growatt - String 2 (Voltage)

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pv2current'] | int / 10 }}" 
    unique_id: growatt_string2_current
    device_class: current
    unit_of_measurement: "A"
    name: Growatt - String 2 (Stroom)

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvpowerin'] | int / 10 }}" 
    unique_id: growatt_actual_input_power
    device_class: power
    unit_of_measurement: "W"
    name: Growatt - Input wattage (Actueel)

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvpowerout'] | int / 10 }}" 
    unique_id: growatt_actual_output_power
    device_class: power
    unit_of_measurement: "W"
    name: Growatt - Output wattage (Actueel)

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvfrequentie'] | int / 100 }}" 
    unique_id: growatt_grid_frequency
    unit_of_measurement: "Hz"
    name: Growatt - Gridfrequentie
    icon: mdi:waveform

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvgridvoltage'] | int / 10 }}" 
    unique_id: growatt_phase_voltage
    device_class: voltage
    unit_of_measurement: "V"
    name: Growatt - Fase voltage

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvenergytoday'] | int / 10 }}" 
    unique_id: growatt_generated_energy_today
    device_class: energy
    unit_of_measurement: "kWh"
    name: Growatt - Opgewekte energie (Vandaag)
    icon: mdi:solar-power

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvenergytotal'] | int / 10 }}" 
    unique_id: growatt_generated_energy_total
    device_class: energy
    unit_of_measurement: "kWh"
    name: Growatt - Opgewekte energie (Totaal)
    icon: mdi:solar-power

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvtemperature'] | int / 10 }}" 
    unique_id: growatt_inverer_temperature
    device_class: temperature
    unit_of_measurement: "°C"
    name: Growatt - Inverter temperatuur

# Onderstaande entiteit wordt niet door alle inverters ondersteund, dus is mogelijk niet beschikbaar.

  - state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvipmtemperature'] | int / 10 }}" 
    unique_id: growatt_ipm_temperature
    device_class: temperature
    unit_of_measurement: "°C"
    name: Growatt - IPM temperatuur
`
For some reason the sensors are not being updated, and have an unavailable status.
It's probably some kind of syntax error, which I'm overlooking, since the energy/growatt topic is still receiving periodic data. 

Update:
I've also tried the new sensors_growatt_nl.yaml config from the latest commit:
https://github.com/johanmeijer/grott/commit/e7e2cf59c06348d4814276081d5b3bf5c7adf4c7
However this seems to have the same problem as my own config rewrite. Sensors shown a status of unavailable, even though data is being received by MQTT.

Besides this, If I'm not mistaken, the wiki documentation (https://github.com/johanmeijer/grott/wiki/Home-Assistant-configuration-examples) needs an update. It still refers to the old format with a `sensor:` key instead of `mqtt:` key.

osos, 2022-09-11:
From my configuration.yaml:
```
mqtt:
  discovery: true
  sensor: !include_dir_merge_list mqtt-sensor
```

This is my sensors_growatt.yaml file (located in the mqtt-sensor directory), optimised for hybrid inverters (profile: t06NNNNXSPH) and includes more or less all parameters with icons etc. Should be possible to use with many other profiles and inverters.

```
# Grott - Home Assistant Growatt sensors 
# 
# This file exposes all sensors from Grott to HA, including dummy sensors for the type of the inverter and the type and serial number of the datalogger (Be aware, the dummy
# sensors have to be set manually) 

#- platform: template
#  sensors:
#    growatt_inverter:
#      unique_id: growatt_invertertype
#      friendly_name: Growatt - Type
#      # Please set the type of your inverter
#      value_template: "SPH3600"
#      icon_template: mdi:select-inverse
#
#- platform: template
#  sensors:
#    growatt_datalogger_type:
#      unique_id: growatt_dataloggertype
#      friendly_name: Growatt - Datalogger type
#      # Please set the type of your datalogger
#      value_template: "ShineWIFI-S"
#      icon_template: mdi:select-inverse

- state_topic: energy/growatt
  unique_id: growatt_datalogger_serial
  name: Growatt - Datalogger serial number
  # Please set the serial number of your datalogger
  value_template: "{{ value_json['values']['datalogserial'] }}"
  icon: mdi:select-inverse

- state_topic: energy/growatt
  value_template: "{{ value_json['device'] }}" 
  unique_id: growatt_serial
  name: Growatt - Serial number
  icon: mdi:select-inverse

- state_topic: energy/growatt
  # If you like to have the date in another format, please change "timestamp_custom('%d-%m-%Y')"
  # For more information: https://docs.python.org/3/library/time.html#time.strftime
  value_template: "{{ as_timestamp(strptime(value_json['time'], '%Y-%m-%dT%H:%M:%S')) | timestamp_custom('%d-%m-%Y') }}" 
  unique_id: growatt_date
  name: Growatt - Date
  icon: mdi:calendar

- state_topic: energy/growatt
  # If you like to have the date in another format, please change "timestamp_custom('%H:%M:%S')"
  # For more information: https://docs.python.org/3/library/time.html#time.strftime
  value_template: "{{ as_timestamp(strptime(value_json['time'], '%Y-%m-%dT%H:%M:%S')) | timestamp_custom('%H:%M:%S') }}" 
  unique_id: growatt_time
  name: Growatt - Time
  icon: mdi:clock-digital

- state_topic: energy/growatt
  value_template: >
    {% if (value_json['values']['pvstatus'] | int == 0) %}
      Waiting
    {% elif (value_json['values']['pvstatus'] | int == 1) %}
      Normal
    {% elif (value_json['values']['pvstatus'] | int == 2) %}
      Fault
[... 275 lines cut ...]
  value_template: "{{ value_json['values']['pactogridr'] | float / 10000 }}"
  unique_id: growatt_export_to_grid
  device_class: power
  unit_of_measurement: "kW"
  name: Growatt - Export to grid (actual)
  icon: mdi:transmission-tower-import

- state_topic: energy/growatt
  value_template: "{{ value_json['values']['etogrid_tod'] | float / 10 }}"
  unique_id: growatt_export_to_grid_today
  device_class: energy
  unit_of_measurement: "kWh"
  name: Growatt - Export to grid (today)
  icon: mdi:transmission-tower-import

- state_topic: energy/growatt
  value_template: "{{ value_json['values']['etogrid_tot'] | float / 10 }}"
  unique_id: growatt_export_to_grid_total
  device_class: energy
  state_class: total_increasing
  unit_of_measurement: "kWh"
  name: Growatt - Export to grid (total)
  icon: mdi:transmission-tower-import

```

johanmeijer, 2022-09-11:
I do not have HA running at this moment (winter project?) so I totally rely on you users of HA for configuration and documentation. 

Please let me know if I have to change the examples and wiki information.

gijs007, 2022-10-09:
The config from OSOS is working, I did have to change some things (energy dashboard, some dashboards) as I was using a config with Dutch entity names...
I think it's better to just stick with the English language in the config, for the entity names at least. As we can always manually override the entity name in Home Assistant.

ZonPV, 2023-01-13:
Yesterday I download the latest (stable) version of HomeAssistant for Docker on my Synology. 
I am new to HomeAssistant. So studied the 2 examples in the Wiki but got lost because of different methods and choices.
Is there a full manual how to get the data from the broker to HA? Or can someone give me a hint in the right direction?

The configuration.yaml is still default.

Setup is a MIC 2000TL-X inverter and ShineWifi-X which sends data to Grott-server. Can read the data in MQTT Explorer and Grott uploads to PVoutput.org, Influxdb and Domoticz.

Thnx a lot!

catch56, 2023-02-09:
I've started a PR to update the configuration examples, only found this issue after doing that and haven't actually got grott working with HA yet so could still be some mistakes in there: https://github.com/johanmeijer/grott/pull/280

NosNL, 2023-02-14:
The config from OSOS is working, .. i have changed voor the mod 5000 (two string) ..
tomorrow testing :-) 

Thanks OSOS