johanmeijer/grott #165: Growatt SPH5000TL3 BH/ layout record

Windwind06, 2022-09-18:
Hi Johann,
I've got a problem getting the hybrid parameter from the SPH5000TL3 BH, such as SOC load etc.
Everything with the pv* values is ok, but the mapping is automatic done with T060136X layout and not with a correct one.
What can I do?
Another question: I user grafana and influxdb (v1.7), but there es only one exporting data (the name of the inverter), no values at all.
What can I do? Is Influxdb v2 the only solution?
Thanks
Windwind06

Windwind06, 2022-09-19:
Hi,
I put the invtype=sph in grott.ini and now I get the extended shp values. Is influxdb v2 necessary for exporting data from grott to influxdb?
Thanks

johanmeijer, 2022-09-19:
Grott support both influxdb 1.x and v2. See:

https://github.com/johanmeijer/grott/wiki/InfluxDB-Support

closed 2023-08-13 by johanmeijer