johanmeijer/grott #166: Concatenating consumptiondata

parhedberg, 2022-09-19:
Hello and thanks for the great work!

I am using a Growatt inverter, but no smart meter. So Growatt itself does not know how much i consume (but it knows my production). Here in Sweden we will (soon enough) get our electrical meter changed to a new one that we can take realtime readings off.. 

Would it be possible to implement something so maybe grott listens to consumption values over MQTT and then concatenate this with production values before sending the data to PVOutput? (one point of sender to pvoutput instead of trying to build another function for that)...

johanmeijer, 2022-09-19:
Maybe that is possible but I think there will be a lot of different meters that then might be needed to support.

I read the values from my electrical with a esp8266 and write it to mqtt. I use nodered to read that mqtt value and sent it to pvoutput. You can sent 2 messages to pvoutput (grott solar and electrical usage) and pvoutput will combine them. 

Grott supports the growatt sdm630 connected to the inverter.

parhedberg, 2022-09-19:
Thnx @johanmeijer, that might be an easy enough solution..

Cloding this ticket..

closed 2022-09-19 by parhedberg