johanmeijer/grott #167: Direct input with rs485

delchrys, 2022-09-21:
Is it possible grott can work with direct input from the rs485 port?
Saw a other project of a rs485-ttl-esp8266-mqtt parser.
Wouldn it be great if that could be send to grott server and then posted to all the options grott has..... And with a plus send to growatt server....
That way we can have as many data as we want by directly reading the rs485 port

johanmeijer, 2022-09-22:
Lets see what is possible. See also issue #168.

delchrys, 2022-09-23:
Yeah I've seen that..lol

catch56, 2023-02-09:
https://github.com/mbs38/spicierModbus2mqtt looks relevant here too.

It's very convenient being able to run grott on existing hardware with no new requirement (especially with supply issues), but a smooth transition from proxying/intercepting the shine dongle to replacing it altogether would be amazing.

My SPH6000 also has a USB socket (which is what the proprietary solarassistant, which outputs mqtt too, can use in addition to rs845), so if the same output can be got from USB, it might allow running grott and the dongle side by side (at least for evaluation).

skinkie, 2023-02-09:
I have a very good configuration for spicierModbus2mqtt but there is a very relevant issue. The RS485 output of Growatt devices is for some of the properties completely wrong. I have an open ticket at Growatt for these issues. So expect the default stuff to work, I can share you a nice csv for the SPH10K and 6000UE, but be aware...

catch56, 2023-02-09:
A couple of other references:

https://github.com/Lyve1981/GrowattUsbModbus - grabs the modbus from the USB port (rather than the RS232 port).

https://github.com/otti/Growatt_ShineWiFi-S - firmware replacement for the dongle.

johanmeijer, 2023-02-12:
I have chosen to keep using the wifishine Lan stick while the inverter is at a location where it is not easy to connect to a computer with a  Serial cable. 

I have also everything at home to build my own wifi stick (an ESP8266 and RD485 interface) but have no time to build it (also because maintaining Grott takes up a lot of my free time). 

I you see other solutions that works better then Grott of course  you should use those,  but please do not open issues here for that.