johanmeijer/grott #168: Integration with growatt2mqtt

nygma2004, 2022-09-22:
Hi Johan,

Last year I built a project that connects to Growatt inverters on the RS485 port and reads status via modbus and pushes the data to MQTT. You can check my project here: https://github.com/nygma2004/growatt2mqtt

I got a question of it would be possible to integrate this project with yours: https://github.com/nygma2004/growatt2mqtt/issues/9

I was not aware of your project before, but if you have some time, maybe we can work on it together if you see the benefit.

Cheers,
Csongor

johanmeijer, 2022-09-22:
Hi Csongor (@nygma2004 ),

A good idea to see if we can integrate out projects. 

Do you already have an idea how and what? 

I am going to look at your project. It was already on my todo list to see if I can read directly from the inverter with an esp. You already did it that saves time 😉

Regards, 
Johan

nygma2004, 2022-09-22:
I am in a middle of a big project, so I only took a 2 minute view so far. I also have a few pull requests so I will need to work on this probably next month. I prepare the data in JSON that gets posted to MQTT, I can make that available on a HTTP endpoint to be pulled, or can also be pushed via HTTP as well. So there are some options.