johanmeijer/grott #175: Question: home assistant solar panels

mechoriet, 2022-09-29:
Hello Johan,

how is it possible to add my grott->mqtt solar panels to the Energy tab in home assistant
cause I don't see it in the list or is there something I need to do that I missed in the wiki using the eng sensors cause the dutch ones didn't work for some reason no error in the home assistant docker on startup
[image]

mechoriet, 2022-09-29:
sorta solved the problem of not having the sensor available to select with https://github.com/johanmeijer/grott/issues/70

but now when my min 2500 turn off cause power is low home assistant keeps for some reason its last value at the solar of the energy page is there a workarround/solution for that

johanmeijer, 2022-09-29:
Can you read dutch? 

At the tweakers forum the last pages of this thread:

https://gathering.tweakers.net/forum/list_message/72953382#72953382 

is about the HA configuration together with Grott. Something will (has)change in HA I do understand. I have to change the wiki and samples with this but does not run HA myself so it is a little bit difficult.

mechoriet, 2022-09-29:
yes I can read dutch

seems like the data is now getting recorded "normally" without implementing the config changes that Mr Magic made to convert it to the new format

but now after the Inverter(Growatt 2500 TL-XE) has turned off(cause its powered of the panels) it keeps the solar output at 0.6kwh which messes with how the Energy distribution is shown
but the energy chart now in the evening 7pm till 10pm is reporting it right  

[image]

So I dunno if its something I have to live with or it could also be something with HA being messed by trying to get to work :)

jupppo, 2022-11-11:
You need to state the following in mqtt sensor definition in order to use this properly:

    state_class: "total_increasing" 
instead of
    state_class: "measurement"

Thats wrong in the mqtt templates.