johanmeijer/grott #176: Question: Can Grott be used to set inverter modes?

sumitkm, 2022-09-29:
Apologies if the question makes no sense. I am actually waiting for my Growatt SPH5000 installation in the near future. I was looking up all the integration options (with HA) and chanced upon Grott. Excellent piece of hacking and a fabulous project overall. Congratulations. 

Just wondering if all data is readonly or can I do things send commands to the the inverter to not charge battery at a certain time or not discharge from the battery at certain time? I am mostly concerned about battery drain at night by EV. So I was hoping I could write some automations where battery output is restricted when EV is doing a timed boost. 

I have a myEnergie Zappi, so I can control the charger side. If Grott doesn't support write commands it should still be fine, I can do some jiggery-pokery to get the EV to share 'cheap electricity rates' at night even if it drains part of the battery.

johanmeijer, 2022-10-01:
Grott in standard proxy mode does not change anything in the communication between inverter and growatt . That means also commands that are sent from the growatt site will be sent to the inverter. You can use the blockcmd command parameter to protect the inverter from changes (commands) from China.

At this moment grott in proxy mode does not provide a interface to give commands (yet). 

Under development is grottserver (beta) which replace the growatt Internet server and provides an api to read an set inverter/datalogger settings (see discussions #98).

In the next version (V3) this will be integrated.

Grott can write directly  to mqtt, pvoutput.org an influxdb (v1.x and 2.x). Next to you can build your own extension for other integrations (example for excel/csv an httpserver are available).

I try to develop grott to be usable for everybody (within boundaries) so good ideas are always welcome.

sumitkm, 2022-10-03:
@johanmeijer awesome! Thanks for the patient and detailed response. The blockCmd is epic.

I hope v2 will continue to be maintained, I might even raise my hand to do it. This is because my Growatt Server comes with a 10 year warranty which I am sure they will refuse to acknowledge if I block them from getting the data. So when the merge happens can we use it as a one way feed to Growatt, so they can't claim server is unable to reach plant? 

Come to think of it I could just swap the URL in the wireless dongle they provide, if I ever had to make warranty claims 😈 

Looking forward to v3 APIs as well :)

closed 2022-10-03 by sumitkm

Sdahl1234, 2022-10-03:
I dont think your 10 years warranty has anything to do with them getting your data. What if you didnt buy a wifi-module???