johanmeijer/grott #179: Lots off errors in log

RienduPre, 2022-10-01:
Hi,

I added one of mine Solar inverters (Model MIC 2500TL-X)

The connection is made, but it looses connection again with lots of error in the logging (see below)

I'm using Grott server 2.6.1g

Any help is more than welcome.

`Grott proxy mode started
Hostname : grott-server
IP :  192.168.0.170 , port :  5279 

         - ('192.168.0.54', 30182) has connected

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 5279), raddr=('192.168.0.54', 30182)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 5279), raddr=('192.168.0.54', 30182)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x07\x00\x06\x02\x41\x01\x03\x1f\x35\x2b\x42\x23\x33\x38\x71\x25\x2d\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x2c\x3a\x26\x40\x36\x03\x45\x5f\x45\x26\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x77\x7e\x75\x4b\x40\x41\x75\x61
                 \x74\x74\x3b\x72\x6e\x77\x78\x74\x74\x47\x16\x6f\x77\x46\x64\x74\x47\x13\xc7
                 \x74\x89\x33\x3c\x76\x5c\x5f\x77\x3b\x35\x35\x06\x72\x64\x77\x60\x74\x74\x44
                 \x52\x6f\x4b\x61\x48\x67\xcf\x61\xe7\x77\x61\x37\x39\x00\x46\x2d\x33\x56\x44
                 \x46\x00\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
                 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x63\x31\x76\x75\x40\x94
                 \x6f\x7d\x61\x75\x74\x4b\x72\x58\x77\x7b\x74\x72\x40\x42\x65\x22\x73\xfa\x60
                 \x59\x75\x5f\x7d\x34\x66\xfa\x53\x6c\x6f\x77\x61\x74\x74\x47\x72\x6f\x70\xc2
                 \x7d\x96\x55\xe1\x7c\xfa\x61\x46\x74\x75\x72\x5d\x77\x53\x74\x46\x47\x40\x6f
                 \x45\x61\x46\x74\x75\x72\x5d\x77\x61\x74\x74\x4e\x90\x6f\x77\x26\x3c\x35\x06
                 \x42\x5a\x42\x51\x54\x54\x76\x43\x6e\x46\x61\x74\x74\x47\x61\xf3\x77\x7b\x7d
                 \xc0\x4e\x90\x67\x33\x69\x62\x74\x53\x72\x6a\x7e\x0e\x7c\x88\x60\x62\x48\x67
                 \x46\x64\x53\x57\x55\x7f\x50\x71\x74\xe2\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
                 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x75\x47\x72\x6f\x78\x60\x74\x6d
                 \x47\x72\x6f\x77\x61\x74\x7f\xff\x7e\x5b\x77\x61\x37\x39\x00\x46\x2d\x33\x56
                 \x44\x46\x00\x42\x5f\x47\x51\x44\x44\x77\x42\x5f\x47\x51\x44\x44\x77\x42\x5f
                 \x47\x51\x44\x44\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x6d\x35
[... 61 lines cut ...]

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.170', 39966), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed
`

johanmeijer, 2022-10-01:
Data less then minimum record length, data not processed is not on error. This means that grott skips ti's record because it is not a data record to process.

I am not sure if and why you loose connection. It is not shown in this part of the log.

RienduPre, 2022-10-02:
Hi Johan,

Thanks for the quick response.
The connection problem were caused due to a duplicate ip adres and if the other issues about the record length are 'normal' I will close this issue.

I have a question about connecting 2 Growatt inverters, but I will create a new issue for that.

Thanks,
Rien

closed 2022-10-02 by RienduPre