johanmeijer/grott #182: PVOutput Energy not reset at night

toaomatis, 2022-10-03:
I have recently installed a new PV system with a **Growatt MOD 4000TL3-X** converter.

For two days it [went fine](https://www.pvoutput.org/list.jsp?id=79546&sid=93706), but this morning the PVOutput Energy is not reset to 0: https://www.pvoutput.org/intraday.jsp?id=79546&sid=93706&dt=20221003

I restarted the Docker container, but no success. I do see correct values in my Growatt dashboard: 
[image]

My `docker-compose.yaml` looks like this:
```yaml
version: '3.3'
services:
    grott:
        image: ledidobe/grott
        container_name: grott
        restart: unless-stopped
        ports:
            - "5279:5279"
        environment:
            - ginvtype=tl3
            - gblockcmd=False 
            - gnomqtt=True
            - gmqttip=localhost
            - gmqttauth=False
            - gmqttuser=growatt
            - gmqttpassword=growatt2020
            - gpvoutput=True
            - gpvapikey=....
            - gpvinverters=1
            - gpvinverterid1=systemidX
            - gpvsystemid1=93706
            - ginflux=False
            - ginflux2=False
            - gifdbname=grottdb
            - gifip=localhost
            - gifport=8086
            - gifuser=grott
            - gigpassword=growatt2020
            - giftoken=influx_token
            - giforg=grottorg
            - gifbucket=grottdb
            - gextension=false
            - gextname=grottext
            - 'gextvar = {"ip": "192.168.0.47", "port": "8000"}'
            - TZ=Europe/Amsterdam 
```

toaomatis, 2022-10-03:
I exported all the logs and it turns out the first line send to PVOutput this morning was not correct for the `v1` field:
```
	 - Grott send data to PVOutput systemid:  93706 for inverter:  XXX
		 -  {'X-Pvoutput-Apikey': '....', 'X-Pvoutput-SystemId': '93706'}
		 -  {'d': '20221003', 't': '07:46', 'v1': 10800, 'v2': 0.0, 'v6': 234.9}
	 - Grott PVOutput response: 
		 -  OK 200: Added Status
```
The following update to PVOutput seems correct, but PVOutput already registered the `10800` and therefore (I assume) does not allow any lower values for that day.

```
	 - Grott send data to PVOutput systemid:  93706 for inverter:  XXX
		 -  {'X-Pvoutput-Apikey': '....', 'X-Pvoutput-SystemId': '93706'}
		 -  {'d': '20221003', 't': '07:51', 'v1': 0, 'v2': 0.0, 'v6': 239.3}
	 - Grott PVOutput response: 
		 -  OK 200: Added Status		 
```

johanmeijer, 2022-10-03:
Ok this a tricky one because the inverter has not updated (cleans) it totals and sent wrong value when it starts in the morning. 

Do you use the time of the server (time= server) or the time in the datarecord (default)? If you use time in the datarecord it is really a inverter /datalogger error and it will be hard to correct this with Grott (do we always have to filter the first record, do we have to filter x first records or only one, how do I detect this is the first record of the day). 

Does this happens often? Can you perform an upgrade of your datalogger (shinewifi-x?). Maybe that helps.

toaomatis, 2022-10-04:
Yeah I know it's a tricky one. I have a Growatt Shinelink-X LANbox and can send you the complete logging of the last days.

According to the Grott server logging, it does not capture the Box' timestamp, but uses the servers. 
```
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNXTL3

	 - Grott data record date/time processing started
	 - no or no valid time/date found, grott server time will be used (buffer records not sent!)
```

johanmeijer, 2022-10-05:
If you can give some log data arround the time it goes wrong that will help (especially the growatt original data and the processing after that)

johanmeijer, 2022-10-06:
Indeed do not share your pvoutput credentials with the rest of the world ;)

You can sent it to grott@ledidobe.nl or does github provide a private channel?

johanmeijer, 2022-10-06:
Thinking about it  I will remove that information from the logging in the next release.

toaomatis, 2022-10-06:
As far as I know, GitHub does not provide a private channel (at least not for free or medium tiers). 
I already approached you via Tweakers, but I can send it via mail as well. 

Seems like a good idea to remove the sensitive information when logging :+1:

johanmeijer, 2022-10-06:
Tweakers is ok

johanmeijer, 2022-10-06:
@toaomatis Thanks for the log.

This is a nasty one to solve while the shinelink does not put datetime in the record and grott can not see if this record a valid one or not. 

What we can try is to use the functionality of grott that disable sendig the day totals to PVOoutput. PVoutput now has to calculate the totals itself. This was designed for somethings else (show better / more understandable  averages) but maybe it works for you also: 

You can enable this by: 

 - Add in grott ini:  [PVOutput section] pvdisv1 = True
 
 or
 - Docker environmental gpvdisv1 = "True" (docker: -e gpvdisv1 = "True")
 
 For discussion on this functionality see issue  #52
 
 Hopefully this helps. 
 
 Did you already upgraded the Shinelink box (and RF sticks if possible)?  Maybe that solve the issue also. I do not have a Shinelink so I can not help with how to upgrade.

toaomatis, 2022-10-07:
@johanmeijer not sure how I can update either my Shinelink nor MOD 4000TL3-X inverter.

I do see that my Shinelink has Version V1.0, but I'm unable to log into the inverter to check which software / firmware version is installed.

johanmeijer, 2022-10-10:
I have upgraded my datalogger (wifishine-s) via the growatt server side, datalogger settings: 

[image]

toaomatis, 2022-10-13:
Unfortunately I am unable to do so. I do not have (or don't know here to find) a `Key to Save`. 

When I try to apply any Setting, the software gives me an error message: Key Wrong

johanmeijer, 2022-10-13:
The key is growatt + date (e.g growatt20221012).

toaomatis, 2022-10-17:
Unfortunately yesterday was another day with a ghost message at the start of day. This time I noticed it to late and was not able to delete the data for that day: https://www.pvoutput.org/intraday.jsp?id=79546&sid=93706&dt=20221017

I have not (yet) set the `gpvdisv1 = "True" ` as per your suggestion, while I'm still on the `ledidobe/grott:latest` Docker image

toaomatis, 2022-10-17:
Small update while I was going through the logging to see if I could make distinction between a good and a ghost telegram.

The ghost telegram appears more than I expected based on the PVOutput data, but sometimes PVOutput rejects a new status when the `v1` is too high:
```
// 20221007
- Grott PVOutput response: 
 -  Bad request 400: Energy value [16400] too high for [07:47]
// 20221009
- Grott PVOutput response: 
 -  Bad request 400: Energy value [15900] too high for [07:46]
// 20221010
- Grott PVOutput response: 
 -  Bad request 400: Energy value [16300] too high for [07:45]
```

See https://pvoutput.org/help/api_specification.html?highlight=addstatus#bad-request-400-energy-value-energy-too-high-for-time

While yesterdays ghost telegram only had `'v1': 6800` it was accepted because it was only 1.199kWh/kW 

A quick fix (for me at least) would be to skip uploading a new status to PVOutput in case the `'v2': 0.0`

johanmeijer, 2022-10-17:
I am not sure if do skip the record if the value if V2 = 0.0 does not give bad side affects in other situations. Lets me think about it. 

It still would be nice if you try gpvdisv1 = "True"  but you have to use the grott:beta version.

toaomatis, 2022-10-18:
Okay. I'll give the beta a try using the `gpvdisv1 = "True"` parameter.

mrfixit1, 2023-10-31:
Is anyone still working on or thinking about this one?

I recently switched to Grott and have the same issue. 
[image]

Config: MIC 2500TL-X inverter, ShineLanBox, Home Assistant OS 2023.10.5 on pi 3b, Grott stable branch (2.7) add-on with default parameters, MQTT add-on and integration.

johanmeijer, 2023-10-31:
?