johanmeijer/grott #186: ShineMaster/SmartEnergyManager reading the Smartmeter

Konni1234, 2022-10-11:
Hello, 

I have a ShineMaster with a connected Chint DTSU Meter, can I get the data from it ?

Curently I can readout only the data from the 2 Inverters.

Konni1234, 2022-10-11:
It seems to be that record.

`- Growatt original Data:
		 \x00\x01\x00\x06\x00\xb4\x24\x1b\x03\x2b\x2b\x47\x23\x33\x23\x77\x46\x5c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x30\x77\x7e\x7f\x51\x56\x4f\x77\xeb\x47\x4d\x7e\x5c\x5c\x5b\x52\x4d\x4d
		 \x69\x41\x43\x44\x58\x4c\x5a\x7f\x5e\x5d\x44\x51\x5a\x40\x6b\x40\x5c\x47\x4f
		 \x4d\x58\x75\x41\x5f\x59\x52\x58\x46\x69\x40\x43\x45\x4f\x44\x58\x75\x5c\x5d
		 \x5b\x4c\x45\x41\x76\x41\x41\x47\x4d\x59\x41\x75\x47\x41\x47\x4d\x59\x40\x71
		 \x42\x41\x40\x4d\x59\x41\x75\x45\x41\x43\x4d\x59\x46\x7e\x5c\x5c\x5b\x50\x5a
		 \x46\x6b\x5f\x5c\x46\x4f\x43\x58\x76\x5c\x5f\x5b\x4c\x44\x5a\x7e\x4b\x59\x5b
		 \x4c\x44\x5a\x7e\x4b\x57\x5b\x4c\x44\x5a\x7e\x4b\x5e\x5b\x4c\x44\x5a\x7e\x4b
		 \x57\x5b\x55\x4d\x5a\x77\x5e\x5f\x59\x51\x58\x41\x69\x42\x43\x9a\xcc
	 - Grott automatic protocol detection
	 - Grott data record length 188
	 - layout   :  T06241b
	 - no matching record layout found, try generic
	 - Record layout used :  T06241b
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 0001000600b4241b4459443042475730343300000000000000000000000000000000000000000
		 047160a0b162420008a3339392e332c3339392e332c3339382e382c3233302e342c3233302e39
		 2c3233302e332c322e322c322e302c322e322c2d313531332e302c2d3532352e302c2d3436302
		 e372c2d3532372e342c2d32392e332c312e322c2d33312e372c312e302c2d302e3939362c2d30
		 2e3939382c2d302e3939312c2d302e3939382c34392e302c302e302c352e302cedad
	 - Grott data ack record or data record not defined no processing done`

The Meterdevice is a Chint DTSU666.

I have also done some screenshots from the same time as the record.

[https://www.elko-tours.de/wp-content/uploads/2022/10/IMG_6409.png](url)
[https://www.elko-tours.de/wp-content/uploads/2022/10/IMG_6410.png](url)
[https://www.elko-tours.de/wp-content/uploads/2022/10/IMG_6411.png](url)
[https://www.elko-tours.de/wp-content/uploads/2022/10/IMG_6412.png](url)
[https://www.elko-tours.de/wp-content/uploads/2022/10/IMG_6413.png](url)
[https://www.elko-tours.de/wp-content/uploads/2022/10/IMG_6414.png](url)

Konni1234, 2022-10-12:
Data und Excel with historical data

	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\xad\x24\x1b\x03\x2b\x2b\x47\x23\x33\x23\x77\x46\x5c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x30\x77\x7e\x78\x4d\x5d\x41\x77\xe2\x47\x4d\x71\x5c\x57\x5b\x52\x4d\x41
		 \x69\x4a\x43\x44\x58\x41\x5a\x7f\x5e\x5d\x45\x58\x5a\x45\x6b\x40\x5d\x4e\x4f
		 \x45\x58\x75\x40\x57\x59\x51\x58\x44\x69\x41\x43\x47\x4f\x47\x58\x77\x5c\x5c
		 \x5b\x4c\x45\x46\x70\x5c\x58\x5b\x4c\x42\x42\x69\x4b\x43\x4e\x4f\x40\x58\x6a
		 \x45\x5f\x59\x51\x58\x47\x74\x5c\x5d\x5b\x4c\x40\x5a\x7f\x5e\x5b\x45\x4f\x40
		 \x58\x6a\x46\x41\x45\x4d\x59\x44\x69\x47\x5b\x43\x4d\x59\x44\x69\x4a\x56\x43
		 \x4d\x44\x5a\x76\x46\x5b\x5b\x4c\x44\x5a\x7e\x40\x57\x5b\x54\x44\x5a\x77\x5e
		 \x5f\x59\x51\x58\x4d\x69\x42\x43\xfe\x86
	 - Grott automatic protocol detection
	 - Grott data record length 181
	 - layout   :  T06241b
	 - no matching record layout found, try generic
	 - Record layout used :  T06241b
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 0001000600ad241b4459443042475730343300000000000000000000000000000000000000000
		 047160a0c0a2f2e00833339362e382c3339352e382c3339352e382c3232392e312c3232392e31
		 2c3232382e302c302e332c302e332c302e332c2d3132372e372c2d36362e392c392e342c2d373
		 02e302c33332e322c2d342e382c34322e342c2d342e322c2d302e3534342c2d302e3839342c30
		 2e3134342c2d302e3932382c35302e302c302e302c392e302c89e7
	 - Grott data ack record or data record not defined no processing done

[https://www.elko-tours.de/wp-content/uploads/2022/10/DYD0BGW043_36-Electric-meter-data-2022-10-12_2022-10-12.xls][(url)](https://www.elko-tours.de/wp-content/uploads/2022/10/DYD0BGW043_36-Electric-meter-data-2022-10-12_2022-10-12.xls)

Konni1234, 2022-10-13:
Hello, 

here is a sniplett from the complete Log (I habe in the config invtype sph set, because one of the inverters is a sph).

If you need, I can give you access via ssh to the system.
[https://www.elko-tours.de/wp-content/uploads/2022/10/grottproxylog.txt][(url)](https://www.elko-tours.de/wp-content/uploads/2022/10/grottproxylog.txt)

johanmeijer, 2022-10-13:
Thanks,

I uploaded a layout processing file for the smartmeter. You can find it in the example directory:

https://github.com/johanmeijer/grott/blob/2.7-(Beta)/examples/Record%20Layout/tchntthree.json  

If you copy this file to your Grott main directory (and restart Grott) it will be used to process the smartmeter data (layout T06241b). 

I did my best to map all values (use the names in the screenshots) but I am not totally convinced that everything is mapped well. 

I can not find the reverse/positive energy totals which are in the spreadsheet (maybe this calculated by Growatt). 

And I think there are multiple keywords which point to the same values. I mapped them all where possible. 

The following keywords point to the same values: 

      active_energy, positiveActiveTodayEnergy, posiActivePower  

      reactive_energy, reverseActiveTodayEnergy, reverActivePower    	
 
reactive_power and active_power also point to the same field but reactive only has a value when negative and active when positive (experience from other smart meters). 

ReverseActiveEnergy shows a positive value in the screenshot but is negative in "real life". Grott is not able yet to reformat this to a positive value (so the real value is -1513, the screenshots show 1513). 

Please test the layout and give feedback.

Konni1234, 2022-10-13:
Thanks, it's working :)

closed 2022-10-13 by Konni1234

johanmeijer, 2022-10-13:
@Konni1234 hopefully it is what you want. 

In version 2.7.7 (under construction) you will be able to set inverter type per inverter (see also: pull request #169)

Konni1234, 2022-10-14:
Yes, I've been waiting for that.

If it is useful, I could also connect an SDM630 to the Shinemaster this afternoon as a test.

johanmeijer, 2022-10-14:
The SDM630 data should be processed bij Grott. The challange migth be it has deffirent devicenummer with the shinemaster (for the Chint it was 24 (in T06241b). With other dataloggers (Like shinewifi/link) it uses 01 or 5x for that. I am not sure how Grott reacts on that. 

But that is usefull information so I can change Grott to be more generic.

Konni1234, 2022-10-14:
On the Shinemaster the SDM630 has folowing ID: T06221b

After changing the ID in grottconf from "T06501b" to 6221b the SDM630 reads out correctly.

`	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\xf7\x22\x1b\x03\x2b\x2b\x47\x23\x33\x23\x77\x46\x5c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x34\x77\x7e\x7a\x57\x61\x77\x77\xac\x44\x5a\x73\x43\x43\x47\x4f\x44\x47
		 \x6b\x5f\x5c\x41\x53\x5a\x40\x6b\x5f\x5b\x40\x52\x5a\x40\x6b\x5f\x5c\x41\x54
		 \x5a\x47\x6b\x42\x41\x47\x4d\x45\x41\x7f\x5c\x57\x5b\x51\x5a\x44\x6b\x41\x59
		 \x45\x4f\x40\x58\x73\x4b\x56\x59\x52\x58\x47\x71\x47\x41\x43\x4d\x59\x44\x69
		 \x4b\x43\x5a\x51\x5a\x4d\x6b\x5f\x5f\x59\x58\x58\x46\x75\x4a\x41\x40\x4d\x46
		 \x46\x7e\x5c\x5a\x5b\x53\x46\x4d\x69\x45\x43\x46\x4f\x41\x58\x75\x5c\x5e\x5b
		 \x50\x5a\x41\x6b\x5f\x5e\x45\x51\x45\x5a\x73\x5e\x5e\x40\x53\x5a\x45\x6b\x43
		 \x58\x45\x4f\x45\x58\x6a\x42\x41\x4e\x4d\x40\x4d\x69\x4b\x43\x47\x4f\x44\x58
		 \x77\x5c\x5c\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x44\x5a\x73\x5e\x5f\x59\x51
		 \x58\x44\x69\x42\x43\x47\x4f\x44\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41
		 \x47\x4d\x44\x5a\x73\x47\x43\x44\x58\x42\x5a\x7e\x5e\x5c\x4e\x56\x5a\x43\x6b
		 \x41\x56\x40\x4f\x45\x58\x91\x60
	 - Grott automatic protocol detection
	 - Grott data record length 255
	 - layout   :  T06221b
	 - Record layout used :  T06221b
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 0001000600f7221b4459443042475730343300000000000000000000000000000000000000000
		 043160a0e10131800cd302e34312c302e30332c2d3336322e342c2d3437332e342c2d3336352e
		 332c302e302c3135382e382c302e302c3336322e342c3439392e332c3336352e342c2d302e392
		 c2d302e392c2d302e392c3232382e372c3232392e352c3232392e372c312e352c322e312c312e
		 352c2d313230312e342c3137322e312c3137322e312c2d302e392c34392e392c302e302c302e3
		 32c302e302c302e302c302e342c302e302c302e302c302e302c302e302c302e302c302e302c30
		 2e34352c3339362e392c3339372e372c3339372e312cd612
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06221b

	 - Grott server date/time used
	 - Grott values retrieved:
		 -  datalogserial        :  DYD0BGW043
		 -  active_energy        :  0.41
		 -  reactive_energy      :  0.03
		 -  activePowerL1        :  -362.4
		 -  activePowerL2        :  -473.4
		 -  activePowerL3        :  -365.3
		 -  reactivePowerL1      :  0.0
		 -  reactivePowerL2      :  158.8
		 -  reactivePowerL3      :  0.0
		 -  apperentPowerL1      :  362.4
		 -  apperentPowerL2      :  499.3
		 -  apperentPowerL3      :  365.4
		 -  powerFactorL1        :  -0.9
		 -  powerFactorL2        :  -0.9
		 -  powerFactorL3        :  -0.9
		 -  voltageL1            :  228.7
		 -  voltageL2            :  229.5
		 -  voltageL3            :  229.7
		 -  currentL1            :  1.5
		 -  currentL2            :  2.1
		 -  currentL3            :  1.5
		 -  power                :  -1201.4
		 -  active_power         :  0
		 -  reverse_active_power :  -1201.4
		 -  apparent_power       :  172.1
		 -  reactive_power       :  172.1
		 -  power_factor         :  -0.9
		 -  frequency            :  49.9
		 -  posiActivePower      :  0.0
		 -  reverActivePower     :  0.3
		 -  posiReactivePower    :  0.0
		 -  reverReactivePower   :  0.0
		 -  apparentEnergy       :  0.4
		 -  totalActiveEnergyL1  :  0.0
		 -  totalActiveEnergyL2  :  0.0
		 -  totalActiveEnergyL3  :  0.0
		 -  totalRectiveEnergyL1 :  0.0
		 -  totalRectiveEnergyL2 :  0.0
		 -  totalRectiveEnergyL3 :  0.0
		 -  total_energy         :  0.45
		 -  l1Voltage2           :  396.9
		 -  l2Voltage3           :  397.7
		 -  l3Voltage1           :  397.1
		 -  device               :  SDM630`