johanmeijer/grott #187: Help with getting the input registers from growatt inverter

GiovaniAgsolve, 2022-10-13:
Hello there,

Just to the point, I suceed to get the holding registers from the Inverter and the holding registers from the Datalogger with the **grottserver** and GET requisitions, but **im not sure how to get the input registers from the Inverter** using the GET requisitions.

Please, can someone help me with that?

My growatt inverter is a SPF3500 that is equivalent to the SPF5000.
The registers list is here: 

The datalogger/inverter is communicating with a raspberry pi3 that have the grottserver.py running.

Impressive job so far, @johanmeijer and community 

best regards

johanmeijer, 2022-10-13:
@GiovaniAgsolve The input registers are sent in the data record and that is what Grott decodes and send to MQTT etc.

I am not aware of a command to retrieve these registers via the datalogger network interface. But if somebody knows we can built this in the API interface. 

In a future Grott / Grottserver release I can store the latest values sent by the datalogger in Grott and make an API query available for this if we think it adds value!

I am right that there is no registers list is include in your comment above?

GiovaniAgsolve, 2022-10-13:
> Am I right that there is no registers list is include in your comment above?

Oh, sorry, I forgot to send the link with the Input Registers list: https://docplayer.net/199800433-Growatt-offgrid-spf5000-modbus-rs485-rtu-protocol.html

> @GiovaniAgsolve The input registers are sent in the data record and that is what Grott decodes and send to MQTT etc.

I will try to make the Grott work together with the grottserver and debug. 

I was just trying to get the input registers with the grottserver running and using GET requisitions.

Thanks in advance!

johanmeijer, 2022-10-13:
Grottserver is build to work together with Grott only replace the IP in the [Growatt] section of grott.ini with the ip address of your grottserver (or localhost f it runs on the same server).  

The grottserver docker container runs both Grott and Grottserver. 

The idea of the datalogger (shinewifi, shinelan, shinelink) is to sent (push) the input regs to the growatt internet site to create the graphics / reports and are used by the growatt phone app). 

Reading/manipulating the holding registers (and the registers of the datalogger) are there for setting/reading the devices config remotely (I think).

GiovaniAgsolve, 2022-10-14:
> Grottserver is build to work together with Grott only replace the IP in the [Growatt] section of grott.ini with the ip address of your grottserver (or localhost f it runs on the same server).
>
> The [...]

Things are starting to work with grottserver and grott (sniffer mode)

Thank you very much :)

closed 2022-10-19 by GiovaniAgsolve